rtl-airband/RTLSDR-Airband #190: Highpass/Lowpass filters have no effect on "mixer"

cell21633, 2020-09-23:
I have noticed CTCSS coming thru my mix when my feed is set to combine to a stereo mix thru the mixer. Verified the highpass/lowpass filters work as designed when devices are set to output icecast directly. Or can someone verify my config? Thanks.

```
devices:
({
  type = "rtlsdr";
  index = 0;
  gain = 20;
  centerfreq = redacted;
  correction = 0;
  channels:
  (
    {
      disable = false;
      freq = redacted;
      modulation = "nfm";
      lowpass = 0;
      highpass = 300;
      afc = 1;
      outputs: (
        {
          type = "mixer";
          name = "mixer0";
          ampfactor = 4.0;
          balance = -1.0;
        }
      );
    },
    {
      disable = false;
      freq = redacted;
      modulation = "nfm";
      lowpass = 0;
      highpass = 300;
      afc = 1;
      outputs: (
        {
          type = "mixer";
          name = "mixer0";
          ampfactor = 4.0;
          balance = 1.0;
        }
      );
    }
  );
 }
);

mixers: {
 mixer0: {
  outputs: (
   {
    disable = false;
    type = "icecast";
    server = "redacted";
    port = redacted;
    mountpoint = "redacted";
    username = "source";
    password = "redacted";
   }
  )
 }
};
```

charlie-foxtrot, 2020-09-23:
If you want `lowpass` / `highpass` to apply to the mixer output, then it needs to be set in the mixer configuration, so like this:

```
mixers: {
 mixer0: {
  lowpass = 0;
  highpass = 300;
  outputs: (
   {
...
```

_**But**_ if all you are trying to do is suppress CTCSS and you're willing to run on the `unstable` branch, I would recommend disabling the low/high pass filters (setting to `-1`) for both the channels and the mixer, then using `notch` at the correct CTCSS frequency in Hz per-channel - see #159.  So something like this:

```
devices:
({
  type = "rtlsdr";
  index = 0;
  gain = 20;
  centerfreq = redacted;
  correction = 0;
  channels:
  (
    {
      disable = false;
      freq = redacted;
      modulation = "nfm";
      lowpass = -1;
      highpass = -1;
      notch = CTCSS frequency in Hz;
      afc = 1;
      outputs: (
        {
          type = "mixer";
          name = "mixer0";
          ampfactor = 4.0;
          balance = -1.0;
        }
      );
    },
    {
      disable = false;
      freq = redacted;
      modulation = "nfm";
      lowpass = -1;
      highpass = -1;
      notch = CTCSS frequency in Hz;
      afc = 1;
      outputs: (
        {
          type = "mixer";
          name = "mixer0";
          ampfactor = 4.0;
          balance = 1.0;
        }
      );
    }
  );
 }
);

mixers: {
 mixer0: {
  lowpass = -1;
  highpass = -1;
  outputs: (
   {
    disable = false;
    type = "icecast";
    server = "redacted";
    port = redacted;
    mountpoint = "redacted";
    username = "source";
    password = "redacted";
   }
  )
 }
};
```

cell21633, 2020-09-23:
hmm, maybe there is something wrong with my build, because the filtering doesn't work either when placing it in the mixers group. i shall try the unstable branch! thanks!

> If you want `lowpass` / `highpass` to apply to the mixer output, then it needs to be set in the mixer configuration, so like this:

closed 2020-09-23 by cell21633