rtl-airband/RTLSDR-Airband #193: Error compile

Chrisoo, 2020-11-12:
pi@raspberrypi3:~/RTLSDR-Airband $ make PLATFORM=rpiv2 NFM=1
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.2.0-1-g0e771ec\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp: In instantiation of ‘size_t AFC::check(const FFT_RESULTS*, size_t, float, unsigned char) [with FFT_RESULTS = GPU_FFT_COMPLEX; int STEP = 1; size_t = unsigned int]’:
rtl_airband.cpp:268:32:   required from ‘void AFC::finalize(device_t*, int, const FFT_RESULTS*) [with FFT_RESULTS = GPU_FFT_COMPLEX]’
rtl_airband.cpp:650:34:   required from here
rtl_airband.cpp:228:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if (bin < -STEP)
        ~~~~^~~~~~~
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.2.0-1-g0e771ec\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM -DWITH_RTLSDR   -c -o input-common.o input-common.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.2.0-1-g0e771ec\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM -DWITH_RTLSDR   -c -o input-helpers.o input-helpers.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.2.0-1-g0e771ec\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM -DWITH_RTLSDR   -c -o input-file.o input-file.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DRTL_AIRBAND_VERSION=\"v3.2.0-1-g0e771ec\" -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM -DWITH_RTLSDR   -c -o output.o output.cpp
output.cpp: In function ‘void shout_setup(icecast_data*, mix_modes)’:
output.cpp:94:37: error: ‘SHOUTERR_RETRY’ was not declared in this scope
  if (ret == SHOUTERR_BUSY || ret == SHOUTERR_RETRY)
                                     ^~~~~~~~~~~~~~
output.cpp:99:41: error: ‘SHOUTERR_RETRY’ was not declared in this scope
  while ((ret == SHOUTERR_BUSY || ret == SHOUTERR_RETRY) && shout_timeout-- > 0) {
                                         ^~~~~~~~~~~~~~
<wbudowane>: polecenia dla obiektu 'output.o' nie powiodły się
make: *** [output.o] Błąd 1

szpajder, 2020-11-13:
Sorry about that.

Please try the latest unstable branch, I've just pushed a fix there.

szpajder, 2020-11-13:
Released in 3.2.1.

Closing.

closed 2020-11-13 by szpajder