rtl-airband/RTLSDR-Airband #197: Using a config that defines two input "devices" both using the same "file" often doesn't produce the same output

charlie-foxtrot, 2020-11-24:
See sample config below.  Running on a mostly idle raspberry pi 4, some times `first_*.cs16` and `second_*.cs16` (and their corresponding `.mp3` files) match each other, but most times they do not.

Running multiple times, the `second_*.cs16` files match each other more often than the `first_*.cs16` files match each other.

Looking at the file sizes, the differences are multiples of 8000 bytes (the `WAVE_RATE`).

```
$ ll */*.cs16 | cut -d ' ' -f 4,8 | sort
1568000 output_9/first_20201123_16.cs16
1608000 output_2/first_20201123_16.cs16
1616000 output_1/first_20201123_16.cs16
1632000 output_4/first_20201123_16.cs16
1640000 output_5/first_20201123_16.cs16
1640000 output_8/first_20201123_16.cs16
1640000 output_9/second_20201123_16.cs16
1648000 output_1/second_20201123_16.cs16
1648000 output_3/first_20201123_16.cs16
1648000 output_6/first_20201123_16.cs16
1648000 output_7/first_20201123_16.cs16
1648000 output_7/second_20201123_16.cs16
1656000 output_2/second_20201123_16.cs16
1656000 output_3/second_20201123_16.cs16
1656000 output_4/second_20201123_16.cs16
1656000 output_5/second_20201123_16.cs16
1656000 output_6/second_20201123_16.cs16
1656000 output_8/second_20201123_16.cs16
```

```
localtime = true;
multiple_demod_threads = true;
devices:
(
  {
    type = "file";
    centerfreq = 151.10250;
    sample_rate = 2.40;
    filepath = "./capture.dat";
    channels:
    (
      {
        freq = 151.14500;
        modulation = "nfm";
        lowpass = -1;
        highpass = -1;
        outputs:
        (
          {
            type = "rawfile";
            directory = "./output/";
            continuous = true;
            filename_template = "first";
          },
          {
            type = "file";
            directory = "./output/";
            continuous = true;
            filename_template = "first";
          }
        );
      }
    );
  },
  {
    type = "file";
    centerfreq = 151.10250;
    sample_rate = 2.40;
    filepath = "./capture.dat";
    channels:
    (
      {
        freq = 151.14500;
        modulation = "nfm";
        lowpass = -1;
        highpass = -1;
        outputs:
        (
          {
            type = "rawfile";
            directory = "./output/";
            continuous = true;
            filename_template = "second";
          },
          {
            type = "file";
            directory = "./output/";
            continuous = true;
            filename_template = "second";
          }
        );
      }
    );
  }
);
```

charlie-foxtrot, 2020-11-24:
This feels like a race condition between he `demodulate` thread(s) and the output thread, I'm going to start looking there

charlie-foxtrot, 2020-11-24:
Can also reproduce with just a single "device" that outputs to both "file" and "mixer".  `mp3_cond` being set from multiple places (once per demodulate thread plus in the mixer thread) is likely the culprit.

charlie-foxtrot, 2020-11-25:
Here is what I've gathered so far:

- There is a thread per input "device", one or more demodulate threads, one output thread, and one mixer thread.
- Data is passed from the device thread(s) to demodulate thread(s) using a large circular buffer with a mutex per device
- Data is passed from the demodulate thread(s) to the output thread using `waveout` and `iq_out`, and although these arrays are `2 * WAVE_BATCH + AGC_EXTRA` large, writes always start at the beginning, are `WAVE_BATCH` values at a time, and there is no mutex.
- Each demodulate thread will signal `mp3_cond`
- The output thread waits for `mp3_cond`
- Data is passed from the output thread to the mixer thread by copying from `waveout` to the mixer's `wavein` that is protected by a mutex
- Data is passed from the mixer thread back to the output thread once again using `waveout` and there is no mutex
- The mixer thread also signals `mp3_cond`

So it looks like the problem is the data passing between the demodulate thread(s), mixer thread, and output threads are not thread safe and the output thread can be woken up by any one of the demodulate threads or the mixer thread and start processing data that may not be ready yet.

Two options that come to mind:
- Create an output thread for each demodulate and mixer thread and use different signals for each
- Generalize the already thread safe circular buffer code and use it between the demodulate, mixer, and output threads as well

@szpajder any preference or other thoughts?

szpajder, 2020-11-29:
>the output thread can be woken up by any one of the demodulate threads or the mixer thread and start processing data that may not be ready yet.

If there is any data to process from a particular device, `dev->waveavail` is set to 1. 

If there is any data to process from a mixer, `mixer.channel->state` is set to `CH_READY`.

Mixer loop and device loop in `output_thread` skips mixers and devices which don't have these flags set. Hence, no such thing as "processing data that may not be ready yet" can happen.

charlie-foxtrot, 2020-11-29:
at a minimum what can (and does) happen is the demodulate thread signals the output thread then re-writes to the same memory before the output thread is done reading.

I suppose my different file sizes by multiples of `WAVE_RATE` could be explained by loosing full blocks, ie the demodulate thread signals three times but the output thread only catches the first and third while the second gets overwritten.

szpajder, 2020-11-29:
Yes, it may happen when using an I/Q file as an input, because sample blocks are then delivered to the output thread with blazing speed. However during normal operation (ie. with an SDR) it's probably a non-issue.

In GNUradio there is a throttle block to deal with such problems. It is often required when reading samples from a file, but completely unnecessary when reading from a real radio.

charlie-foxtrot, 2020-11-30:
Confirmed I'm only loosing (and / or corrupting) output not generating new output.  Opened a PR to add some logging to verify this doesn't happen during normal operation: #201

charlie-foxtrot, 2020-12-02:
I'll start off by saying that what I'm doing may not be "normal", but I am seeing a somewhat constant low level of errors for the mixer input overrun as well as occasional overruns on both mixer and device outputs.  Im currently running 76 frequencies across 3 devices and into 7 mixers with a combination of file and ice cast outputs on raspberry pi 4.

Here is what the errors look like over a 6 hour window:

[image]

I tried first pulling off the mixer output into its own thread but that just moved errors to the the output overruns.  I have also tried an output thread for each device plus a dedicated output thread for the mixers and thats looking better.

charlie-foxtrot, 2020-12-03:
Opened #202 that makes things look much better when enabled:

[image]

charlie-foxtrot, 2021-01-10:
with the addition of #215 to throttle the input file, the majority of runs now have the same output.

closed 2021-01-10 by charlie-foxtrot