rtl-airband/RTLSDR-Airband #198: [Feature Request] Temporary Filenames during Write

Staja, 2020-11-25:
It would be useful to define a temporary filename extension for files still being written (output: file) and then to be renamed to the final filename after the file has been finished being written to.

This would be useful for directories being watched by other applications (rsync etc.) to exclude files still being written to.

szpajder, 2020-11-29:
Implemented in unstable branch. Please give it a try.

It appends `.tmp` suffix to the file currently being written to. When the file is closed, the file is renamed. If append mode is enabled and the `.mp3` file which is about to be opened already exists, it is renamed to `.mp3.tmp` and back to `.mp3` when closed.

Staja, 2020-12-01:
I'm running [ad58d24b96fab002c621192aca52160920f40a05](https://github.com/szpajder/RTLSDR-Airband/commit/ad58d24b96fab002c621192aca52160920f40a05) now, so far so good (except for #200 but that's unrelated.

Thank you for this feature, this makes rsync easier to use. For those wanting to use rsync to upload recorded files to remote target and then delete the local files, this is what I've been using:

Setup below assumes the following, change as per your requirements:
- a file output to `/var/local/recordings` on local system, same path exists on remote system
- `split_on_transmission` as true
- are using systemd on the system using rtl_airband.
- rtlairband user on local system and remote system
- Remote system files will be transferred to is `archive.local`
- Actual rsync command is spun out into bash script apart from service unit
- Said service unit uses it's own user
- SSH utilises pubkey authentication (in this case, using a ECC key)

**/etc/systemd/system/upload-recordings.service**
```
[Unit]
Description=Upload of Recordings to Remote Server for archiving

[Service]
User=rtlairband
Group=rtlairband

ExecStart=/home/rtlairband/upload-recordings.sh
```

**/home/rtlairband/upload-recordings.sh**
Actual rsync script.
```
#!/bin/bash
/usr/bin/rsync --remove-source-files --exclude '*.tmp' --omit-dir-times -rltzv -e "ssh -i /home/rtlairband/.ssh/id_ed25519" /var/local/recordings/. rtlairband@archive.local:/var/local/recordings
```

**/etc/systemd/system/upload-recordings.path**
This triggers a one-off service (which is our rsync) each time [a file which was open for writing gets closed](https://www.freedesktop.org/software/systemd/man/systemd.path.html) in /var/local/recording
```
[Unit]
Description=Triggers upload on new rtl_airband Recordings upon new Recordings

[Path]
PathChanged=/var/local/recordings
Unit=upload-recordings.service

[Install]
WantedBy=default.target
```
Make sure to enable and start relevant systemd services etc. with systemctl.

One may wish to have a systemd timer (every few minutes perhaps?) that runs another rsync command that uploads any files that may not have been uploaded due to a network fault, system restart etc.

szpajder, 2021-10-18:
Released in v4.0.0.

Closing.