rtl-airband/RTLSDR-Airband #200: SoapySDR readStream failed: OVERFLOW, disabling

Staja, 2020-11-26:
`SoapySDR device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]': readStream failed: OVERFLOW, disabling`

Issue looks similar to https://github.com/pothosware/SoapyAirspyHF/issues/11

This is with a brand new Airspy R2.

Journal:
```
Nov 26 19:16:33 redacted systemd[1]: Started SDR AM/NFM demodulator.
Nov 26 19:16:33 redacted rtl_airband[...]: SoapySDR: device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]': using native sample format 'CS16' (fullScale=65536.0)
Nov 26 19:16:33 redacted rtl_airband[...]: RTLSDR-Airband version 3.2.1 starting
...
Nov 26 19:16:41 redacted rtl_airband[...]: SoapySDR: device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]': sample rate set to 2500000 sps
Nov 26 19:16:41 redacted rtl_airband[...]: SoapySDR: device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]': AGC off (requested: off)
Nov 26 19:16:41 redacted rtl_airband[...]: SoapySDR: device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]': gain 'LNA' set to 15.0 dB
Nov 26 19:16:41 redacted rtl_airband[...]: SoapySDR: device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]': gain 'MIX' set to 15.0 dB
Nov 26 19:16:41 redacted rtl_airband[...]: [INFO] Using format CS16.
Nov 26 19:16:41 redacted rtl_airband[...]: SoapySDR: device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]': gain 'VGA' set to 5.0 dB
Nov 26 19:16:41 redacted rtl_airband[...]: SoapySDR: device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]' initialized
Nov 26 19:16:41 redacted rtl_airband[...]: SoapySDR: device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]' started
...
Nov 27 05:05:09 redacted rtl_airband[...]: SoapySDR device 'driver=airspy,serial=ffffffff:ffffffff,label=AIRSPY [ffffffff:ffffffff]': readStream failed: OVERFLOW, disabling
Nov 27 05:05:09 redacted rtl_airband[...]: Disabling device output
...
Nov 27 05:05:09 redacted rtl_airband[...]: All receivers failed, exiting
Nov 27 05:05:09 redacted rtl_airband[...]: Cleaning up
Nov 27 05:05:09 redacted rtl_airband[...]: Input threads closed
Nov 27 05:05:09 redacted rtl_airband[...]: Disabling device output
Nov 27 05:05:09 redacted rtl_airband[...]: O
Nov 27 05:05:09 redacted systemd[1]: rtl_airband.service: Succeeded.
```

szpajder, 2020-11-29:
It probably is a similar issue to the one you've pointed out, but I don't have such a device at hand to confirm this. 

Nevertheless, it's not something that can be fixed in rtl_airband's code.

Staja, 2020-11-29:
I've noticed that turning off manual squelch (it was flapping alot during these incidents) and relying on automatic squelch stopped the problem from occurring. I guess it doesn't like it when it's flapping?

szpajder, 2020-11-29:
Squelch and device handling are completely separate tasks. They run in separate threads and do not interact with each other.

Try adding `multiple_demod_threads = true'` in the top level of the config and check if it makes any difference.

Staja, 2020-12-01:
No improvement, in fact, it's less stable with that then just turning off automatic squelch alone.

Staja, 2020-12-09:
Looks like the Airspy R2 was running firmware years out of date, I've upgraded it to _AirSpy NOS v1.0.0-rc10-6-g4008185 2020-05-08_, we'll see if that improves things.

Staja, 2020-12-10:
No change in reliability with the new firmware.

szpajder, 2021-05-16:
Related: https://github.com/szpajder/RTLSDR-Airband/issues/247