rtl-airband/RTLSDR-Airband #203: "Icecast metadata updates in scan mode" seems not to work

ger-pakon, 2020-12-04:
Hey,

first thank you very much for your great work! 
I tried to configure, that the labels for the frequencies are shown by my own icecast server, but this din't work.

Is there any point, where i can troubleshoot this?

Thank you,
Pakon

szpajder, 2020-12-05:
Can you post your config file (without passwords, etc)?

Which version / branch of the program are you using?

ger-pakon, 2020-12-05:
Hey, I'm using the latest stable.

```
devices:
({
  type = "rtlsdr";
  index = 2;
  gain = 200.0;
  mode = "scan";
  multiple_demod_threads = true;
  channels: (
      {
      freqs = ( freq1, freq2, freq3, freq4, freq5);
      labels = ( "Label freq1", "Label freq2", "Label freq3", "Label freq4", "Label freq5");
      modulation = "am";
      afc = 0;
      squelch = 425;
      lowpass = 3200;
      highpass = 200;
      outputs: (
        {
          type = "icecast";
          server = "192.168.X.XX";
          port = 8000;
          mountpoint = "XXX.mp3";
          name = "ATC";
          genre = "ATC";
          description = "Tower and Ground";
          username = "<hidden>";
          password = "<hidden>";
          send_scan_freq_tags = true;
          shout_metadata_delay = 1;
        }
      );
    }
  );
});
```

szpajder, 2020-12-05:
Works fine for me. I can see metadata updates and frequency labels displayed in the player.

Does your player actually display icy-info updates as they happen? It might be worth taking a traffic dump with tcpdump to verify if the updates are sent to the server and then back to the client.

ger-pakon, 2020-12-05:
Hey,

thanks for testing. I even can't see something with tcpdump on the local interface. For which keywoards should I grep? I tried to grep for the labels. Which player do you use? I use VLC and an "auna" Internet Radio.

Thanks

szpajder, 2020-12-05:
I use mpv, but just checked VLC and it works equally well.

When the source sends metadata update to the server, it does that with a HTTP GET request to `/admin/metadata?mode=updinfo&mount=<mount_name>&song=<label>` . The server in turn sends updates to client with `StreamTitle=<label>`, so grep for that. For example you can do something like this on the client machine:

```
tcpdump -As0 -lni <interface_name> host <icecast_server_ip> and port 8000|grep --line-buffered StreamTitle
```

ger-pakon, 2020-12-05:
Hey,

I could do some tests:
1) Update the Streamtitle with curl is working and will shown on all my clients.
2) the `/admin` path is protected with basic auth, but i never set credentials 
  - I tried the same password and user as for user source (admin renamed to source) - nothing happened
3) I could see in tcpdump, when curl send the request to update the metada,I also could see this in the icecast access.log
4) I can't see any update from the rtlsdr-airband (lan/loopback interface) to the icecast server (of course after someone's speaking for serveral seconds)

So I think the problem is between icecast (2.4.4.) and the rtlsdr-airband. :-(

Thank you

szpajder, 2020-12-05:
I'm also on icecast 2.4.4 and it works fine.

Which libshout version are you using on the machine where rtl_airband is running? If it's 2.4.2 or later, then can you please try downgrading it to 2.4.1?

ger-pakon, 2020-12-05:
Hey,

its the following installed by ubuntu:

```
dpkg -l | grep libshout
ii  libshout3:amd64                                             2.4.3-1                             amd64        MP3/Ogg Vorbis broadcast streaming library
ii  libshout3-dev:amd64                                         2.4.3-1                             amd64        MP3/Ogg Vorbis broadcast streaming library (development)
```

It seems, I have to compile it for myself. Or will it been supported, soon?

Thanks,
Pakon

szpajder, 2020-12-05:
2.4.3 is buggy, don't use it. See here: https://github.com/szpajder/RTLSDR-Airband/issues/143

2.4.4 is still buggy, but less. 2.4.1 works best as of now.

ger-pakon, 2020-12-05:
I use:

```
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal
```

I downgraded to:
```
- http://ftp.de.debian.org/debian/pool/main/libs/libshout/libshout3_2.4.1-2_amd64.deb
- http://ftp.de.debian.org/debian/pool/main/libs/libshout/libshout3-dev_2.4.1-2_amd64.deb
```
I will post, if it works with it. :-)

ger-pakon, 2020-12-06:
Hey,

just wanted to let you know, that it is working with the downgraded packages, I mentioned before on Ubuntu 20.04.

Thanks for your help.

closed 2020-12-06 by ger-pakon

HagarViking, 2022-03-14:
Has this been corrected in libshout 2.4.5?

hypeARM, 2022-12-20:
How to downgrade libshout packages?

I'm in: rpiv3 armhf

feelingwalnut, 2022-12-20:
this is a question for the package manager of your distribution, which will likely be covered by their help/man files.
likely several other packages will need downgraded, then they will have their own dependencies that also need downgraded.

usually with something like sudo apt install <package>=<version>
or manually download the *.deb and sudo dpkg --install <filename.deb>
that'd be instructions for debian based linux distros.