rtl-airband/RTLSDR-Airband #207: Buffer Overflow message

khanab85, 2020-12-19:
Hi There,

Im trying to get rtl_airband to work. unfortunately, I'm facing an error that I can not find anywhere in the codebase or google.

As soon the program starts, I receive every few seconds a 
**Warning: buffer overflow**

Setup:
RPi3, 3 R820T Dongles - works fine with SoapySDR, rtl_tcp, dump1090 (even if all in parallel) 
rtl_airband version 3.2.1

Here are some logs that might help.

pi@rpi007:~ $ sudo rtl_airband -F -e
usb_claim_interface error -6
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Found Rafael Micro R820T tuner
[INFO] Opening Generic RTL2832U OEM :: stx:978:0...
Found Rafael Micro R820T tuner
SoapySDR: device 'driver=rtlsdr,serial=stx:978:0,buffers=5': using native sample format 'CS8' (fullScale=128.0)
RTLSDR-Airband version v3.2.1 starting
Connecting to 192.168.1.105:8000/xxx.mp3...
Connected to 192.168.1.105:8000/xxx.mp3
usb_claim_interface error -6
[INFO] Opening Generic RTL2832U OEM :: stx:978:0...
Found Rafael Micro R820T tuner
Exact sample rate is: 1048000.000278 Hz
[R82XX] PLL not locked!
SoapySDR: device 'driver=rtlsdr,serial=stx:978:0,buffers=5': sample rate set to 1048000 sps
SoapySDR: device 'driver=rtlsdr,serial=stx:978:0,buffers=5': AGC off (requested: off)
SoapySDR: device 'driver=rtlsdr,serial=stx:978:0,buffers=5': gain set to 25.0 dB
SoapySDR: device 'driver=rtlsdr,serial=stx:978:0,buffers=5' initialized
[INFO] Using format CS8.
SoapySDR: device 'driver=rtlsdr,serial=stx:978:0,buffers=5' started
**Warning: buffer overflow
Warning: buffer overflow
Warning: buffer overflow
...**

khanab85, 2020-12-19:
Update:

I have recompiled with Plattform **generic-arm7**

charlie-foxtrot, 2020-12-19:
For some reason github's search is missing it, but the code is here: https://github.com/szpajder/RTLSDR-Airband/blob/master/input-helpers.cpp#L67

The warning was added to call attention to the device writing data faster than the demodulation thread was reading it.  Things will work mostly fine despite the warning, but it does mean you're loosing samples.

Given you're running multiple devices, you can have a demodulation thread per device by adding `multiple_demod_threads = true;` to the top level of the config file (ie before the `devices` section).  That should help reduce the overflows.

Depending on how many outputs you have, you may also want to use `multiple_output_threads = true;` but that is only available in the `unstable` branch.

charlie-foxtrot, 2020-12-19:
> Update:
> 
> I have recompiled with Plattform **generic-arm7** without, the warning disappeard.

Did **generic-arm7** made it go away or removing **generic-arm7** made it go away?

More compile-time optimizations will make the demodulation thread faster and could allow a single thread to keep up.  Comments above about enabling multiple demodulation threads still holds.

khanab85, 2020-12-19:
thanks charlie-foxtrot for the explanation!

adding the **generic-arm7** made the error appear less often. It works fine although the warning keeps popping up.

closed 2020-12-19 by szpajder