rtl-airband/RTLSDR-Airband #208: Can't setup 2 devices for 2 different outputs in RTL-Airband

danielaregay, 2020-12-20:
# This is a minimalistic configuration file for RTLSDR-Airband. 
# More complex configurations are possible.
# Refer to https://github.com/szpajder/RTLSDR-Airband/wiki
# for description of keywords and config syntax.
# Just a single RTL dongle with one FM channel
# sent to a single Broadcastify stream. 
# Note that the center freq is 100 KHz offset from the receive freq
# Also, you must enter your server number, stream name, mountpoint and 
# password from your Broadcastify stream "Technicals"
devices:
({
  mode = "multichannel";
  type = "rtlsdr";
  index = 0;
  gain = 39;
  centerfreq = 118.9;
  channels:
  (
    {
      squelch = 59
      freq = 119.7;
      labels = "Approach/Departure";
        modulation = "am";
        highpass = 70;
             outputs:
(       
       {
       type = "icecast";
       server = "audio-in.liveatc.net";
       port = 8010;
       mountpoint = "xxxx_app";
       name = "HAAB Approach";
       genre = "ATC";
       username = "source";
       password = "password";
       send_scan_freq_tags = true;
       description = "Approach";      
      },
{
type = "file";
directory = "/home/pi/Approach";
filename_template = "Approach";
continuous = true;
append = true;
}

);
}, 
 {
       squelch = 56
       freq= 121.9;
       labels= "Ground";
       modulation= "am";
       highpass= 70;
 outputs: (
{    
          type = "icecast";
          server = "audio-in.liveatc.net";
          port = 8010;
          mountpoint = "xxxx_gnd";  # Note: Do not include a leading slash "/"
          name = "Ground";
          genre = "ATC";
          username = "source";
          password = "password";
},
{
type = "file";
directory = "/home/pi/Ground";
filename_template = "Ground";
continuous = true;
append = true;
}
       
);   
     }
      
);
}
);

(
{
   type = "rtlsdr";
  index = 1;
  gain = 44;
  centerfreq = 118.2;
  channels:
  (
    {
      squelch = 59
      freq = 118.1;
      labels = "Tower";
        modulation = "am";
        highpass = 70;
             outputs:
(       
       {
       type = "icecast";
       server = "audio-in.liveatc.net";
       port = 8010;
       mountpoint = "xxxx_twr";
       name = "Tower";
       genre = "ATC";
       username = "source";
       password = "password";
       send_scan_freq_tags = true;
       description = "Tower";      
      },

{
type = "file";
directory = "/home/pi/Tower";
filename_template = "Tower";
continuous = true;
append = true;
}
);
}
);
}
);

spykeboi, 2021-03-17:
I have this issue, shows syntax error for final line

szpajder, 2021-03-18:
Unnecessary parentheses between `device` sections.

[Here](https://github.com/szpajder/RTLSDR-Airband/blob/master/config/two_dongles_multiple_outputs.conf) is a working example for two devices.

closed 2021-03-18 by szpajder