rtl-airband/RTLSDR-Airband #221: [BUG] Wrong time in output file name. Doesn't use a time zone.

atlex, 2021-01-26:
I have wrong output file names. A file name doesn't include a time zone. In my case it's GMT +3.
I expect to have output file names with correct time zone.

RTLSDR-Airband version v3.2.1-19-g04104c8
Raspberry Pi 4B

$ ls -lah
-rw-r--r--  1 root root 2.1K Jan 26 09:44 144_20210126_064404_14450000.mp3
-rw-r--r--  1 root root 3.1K Jan 26 09:44 144_20210126_064417_14450000.mp3
-rw-r--r--  1 root root 3.7K Jan 26 10:00 144_20210126_070004_14450000.mp3
-rw-r--r--  1 root root 1.2K Jan 26 10:27 144_20210126_072716_14450000.mp3

$ date
Tue Jan 26 11:03:22 MSK 2021

$ timedatectl
Local time: Tue 2021-01-26 11:03:23 MSK
Universal time: Tue 2021-01-26 08:03:23 UTC
RTC time: n/a
Time zone: Europe/Moscow (MSK, +0300)
System clock synchronized: yes
NTP service: active
RTC in local TZ: no

rtl_airband.conf
...
{
  freq = 144.5;
  modulation = "nfm";
  bandwidth = 10000;
  outputs: 
  (
    {
      type = "file";
      directory = "/home/pi/recs/144500";
      filename_template = "144";
      include_freq = true;
      continuous = false;
      split_on_transmission = true;
      append = false;
    }
    ,
    {
      type = "mixer";
      name = "mixer1";
    }
  );
}
...

szpajder, 2021-01-26:
Add "localtime = true;" at the top level of your config.

atlex, 2021-01-27:
Thank you @szpajder ! it works for me! It looks like not documented feature  :)
Closed.

closed 2021-01-27 by atlex

szpajder, 2021-01-27:
Indeed, that's an omission. Thanks for pointing this out.

I have documented it [here](https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-file-outputs#local-timestamps-in-file-names).