rtl-airband/RTLSDR-Airband #234: [BUG] Warning: Buffer Overflow after a few hours

danielaregay, 2021-03-21:
After running for a few hours, there is no output from rtl_airband. When I run _sudo systemctl status rtl_airband_ , I get several "Warning Buffer Overflow" messages repeatedly. The only solution to that is to restart the service. Is there any method that can stop the problem? I have attached my rtl_airband.conf file.
[rtl_airband.txt](https://github.com/szpajder/RTLSDR-Airband/files/6177924/rtl_airband.txt)

Thanks in advance

szpajder, 2021-03-22:
Buffer overflows indicate that your computer is too slow to handle incoming samples in real time. Or maybe you are running too many other things on it.

LoganSound, 2021-03-28:
I'm learning systems right now, when you open /etc/systems/system/rtl_airband.service with a text editor does it have:
[Service]
Restart=always

Or 

Restart=on-failure

I've had problems with the latter but I did roll my own service which may not have been the best course of action😉

closed 2021-05-08 by szpajder