rtl-airband/RTLSDR-Airband #237: [FEATURE] RTL_TCP input source support

atlex, 2021-04-19:
It would be great to have the RTL_TCP input source support https://github.com/c4software/raspberry-rtlsdr-server
Like
devices:
({
  type = "rtl_tcp";
...
});

It's very handy to have the RTLSDR-Airband on Android like this https://github.com/merbanan/rtl_433/issues/892#issuecomment-502057557

I have created the guide how to install and configure the RTLSDR-Airband on Android. There are no compile issues and rtl_airband starts perfectly on Android 10. But the RTLSDR-Airband can't be configured to receive RTL_TCP as input :(

szpajder, 2021-04-19:
There is no rtl_tcp support, but there is SoapySDR support which means you can use SoapyRemote instead of rtl_tcp.

atlex, 2021-04-20:
> There is no rtl_tcp support, but there is SoapySDR support which means you can use SoapyRemote instead of rtl_tcp.

Unfortunately the SDR Driver https://play.google.com/store/apps/details?id=marto.rtl_tcp_andro can provide only RTL_TCP :(

nmaster2042, 2022-03-14:
There is a new SoapySDR driver for rtl_tcp: https://github.com/pothosware/SoapyRTLTCP