rtl-airband/RTLSDR-Airband #240: [BUG] rtl_airband: psymodel.c:576: calc_energy: Assertion `el >= 0' failed.

pschonmann, 2021-04-28:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit):
    - 3.2.1
  - `make` options used to build the program:
    - PLATFORM=rpiv2 NFM=1 make 
  - Hardware platform (eg. x86_64, Raspberry Pi v4):
    - Raspberry Pi 2 Model B Rev 1.1
  - Operating system name and version: 
    - Raspbian buster 10

**What happened?**
  -  After little time when i run RTLSDR-Airband i got error rtl_airband: psymodel.c:576: calc_energy: Assertion `el >= 0' failed.

**What you expected to happen?**
  - Not to throw error

**Steps to Reproduce**
 - Just ran the RTLSDR-Ariband with -f
 - 
**Additional context**
Add any other relevant information about the problem here.

**Your rtl_airband.conf file**
Remove passwords, server addresses and other private information.
```stats_filepath = "/home/pi/rtl_airband_stats.txt";
log_scan_activity = true;
devices:
({
  mode = "scan";
  type = "rtlsdr";
  index = 0;
  gain = 26;
  correction = 0;
  channels:
  (4
    {
      freqs = (67.600, 67.825, 67.9125, 68.200, 68.650, 69.075, 70.600, 70.825, 70.9125, 71.200, 71.650, 72.075, 67.950, 68.250, 70.950, 71.250, 80.975, 81.100, 81.150, 81.200, 81.225, 81.275, 81.350, 81.375, 173.075  );
          modulation = "nfm"
      bandwidth = 25000;
      outputs: (
        {
          type = "file";
      directory = "/home/pi/records/";
      include_freq = true ;
      filename_template = "BMIS";
        }
      );
    }
  );
 }
);
```

szpajder, 2021-04-29:
This seems to be a bug in lame encoding library, not in rtl_airband.

https://sourceforge.net/p/lame/bugs/498/

charlie-foxtrot, 2021-05-04:
FWIW the only times I have seen this was while trying different build flags (ie turning on / off `DEBUG`) and it went away after doing a full / clean rebuild.

Hermite123456, 2021-05-07:
> This seems to be a bug in lame encoding library, not in rtl_airband.
> 
> https://sourceforge.net/p/lame/bugs/498/

Hello, I am having the same issue (or similar issue)...

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

pi@raspberrypi:~/RTLSDR-Airband-3.2.1 $ make clean
make -C hello_fft clean
make[1]: Entering directory '/home/pi/RTLSDR-Airband-3.2.1/hello_fft'
rm -f *.o *.a
make[1]: Leaving directory '/home/pi/RTLSDR-Airband-3.2.1/hello_fft'
rm -f *.o rtl_airband

pi@raspberrypi:~/RTLSDR-Airband-3.2.1 $ make PLATFORM=armv8-generic
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DWITH_RTLSDR   -c -o input-common.o input-common.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DWITH_RTLSDR   -c -o input-helpers.o input-helpers.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DWITH_RTLSDR   -c -o input-file.o input-file.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DWITH_RTLSDR   -c -o output.o output.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DWITH_RTLSDR   -c -o config.o config.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DWITH_RTLSDR   -c -o util.o util.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DWITH_RTLSDR   -c -o mixer.o mixer.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DWITH_RTLSDR   -c -o input-rtlsdr.o input-rtlsdr.cpp
g++ -rdynamic  rtl_airband.o input-common.o input-helpers.o input-file.o output.o config.o util.o mixer.o input-rtlsdr.o  -lrt -lm -ldl -lvorbisenc -lmp3lame -lshout -lpthread -lconfig++ -lfftw3f -lrtlsdr -o rtl_airband

pi@raspberrypi:~/RTLSDR-Airband-3.2.1 $ sudo make install
install -d -o root -g root /usr/local/bin
install -o root -g root -m 755 rtl_airband /usr/local/bin
install -d -o root -g root /usr/local/etc
test -f /usr/local/etc/rtl_airband.conf || install -o root -g root -m 600 config/basic_multichannel.conf /usr/local/etc/rtl_airband.conf

 *** Done. If this is a new install, edit /usr/local/etc/rtl_airband.conf to suit your needs.

pi@raspberrypi:~/RTLSDR-Airband-3.2.1 $ journalctl -efn -u rtl_airband
May 07 13:01:51 raspberrypi systemd[1]: Started SDR AM/NFM demodulator.
May 07 13:01:51 raspberrypi rtl_airband[3944]: RTLSDR-Airband version 3.1.0 starting
May 07 13:01:53 raspberrypi rtl_airband[3944]: Found Rafael Micro R820T tuner
May 07 13:01:53 raspberrypi rtl_airband[3944]: [R82XX] PLL not locked!
May 07 13:01:54 raspberrypi rtl_airband[3944]: Allocating 10 zero-copy buffers
May 07 13:01:54 raspberrypi rtl_airband[3944]: Connecting to audio.rb24.com:8000/EXTVHF000464...
May 07 13:01:54 raspberrypi rtl_airband[3944]: Connected to audio.rb24.com:8000/EXTVHF000464
May 07 13:01:54 raspberrypi rtl_airband[3944]: Device #0: gain set to 32.80 dB
May 07 13:01:54 raspberrypi rtl_airband[3944]: RTLSDR device 0 initialized
May 07 13:02:05 raspberrypi rtl_airband[3944]: rtl_airband: psymodel.c:576: calc_energy: Assertion `el >= 0' failed.
May 07 13:02:05 raspberrypi systemd[1]: rtl_airband.service: Main process exited, code=killed, status=6/ABRT
May 07 13:02:05 raspberrypi systemd[1]: rtl_airband.service: Failed with result 'signal'.

szpajder, 2021-05-08:
I am unable to reproduce this during normal operation, but I found that libmp3lame triggers this assertion whenever an input sample equals NaN or infinity. I'm not yet sure what causes these values to appear in the audio buffer, but let's at least check if this is the root cause or not. I've pushed a change (commit 72a14cf4) which turns NaN and inifity samples into zeros. Can you please try the latest code from unstable branch and report back if the issue is solved? Just note that the build system in unstable has recently changed, so compilation instructions from the Wiki no longer apply. Please use instructions from README.md in unstable branch.

Hermite123456, 2021-05-08:
Thanks for the prompt reply.

I recompiled the unstable code. I am not seing the error anymore, but no
audio happens. Looks like it's waiting

May  8 11:44:28 raspberrypi systemd[1]: Started SDR AM/NFM demodulator.
May  8 11:44:30 raspberrypi kernel: [  612.036200] r820t 12-001a:
destroying instance
May  8 11:44:30 raspberrypi rtl_airband[5508]: Detached kernel driver
May  8 11:44:30 raspberrypi kernel: [  612.040342] dvb_usb_v2: 'Realtek
RTL2832U reference design:1-1.2' successfully deinitialized and disconnected
May  8 11:44:30 raspberrypi rtl_airband[5508]: Found Rafael Micro R820T
tuner
May  8 11:44:30 raspberrypi rtl_airband[5508]: [R82XX] PLL not locked!
May  8 11:44:30 raspberrypi rtl_airband[5508]: Allocating 10 zero-copy
buffers

  ... 2 minutes of inactivity ...

.... nothing happens at this point, audio burps one click (no actual
airband voice) every 15 seconds or so  ...

.... once I issue the stop command it looks like the program resumes and
continues. Looks like a hang state.  ...

May  8 11:46:26 raspberrypi systemd[1]: Stopping SDR AM/NFM demodulator...
May  8 11:46:27 raspberrypi kernel: [  729.557039] usb 1-1.2: dvb_usb_v2:
found a 'Realtek RTL2832U reference design' in warm state
May  8 11:46:27 raspberrypi kernel: [  729.613614] usb 1-1.2: dvb_usb_v2:
will pass the complete MPEG2 transport stream to the software demuxer
May  8 11:46:27 raspberrypi kernel: [  729.613634] dvbdev: DVB: registering
new adapter (Realtek RTL2832U reference design)
May  8 11:46:27 raspberrypi kernel: [  729.613644] usb 1-1.2: media
controller created
May  8 11:46:27 raspberrypi kernel: [  729.615149] dvbdev:
dvb_create_media_entity: media entity 'dvb-demux' registered.
May  8 11:46:27 raspberrypi kernel: [  729.629084] i2c i2c-11: Added
multiplexed i2c bus 12
May  8 11:46:27 raspberrypi kernel: [  729.629094] rtl2832 11-0010: Realtek
RTL2832 successfully attached
May  8 11:46:27 raspberrypi kernel: [  729.629139] usb 1-1.2: DVB:
registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
May  8 11:46:27 raspberrypi kernel: [  729.629155] dvbdev:
dvb_create_media_entity: media entity 'Realtek RTL2832 (DVB-T)' registered.
May  8 11:46:27 raspberrypi kernel: [  729.629274] r820t 12-001a: creating
new instance
May  8 11:46:27 raspberrypi kernel: [  729.636194] r820t 12-001a: Rafael
Micro r820t successfully identified
May  8 11:46:27 raspberrypi kernel: [  729.647447] rtl2832_sdr
rtl2832_sdr.0.auto: Registered as swradio0
May  8 11:46:27 raspberrypi kernel: [  729.647457] rtl2832_sdr
rtl2832_sdr.0.auto: Realtek RTL2832 SDR attached
May  8 11:46:27 raspberrypi kernel: [  729.647464] rtl2832_sdr
rtl2832_sdr.0.auto: SDR API is still slightly experimental and
functionality changes may follow
May  8 11:46:27 raspberrypi kernel: [  729.659136] Registered IR keymap
rc-empty
May  8 11:46:27 raspberrypi kernel: [  729.659236] rc rc0: Realtek RTL2832U
reference design as
/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/rc/rc0
May  8 11:46:27 raspberrypi kernel: [  729.659370] rc rc0: lirc_dev: driver
dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
May  8 11:46:27 raspberrypi kernel: [  729.659477] input: Realtek RTL2832U
reference design as
/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/rc/rc0/input2
May  8 11:46:27 raspberrypi kernel: [  729.659769] usb 1-1.2: dvb_usb_v2:
schedule remote query interval to 200 msecs
May  8 11:46:27 raspberrypi rtl_airband[5508]: Reattached kernel driver
May  8 11:46:27 raspberrypi rtl_airband[5508]: RTLSDR-Airband version
4.0.0pre-72a14cf starting
May  8 11:46:27 raspberrypi rtl_airband[5508]: Connecting to
audio.rb24.com:8000/EXTVHF000464...
May  8 11:46:27 raspberrypi rtl_airband[5508]: Connected to
audio.rb24.com:8000/EXTVHF000464
May  8 11:46:27 raspberrypi rtl_airband[5508]: Device #0: gain set to 33.80
dB
May  8 11:46:27 raspberrypi rtl_airband[5508]: RTLSDR device 0 initialized
May  8 11:46:27 raspberrypi rtl_airband[5508]: Got signal 15, exiting
May  8 11:46:27 raspberrypi kernel: [  729.667862] usb 1-1.2: dvb_usb_v2:
'Realtek RTL2832U reference design' successfully initialized and connected
May  8 11:46:27 raspberrypi rtl_airband[5508]: Cleaning up
May  8 11:46:27 raspberrypi rtl_airband[5508]: Input threads closed
May  8 11:46:27 raspberrypi rtl_airband[5508]: Disabling device outputs
May  8 11:46:27 raspberrypi rtl_airband[5508]: Closing connection to
audio.rb24.com:8000/EXTVHF000464
May  8 11:46:27 raspberrypi rtl_airband[5508]: Closing output thread(s)
May  8 11:46:27 raspberrypi systemd[1]: rtl_airband.service: Succeeded.
May  8 11:46:27 raspberrypi systemd[1]: Stopped SDR AM/NFM demodulator.
May  8 11:46:28 raspberrypi systemd-udevd[6054]: Process '/usr/sbin/th-cmd
--socket /var/run/thd.socket --passfd --udev' failed with exit code 1.

szpajder, 2021-05-08:
Please post your config file. And also run the program directly from the shell, in foreground (ie. use the `-f` option) and observe the textual waterfalls - whether they are running and what signal and noise level values do you get.

HagarViking, 2021-05-08:
Did you blacklist the Linux kernel drivers dvb_usb_rtl28xxu, rtl2832 and rtl2830?

Hermite123456, 2021-05-08:
Well, I'm a novice here... so bare with me :) I know nothing about
'blacklist'.

Below is my config followed by the shell command and CTRL-C to stop it.

devices:
({
  type = "rtlsdr";
  # index = 0;
  serial = "00000040";
  sample_rate = 2.4;
  gain =33.8;
  centerfreq = 121000000;
  correction = 0;
  mode = "scan";
  channels:
  (
    {
      modulation = "am";
      bandwidth = 8330;
      #squelch = 25 ;  # Uncomment to set squelch manually
       freqs = (
 118400000,
 118900000,
 119100000,
 119300000,
 119900000,
 120425000,
 121000000,
 121300000,
 121800000,
 121900000,
 122075000,
 122400000,
 122700000,
 123200000,
 123500000,
 124300000,
 124650000,
 125150000,
 125600000,
 126400000,
 126700000,
 126900000,
 132850000
 );

 labels = (
 "St-Hubert Tower",
 "Dorval Approach/Departure",
 "Mirabel Tower",
 "Dorval Tower",
 "Dorval Tower",
 "Dorval Approach/Departure",
 "Dorval Ground/Apron (West)",
 "St-Hubert Tower",
 "Mirabel Ground Advisory",
 "Dorval Ground/Apron (East)",
 "Dorval Ground/Apron (East)",
 "Mirabel Apron",
 "St-Hubert",
 "Sorel-Tracy Tower",
 "Dorval Bombardier Ops",
 "Dorval Tower",
 "Dorval Approach/Departure",
 "Dorval Approach/Departure",
 "Dorval Approach/Departure",
 "St-Hubert Ground",
 "Dorval Quebec FIC",
 "Dorval Approach/Departure",
 "Dorval Approach/Departure"
 );
      outputs: (
        {
 type = "icecast";
 server = "audio.rb24.com";
          port = 8000;
          mountpoint = "EXTVHF000464";
          name = "Hermite’s Transmitter";
          stream_name = "Montréal–Trudeau, St-Hubert, Mirabel Airports"
          genre = "ATC";
          Description = "My local Airports"
          username = ***@***.***";
          password = "xxxxxxxx";
 send_scan_freq_tags = true;
        }
# If you need more outputs, put here

      );
    }
  );
 }
);

***@***.***:~ $ sudo /usr/local/bin/rtl_airband -c
/home/pi/RTLSDR-Airband/config/rtl_airband.conf -f
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Allocating 10 zero-copy buffers

 118.400

 -30/-47  118.900
 -45/-47  118.900
 -43/-47  119.100
 -39/-47  119.100
 -41/-47  119.300
 -47/-47  119.300
 -45/-47  119.900
 -45/-47  119.900
 -45/-46  120.425
 -41/-46  120.425
 -42/-47  121.000 ^CReattached kernel driver
***@***.***:~ $

HagarViking, 2021-05-08:
The log is littered with dvb references so it seems to me it's trying to use the kernel DVB-T driver for the dongle.

https://www.reddit.com/r/RTLSDR/wiki/blacklist_dvb_usb_rtl28xxu

Hermite123456, 2021-05-08:
***@***.***:~ $ journalctl -efn -u rtl_airband

-- Logs begin at Sat 2021-05-08 13:19:54 EDT. --
May 08 13:20:00 raspberrypi systemd[1]: Started SDR AM/NFM demodulator.
May 08 13:20:01 raspberrypi *rtl_airband[580]: usb_claim_interface error -6*
May 08 13:20:01 raspberrypi systemd[1]: rtl_airband.service: Main process
exited, code=exited, status=1/FAILURE
May 08 13:20:01 raspberrypi systemd[1]: rtl_airband.service: Failed with
result 'exit-code'.
^C
***@***.***:~ $ *sudo rmmod dvb_usb_rtl28xxu rtl2832*

***@***.***:~ $ sudo systemctl start rtl_airband

***@***.***:~ $ journalctl -efn -u rtl_airband

-- Logs begin at Sat 2021-05-08 13:19:54 EDT. --
May 08 13:20:00 raspberrypi systemd[1]: Started SDR AM/NFM demodulator.
May 08 13:20:01 raspberrypi* rtl_airband[580]: usb_claim_interface error -6*
May 08 13:20:01 raspberrypi systemd[1]: rtl_airband.service: Main process
exited, code=exited, status=1/FAILURE
May 08 13:20:01 raspberrypi systemd[1]: rtl_airband.service: Failed with
result 'exit-code'.
May 08 13:21:58 raspberrypi systemd[1]: Started SDR AM/NFM demodulator.
May 08 13:22:00 raspberrypi rtl_airband[2789]: Found Rafael Micro R820T
tuner
May 08 13:22:00 raspberrypi rtl_airband[2789]: [R82XX] PLL not locked!
May 08 13:22:00 raspberrypi rtl_airband[2789]: Allocating 10 zero-copy
buffers

szpajder, 2021-05-08:
The config is fine, auto squelch is enabled and the values in the waterfall look OK. So the squelch does not open at all on any channel? Maybe there is simply no traffic? Are you able to confirm this with another radio (eg. a scanner) running simultaneously?

Also note that a single scan round over 23 frequencies takes nearly 5 seconds. If transmissions are very short, you will probably miss most of them.

Hermite123456, 2021-05-08:
> My station was running fine until I started receiving the 'Assertion
> failed error'.
> There are plenty of active frequencies in that scan range.
> Listen to the occasionally 'clicks' on my station here:
>  -37/-42  119.300
>  -39/-42  119.300
>  -38/-42  119.900
>  -39/-43  120.425
>  -45/-46  132.850
>  -46/-46  132.850
>   -8/-47  118.400
>   -8/-40  118.900
>  -33/-40  118.900
>  -34/-42  119.100
>  -37/-42  119.100
> Thanks for trying to debug.

I believe this shows an active frequency?
Does the '*' symbol indicate that it locked on that frequency?

***@***.***:~ $ sudo /usr/local/bin/rtl_airband -c
/home/pi/RTLSDR-Airband/config/rtl_airband.conf -f
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Allocating 10 zero-copy buffers

118.400

 -46/-39  118.400
 -29/-41*** 118.400
 -29/-41*** 118.400
 -29/-41  118.400
 -30/-41  118.400
 -29/-41  118.400
 -29/-41  118.400
 -29/-41  118.400
 -30/-41  118.400
 -29/-41  118.400
 -29/-41  118.400
 -29/-41  118.400 ***@***.***:~ $ ^C
***@***.***:~ $

szpajder, 2021-05-08:
Yes, the asterisk appears when the squelch is open.

Please do these obvious things:

- Blacklist the DVB-T driver to make sure they are not loaded automatically when the device is plugged. Create `/etc/modprobe.d/rtl-blacklist.conf` file and put these lines in there:

```
blacklist r820t
blacklist rtl2832
blacklist rtl2830
blacklist dvb_usb_rtl28xxu
```

- Reboot the machine, replug the dongle, run `lsmod` to list currently loaded modules - verify whether blacklisting actually works (ie. none of the above modules should be present on the list).
- Make sure the dongle actually works. Maybe it broke? Connect it to a machine where you have a GUI, run an SDR console there, verify if you have a signal (remember about setting the gain to the same value as you have it set in `rtl_airband.conf`).
- Is the `correction` value correct for this dongle? Maybe it's not? If it's off by 5 or 10, then it's not a big deal, but if it's off by 80 or more, then you may miss tranmissions due to large tuning error.
- Do you have more than one dongle connected to this computer? If you do, triple-check if rtl_airband is using the correct one. Better off disconnect all others to be 100% sure.
- Make another config file for this dongle, but in multichannel (not scan) mode. Configure one channel with just one frequency which is most active and which you have best reception of. Run the program in foreground, verify if it works. If it does, then switch to scan mode and add more frequencies.
- Remove the `bandwidth` setting, just for a test. Also, setting it to a value as high as 8330 means there is basically no filtering done, so you are better off without that.

Hermite123456, 2021-05-09:
All is good now. Turned out I changed my Antenna and it had a much higher
gain. I had to drop my to 19.7.
The Assertion failed error is gone.
Thanks for all your work.

pschonmann, 2021-05-09:
Commented bandwidth make rtl-airband working again.