rtl-airband/RTLSDR-Airband #259: Some emitters are good but other are very low, even the tower !

harf18, 2021-06-10:
**Describe your environment**

  - RTLSDR-Airband version 3.1.0 starting
  - make PLATFORM=rpiv2
  - Raspberry Pi v3 
  - Raspbian GNU/Linux 10 (buster)
  - External Antenna Tagra GP-AIR 1/4 at 110 meters from the tower off the airport LFBI (France)
  - Key : USB 2.0 TV Stick DVB-T + Dab + FM RTL2832U
  - startup log : Found Rafael Micro R820T tuner / [R82XX] PLL not locked! / Device #0: gain set to 29.70 dB / RTLSDR device 0 initialized

**What happened?**

Some emitters (plane, helicopter, on park, on air...) are very clear but others are very low (unintelligible) like the tower ! however, the external antenna are from 110 metters from the tower.
I tried serveral configuration (gain, correction, afc, squelch...) but i can't improve the result.

**What you expected to happen?**

I expect to hear the tower like the other emitters

**Your rtl_airband.conf file**
devices:
({
  type = "rtlsdr";
  index = 0;
  gain = 30;
  correction = 10;
  centerfreq = 118.5;
  channels:
  (
    {
      freq = 118.5;
      squelch = 30;
      afc = 2;
      outputs: (
        {
          type = "icecast";
          server = "xxxxxxxxxxx";
          port = xxxx;
          mountpoint = "TWR.mp3";
          name = "Tower";
          genre = "ATC";
          username = "xxxxx";
          password = "xxxxx";
        }
      );
    }
  );
 }
);

HagarViking, 2021-06-10:
I'd say try lowering your gain to like 15-20 or even more and see if this improves reception of the tower. 
I have a feeling the dongle is overloaded - being so close and with a decent antenna.

If it improves you can try to increase gain gradually until you hit the limit.

I had issues being a little less than a km from an airport with a less efficient antenna and with that kind of gain. Running two frequencies would even bleed the signal on to both streams. I ended up with around gain 25 as a compromise between not distorting the powerful transmissions and receiving the weakest.

harf18, 2021-06-10:
Lowering the gain solve my problem. 10 is the good value
Thank you very much @HagarViking

closed 2021-06-10 by harf18