rtl-airband/RTLSDR-Airband #260: [BUG]

thowe-switzerland, 2021-07-17:
**Describe your environment**

  - RTLSDR-Airband version: unstable from git, 891dc48
  - `make` options used to build the program: cmake -DPLATFORM=armv8-generic -DSOAPYSDR=1 ../
  - Operating system name and version:  Raspberry Pi OS, Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux

Hardware:
- Raspberry Pi 4B 4GB RAM (new, tried with original and alternative strong enough power supply)
- 2x SDRPlay RSP1A 

Other Software:
- SoapySDR 0.8.0
- SoapySDRPlay 0.3.0
- SDRplay_RSP_API-ARM32-3.07.2
 

**What happened?**

I have a configuration with two RSP1A devices connected via SoapySDR. Each device receives two different ATC frequencies each as channel:
Device 1: TWR (121.025) + CLD (2 Channels) (Center Frequency 121.5, Sample Rate: 3.0) 
Device 2: APP (127.325) + ATIS (2 Channels) (Center Frequency 126.15, Sample Rate: 3.0)

Basically everything works fine:
- The MP3 of all channels are just perfect
- an Icecast stream from a mixer also sounds very good

For a research project I also provide the RAW of the channels TWR and APP as CS16. And here is the problem: The RAW of the TWR channel also sounds good and corresponds to the generated MP3. The RAW of the APP channel, on the other hand, is completely muted with weak signals and extremely noisy with very strong signals. Quite obviously, extremely much weaker signals are recorded in the RAW than they are in the MP3.

I could not explain for a long time how it can be that the MP3 of the channel APP is excellent and the CS16 RAW file contains hardly anything and even the strongest signals contain only weak and noisy. Now I have a hypothesis, which of course can be correct but of course can also be wrong: The APP channel comes from the second RSP1A device. Can it be that due to an error, the RAW data of the same frequency but from the first RSP1A end up in the file? This would explain that the signal is so extremely weak and noisy, since the APP frequency on the first RSP1A would be clearly outside the bandwidth and is still contained at most very weakly. I can't really explain the effect any other way.

I have tried to find an error in the code. So far unsuccessfully.

**What you expected to happen?**

That RAW CS16 files and MP3 have the same audio content.

**Steps to Reproduce**

Configure two RSPA1 via SoapySDR. Configure two channels on each. Record channels each as MP3 and CS16. Demodulate CS16 with csdr. MP3 and manually demodulated CS16 file from the first RSP1A are equivalent. MP3 and manually demodulated CS16 file from second RSP1A are totally different. MP3 good. CS16 extremely weak signal.  

**Additional context**

Nothing special. For specific questions, I will be happy to provide more details or perform tests.

**Your rtl_airband.conf file**

```
fft_size = 1024;

devices:
(
 
 { 
  type = "soapysdr";
  device_string = "driver=sdrplay,serial=200*******,biasT_ctrl=true"
  gain = "RFGR=7,IFGR=42";

  sample_rate = 3.0;
  centerfreq = 121.500;

  channels:
  (
    {
      freq = 121.025;
      outputs: (
        {
           type = "file";
           directory = "/home/pi/rtlsdr_airband_data/twr_mp3/";
           filename_template = "TWR";
           continuous = true;
           append = true;
        },
        {
           type = "rawfile";
           directory = "/home/pi/rtlsdr_airband_data/atco2";
           filename_template = "Tower";
           split_on_transmission = true;
        },
        {
           type = "mixer";
           name = "mixer";
           balance = -0.8;
           ampfactor = 1.0;
        }
      );
    },
    {
      freq = 121.900;
      outputs: (
        {
           type = "file";
           directory = "/home/pi/rtlsdr_airband_data/cld_mp3/";
           filename_template = "CLD";
	       continuous = true;
           append = true;
        }
      );
    }
  );
 },

 {
  type = "soapysdr";

  device_string = "driver=sdrplay,serial=200******,biasT_ctrl=false"
  gain = "RFGR=7,IFGR=40";

  sample_rate = 3.0;
  centerfreq = 126.150;

  channels:
  (
    {
      freq = 127.325;
      outputs: (
        {
           type = "file";
           directory = "/home/pi/rtlsdr_airband_data/arr_dep_mp3/";
           filename_template = "ARR_DEP";
    	   continuous = true;
           append = true;
        },
        {
           type = "rawfile";
           directory = "/home/pi/rtlsdr_airband_data/atco2";
           filename_template = "Approach";
           split_on_transmission = true;
        },
        {
           type = "mixer";
           name = "mixer";
           balance = 0.8;
           ampfactor = 1.0;
        }
      );
    },
    {
      freq = 125.125;
      outputs: (
        {
           type = "file";
           directory = "/home/pi/rtlsdr_airband_data/atis_mp3/";
           filename_template = "ATIS";
           continuous = true;
           append = true;
        },
        {
           disable=true;
           type = "mixer";
           name = "scanner";
           balance = 1.0;
           ampfactor = 1.0;
        }
      )
    }
  )
 }
);

mixers: {
  mixer: {
    outputs: (
         {
          type = "file";
          directory = "/home/pi/rtlsdr_airband_data/twr_arr_dep_mp3/";
          filename_template = "Stereo_L_Twr_R_Arr_Dep";
          continuous = true;
          append = true;
        },
        {
          type = "icecast";
          server = "aaa.bbb.ccc.ddd";
          port = 1234;
          mountpoint = "stream";
          username = "*******";
          password = "******";
        },
    );
  }
};

```

szpajder, 2021-07-17:
>
>
What happens when you shift the center frequency of the second device
slightly (say, by 5 or 10 kHz, either way)? Is there any difference?

Can you please post an example cs16 file with garbled audio?

thowe-switzerland, 2021-07-17:
**Frequency Shift / Offset**
I'll do the test with the slight frequency offset and attach the result as soon as I have it.

**Example Files**
In a ZIP I have attached examples each as MP3, CS16, FLAC (converted from CS16):
[Comparison.zip](https://github.com/szpajder/RTLSDR-Airband/files/6834660/Comparison.zip)

**About the files contained in the ZIP:**
Reference (OK):
- twr-example-1.mp3|cs16|flac --> normal example twr freq.

Problem examples (NOK):
- app-example-1.mp3|cs16|flac --> very weak station missing, distortion
- app-example-2.mp3|cs16|flac --> weak station missing, distortion

These are very typical examples.

thowe-switzerland, 2021-07-17:
**Frequency Shift / Offset**
I shifted the center frequency of the second device by +50kHz to `centerfreq = 126.200;`.

This has **not** changed the situation. The behavior is identical.

szpajder, 2021-07-18:
I have pushed a change to the unstable branch, which might fix this. Can
you please try it out?

A few more questions:

- Are you using an active antenna or an in-line preamp?
- The first device has bias T turned on, while the second one has it off.
Is this intentional?
- Are both receivers connected to the same antenna (via a splitter of some
kind) or do they have separate antennas?

thowe-switzerland, 2021-07-18:
Thanks for the change! I'll build and install the new version right away and report the findings here.

Regarding your questions:

**- Are you using an active antenna or an in-line preamp?**
The antenna is a [Civil Airband Slim Jim from Broadsword Antennas](https://www.facebook.com/commerce/products/4208785675822780/?ref=mini_shop_storefront&referral_code=mini_shop_page_shop_tab_cta). So a special passive dipole. However, I actually have a filtered [Airband LNA from gpiolabs](https://gpio.com/collections/filtered-low-noise-amplifier/products/airband-filtered-low-noise-amplifier-with-bias-tee) in use with powering via Bias-T. Resulting gain 16db after bandpath filter 118-136MHz. I compensate for the 16db gain by increasing the gain reduction at the input of the RSPA1 by about 2 RFGR steps, so that it is not overdriven.

**- The first device has bias T turned on, while the second one has it off. Is this intentional?**
Yes, this is intentional. One of the RSPA1 should feed the LNA. I think one is enough. Maybe you could have both of them powering the LNA in parallel. But it seems unnecessary to me.

**- Are both receivers connected to the same antenna (via a splitter of some kind) or do they have separate antennas?**
The former. I.e. both RSP1A are connected to the same antenna line. I.e. after the filtered LNA I have a simple coax splitter.Since this splitter roughly halves the antenna signal and thus introduces 3db loss for one receiver, I inserted the filtered LNA before the splitter to compensate for the attenuation.

**Scheme:**
```
    |
    |
    |                                                                    |--------- RSPA1 (#1, Bias-T on)  --> Feeding TWR + CLD
     ------------------- [ LNA -- Bandpath Filter 118-136 ] -------------|
   Ant                                                                   |--------- RSPA1 (#2, Bias-T off) --> Feeding APP + ATIS
```

I can not exclude that the two RSPA1 after the LNA are currently somewhat "hot" supplied with RF signal. But I think that the resulting 13db should still be ok, especially since strong out-of-band signals (FM radio etc.) are strongly suppressed thanks to the filter. And I increase the input attenuation accordingly (which corresponds to a reduction of the gain in the internal LNA in RSPA1).

The signals demodulated by RTLSDR_Airband are also clean, well modulated and ok on both RSPA1. It can't be that such a miserable IQ signal results in such perfect MP3... Or can it?

szpajder, 2021-07-18:
This means that you are feeding +5V DC into the second RSP-1A input.

I would grab an SDR console of some kind and verify whether or not this
causes any adverse effects (ie. whether the overall levels on both RSPs
differ much or not).

thowe-switzerland, 2021-07-18:
**Regarding your thoughts about the Bias-T in this setup:**
I will check with the manufacturer. I think I read once that they can handle it and it should have no effect.

Likewise, I will swap the roles of the two regarding Bias-T times and see if then the other has a problem.

**Regarding the new software version:**
I have successfully built and installed the new version. And what can I say: You are once again a hero. :-)

From my point of view the quality of the CS16 files is **now worlds better**. While before the MP3 sounded amazingly much better, they now match each other very closely. Fantastic!

Here is an example, just recorded:
[Comparison-after-cs16fix.zip](https://github.com/szpajder/RTLSDR-Airband/files/6836114/Comparison-after-cs16fix.zip)

**About the files contained in the ZIP:**
app-example-afterfix-1.mp3 --> two transmissions combined: ATC, then ACFT
app-example-afterfix-1a.cs16|flac --> ATC (split mode)
app-example-afterfix-1a.cs16|flac --> ACFT (split mode)

I understand more and more what all is behind it. And impressive how you identified and solved the problem. And in world record time. I think I would like to send you a donation or a gift as a thank you. Thank you very much!

szpajder, 2021-07-18:
Some time ago the way input samples are scaled before being processed was
changed. This was necessary for the squelch config and waterfalls to speak
in dBFS rather than in some random units that were unrelated to anything.
So internally the program now works on samples scaled to the range of -1.0
to +1.0, while before the change it was -127.0 to 127.0. This means the I/Q
samples that are going out to rawfile outputs have to be upscaled back to a
larger range, because the number of integer values between -1 and +1 is...
quite small :-)

I guess that 3.2.1 stable release would also work correctly, since the bug
was introduced later on, in the unstable branch. Thanks for reporting and
verification.

thowe-switzerland, 2021-07-18:
Thanks for the explanations. It all sounds very reasonable. However...

...as always when something is fixed, it is then stuck in another place: For the sake of completeness, I have now checked the tower channel with the new version. This seems now quite clearly overdriven. While the planes are still ok, the (strong) tower is distorted and clipped in flac generated from cs16 - while once again the MP3 sound perfectly.

Actually TWR ATC and APP ATC should be comparably strong. Quite astonishing. While before TWR was ok and APP was not, it is now the other way around.

Here is an example of the TWR frequency recorded with the latest version: 
[twr-after-cs16fix.zip](https://github.com/szpajder/RTLSDR-Airband/files/6836273/twr-after-cs16fix.zip)

szpajder, 2021-07-18:
The program works with CF32 internally. The convention for this format is
that the sample value shall fit in -1.0 to 1.0 range, however rtl_airband
happily works with larger ranges and this is why you get clean MP3 audio.
But when the I/Q data is about to be exported to a rawfile, it gets
converted to CS16 in a simplistic way, ie. a float range -1.0 - +1.0 gets
remapped to -1024 - +1024. However, given that rtl_airband allows larger
float span internally, it may happen that the resulting CS16 samples will
exceed the -1024 - +1024 range. When the signal is very strong, it may even
happen that the CS16 value exceeds the maximum value allowed by the format
which is 32768. Then the signal gets clipped and this is your problem - you
get garbled audio.

I might fix this by changing the sample format in rawfiles to CF32, if you
don't mind. I checked with csdr and it does not seem to care about samples
exceeding -1.0 to +1.0 range. Your demodulation flow will still work, you
just need to omit the convert_s16_f step. Other apps might be picky about
it though - for example, Audacity gives me a clipped signal when I import a
CF32 file with samples exceeding the -1.0-1.0 range.

I still wonder why there is such a large difference between signal strength
on these two receivers / channels. Is this only due to the distance to the
transmitters?

thowe-switzerland, 2021-07-18:
Once again thank you for the explanations!

For the format, I am bound to the CS16 requested by the research project. I do not need this for myself. I only supply that for the automatic speech recognition training.

I will try to investigate the problem with the different signal strengths. E.g. by exchanging the channels of the receivers. Then we will see if the transmitter or my receiver chain is different.

Personally, I can't imagine that the two transmitters are so different. The antenna is 4km away from me. The two frequencies are IMHO about the same strength.... But I will find out.

szpajder, 2021-07-19:
Quick check: run the program in foreground (with -f option) and inspect
textual waterfalls. What signal levels and noise levels does it show on
both receivers in typical conditions?

thowe-switzerland, 2021-07-20:
All channels inactive:
[image]

ATIS (125.125) on 2nd RSP1A active:
[image]

TWR (121.025) and APP (127.325) active (operated coupled by one ATC as low traffic in the evening time)
[image]
[image]

thowe-switzerland, 2021-07-20:
I had an exchange with a representative of the research project regarding the problem with the clipped CS16 files. He gave me the feedback that they are not bound to this particular CS16 format and that others are also possible. I.e. if there is a better suitable format, then this would also fit. You mentioned CF32 as an example.

Regardless of this, the question arises as to why the signal strengths differ so much between the two RSP1As. I will analyze this further. At the moment I can not make any physical tests or changes, because I am remote and can only access the installation via command line shell.

szpajder, 2021-07-20:
You have 0dBFS signal level at 121.025 which means the receiver is
saturated and that's why I/Q data is now clipped.

That preamp is doing you more harm than good.

thowe-switzerland, 2021-07-20:
Yes, I saw that too and will increase the attenuation in the RSP1A a bit. I know that I drive a little "too hot" there. Fortunately, you can hardly hear it in the MP3.

With the LNA I want to compensate actually only the halving of the signal because of the antenna splitter. Maybe I will power the LNA directly and insert an RF attenuator. Because with attenuator after the LNA, bias-T powering doesn't work anymore.

And I am of course deliberately looking to optimize the (internal/external) LNA gain to the limit before overdrive. The signal-to-noise ratio gets much better - stations become audible that would otherwise be inaudible. But of course the total gain over all should not be too high. I can easily correct this by attenuating the internal LNA of the RSP1A one step more. The RSPA1A is very well controllable. Basically, it doesn't matter whether the external LNA or the internal LNA provides the gain. About two gain reduction steps further down, the RSP1A also clips alone without the external LNA. 

But yes: I'm going a bit over the top on the first one, and need to correct that right away. Either way, the total gain over all should not be so high as to overdrive the intermediate stage or ADC. I'll increase the RFGR (Radio Frequency Gain Reduction) of the first RSP1A, which reduces the overall gain before the intermediate stage by about 7db.

And a completely different question is of course, why the second one gets fed such a completely different level. I will investigate this question.

charlie-foxtrot, 2021-07-20:
From earlier you mentioned

> Actually TWR ATC and APP ATC should be comparably strong. Quite astonishing. While before TWR was ok and APP was not, it is now the other way around.

but the two frequencies having comparable strengths may not be accurate.  The controller operating both frequencies may be sitting in the same place, but that does not mean their signal is transmitting from the same location(s) and/or at the same power.  Some airports broadcast their tower frequency from an antenna on the tower, but broadcast their approach frequency from a nearby hilltop.  Furthermore, approach may need to cover a larger area so be transmitted at a higher power.

szpajder, 2021-07-20:
> Yes, I saw that too and will increase the attenuation in the RSP1A a bit. I know that I drive a little "too hot" there. Fortunately, you can hardly hear it in the MP3.

This is because rtl_airband can cope with floating point samples exceeding 1.0, but they get clipped when exported to a rawfile.

>And a completely different question is of course, why the second one gets fed such a completely different level. I will investigate this question.

SDRPlay radios have a bandpass antialiasing filter at the front end. According to the API documentation, this filter supports following bandwidth settings: 0.2, 0.3, 0.6, 1.536, 5.0, 6.0, 7.0 and 8.0 MHz. Your choice of a sampling rate (3 Msps) is a bit unfortunate because it causes the filter to be set to 1.536 MHz which means your spectrum looks like this:

[image]

This shows that your channel of interest is outside of the filter passband and is significantly attenuated. This problem is not present on the first device because the channels on it are much closer to the center frequency, hence they both fit in the passband.

To fix the issue you should increase the sampling rate on the second device up to at least 5 Msps. This will cause the 5.0 MHz filter to kick in and the result should look like this:

[image]

Probably you need to crank up `fft_size` as well to narrow down the channel bins (otherwise you will pick more noise and the sensitivity will drop). Obviously this will affect the bin size on the first device too (because `fft_size` setting is global), so if this will have any adverse effects (like distorted audio) you need to set the first device to 5 Msps as well, so that the bin widths match.

szpajder, 2021-07-20:
I have changed the sample format in rawfile outputs to CF32. Please check
the latest unstable branch and adjust your csdr processing flow accordingly
(just drop the initial convert_s16_f step).

thowe-switzerland, 2021-07-20:
Thank you so much for this incredible support I am receiving here. I am deeply impressed and learning a lot.

I was not aware that at 3 MSPS the narrower filter of the RSP1A provides attenuation of the edges. Of course, this explains why the two more widely spaced frequencies on the second RSP1A experience an overall signal attenuation of about 14dB compared to the signals on the first. This mystery is thus solved. Cool. :-)

Background: Originally I had wanted to build this new receive setup (which fortunately is still under construction and not productive, so I can experiment well) first on a single RSPA1A receiving all 4 frequencies at once. But in doing so I ran into two problems:
1. buffer overflows occurred quite often, which became less frequent the smaller the MSPS was
2. with increasing MSPS the RSP1A changes gradually from a maximum of 14 bit dynamic range over 12 bit, 10 bit down to 8 bit. So with 7 or 8 MSPS, I had less dynamic range.

Since I receive very close to the transmitters and in the approach path (details below), I have to fight with extreme level differences. Especially the 8-bit RTL SDR dongles in the current feed suffer from this, which leads to the fact that you have to decide when setting the gain whether you want to listen to the close stations without clipping (which I currently have chosen) or the weak far away ones with less noise. To solve both problems (buffer overflows, dynamic range) I decided to use two RSP1A.

What I wonder: in the edges, both signal and noise are lowered by the filter. Am I losing a lot of signal to noise ratio by using the current suboptimal config? So much that it's worth taking the dynamic range loss from 14 bits to 12 bits? I'll give this a practical try. Since the two devices have the identical signal, I can have both configurations receive in parallel and compare the results 1:1. 

About the transmitter location: Thanks for the hint @charlie-foxtrot. This is effectively true for our airport as well. The transmitting and receiving antennas are clearly separated and stationed on different terrain elevations. I know these very well.

- Airport: 510 m.a.s.l.
- Transmitting antenna: 590 m.a.s.l. 1km east of the airport
- Receiving antenna: 800 m.a.s.l. 3.5km south of the airport
- Receiving antenna of my feeder 3.5km east of the transmitting antenna at 560 m.a.s.l.

Fortunately there are small elevations between the transmitting antenna and my receiving antenna. Otherwise the problem with the extreme dynamic range would be even bigger.

By the way, this is how the setup with the two RSPA1 and the RPi4 looks like. Everything in a metal case, which protects the electronics under the roof and dissipates the heat.  Blue the thermal pads, green the temperature and humidity logger.

[image]

And thanks for the upgrade to CF32 RAW format. I hope to test that tomorrow.

thowe-switzerland, 2021-07-20:
Looked up the resolution in relation to MSPS again. 5.0 MSPS should also provide 14 bit resolution. I.e. except more buffer overflows thus no disadvantage. Great.

szpajder, 2021-07-20:
> What I wonder: in the edges, both signal and noise are lowered by the filter. Am I losing a lot of signal to noise ratio by using the current suboptimal config?

Indeed, both signal and noise are lowered by the filter by the same amount, however the quantization noise level stays the same, which means your SNR drops down slightly. But unless you go down too much with the gain (or rather: go up too much with the gain reduction), then you should still be fine. Just make sure that the noise floor level on weak channels is a couple of dB (say, 10 dB) higher than the noise floor level on these channels with the antenna disconnected.

If increasing the sample rate is not feasible due to computational overhead, then you may:

1. Decrease gain reduction on the second receiver to compensate for the loss introduced by the antialiasing filter. This should work reasonably well, provided that you don't have any strong blockers in the passband.
2. Put the LNA in between the splitter and the second RSP1A - as another way of compensating that loss. Having an LNA in front of the first RSP1A is probably not critical, since it's receiving Tower and Clearance Delivery, which, by definition, covers a smaller area than Approach (so you aren't doing any sort of "DX'ing" here).
3. Just do nothing - if you are happy with the audio quality then why fix something which is not broken? With CF32 rawfiles, you should be able to pull equally good audio out of these weak channels too.

thowe-switzerland, 2021-07-21:
Despite the vacation schedule, I got around to installing the new version with CF32 RAW today and doing tests with 5 MSPS vs. 3 MSPS.

I configured one RSP1A with 5 MSPS and one with 3 MSPS to receive APP and ATIS, optimized the gains and then compared. There is a razor thin advantage for the 5 MSPS. But almost non-existent and only noticeable with extremely weak signals. In practice, the APP channel sounds identical.

So I will stick to the the suggested variant 1 and raise the gain of the second RSP1A by reducing RFGR by two steps to compensate the narrow filter.

The resulting CF32 files now really allow the production of high quality FLAC files. I'm amazed at the detail you can hear from the cockpits. As a quick example, I've attached a takeoff clearance for a rescue helicopter. I am very happy with it!
[twr-example-from-cf32-1.zip](https://github.com/szpajder/RTLSDR-Airband/files/6858840/twr-example-from-cf32-1.zip)

Other observation: During testing I often used "rtl_airband -f" without further optimization. Started like this, completely without task scheduler optimization for realtime and without "nice", the Icecast stream stalls about once per two minutes for about 5 seconds and then continues where it stalled. This does not occur if I set the round robin task sheduler (rr) with high prio for rtl_airband. But that is another topic for which I may open a discussion.

Note: In log outputs sometimes the format "CS16" is still mentioned. Otherwise: The CF32 are an improvement from my point of view. Thanks!

szpajder, 2021-07-21:
> Note: In log outputs sometimes the format "CS16" is still mentioned.
>
Perhaps you see this in a log message informing about the chosen input
sample format? In this case this is expected - CS16 is the default format
of samples produced by SDRPlay radios.

> Otherwise: The CF32 are an improvement from my point of view. Thanks!
>
Glad to know it helped. Thanks for the feedback and testing.

If you are satisfied with the fix, then please close the issue.

closed 2021-07-21 by thowe-switzerland

fabland, 2021-09-28:
Just a small thing, I am using an airspy with soapysdr and I get the following output:
sudo rtl_airband -f
[INFO] Using format CS16.
The raw output format that is produced is .cf32

szpajder, 2021-09-28:
The [INFO] message is produced by the SoapySDR library. It indicates the
format of samples received from the device. The raw output format is always
CF32, regardless of what the input device supports.