rtl-airband/RTLSDR-Airband #264: [BUG] Mixer seems to play sample section multiple times (Unstable branch)

thowe-switzerland, 2021-07-30:
**Describe your environment**

**RTLSDR-Airband version:** current unstable branch from git, a472673
make options used to build the program: cmake -DPLATFORM=native -DSOAPYSDR=1 ../

**Operating system name and version:** Raspberry Pi OS, Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux

**Hardware:**
Raspberry Pi 4B 4GB RAM (new, tried with original and alternative strong enough power supply)
2x SDRPlay RSP1A

**Other Software:**
SoapySDR 0.8.0
SoapySDRPlay 0.3.0
SDRplay_RSP_API-ARM32-3.07.2

**What happened?**
Sporadically (about 1 to 4 per hour) the mixers seem to play a very short section of their inputs (section of maybe 0.2s length) several times (about two to four times). This results in a short stutter effect.

Example:
[example-repeated-section.zip](https://github.com/szpajder/RTLSDR-Airband/files/6909706/example-repeated-section.zip)

I was able to find out the following through experimentation:
- The repetition affects all input channels of a mixer simultaneously.
- If input channels lead into several mixers, all mixers are affected at the same time. 
- All outputs of a mixer show the problem at the same time. I.e. both the Icecast stream and the MP3 recording of the mixer stutter at the same time.

**What you expected to happen?**
I would expect mixers not to play sample fragments multiple times.

**Steps to Reproduce**
I am unsure what is contributing to the problem and can only describe my setup which is causing the problem for me.

I can't rule out that problems further up the chain are leading to this problem at the back of the mixer. E.g. buffer overflows in the input section. But I am very unsure where the problem comes from, since the direct MP3 of a channel (without mixer) are without this effect. But these direct recordings of a channel sporadically (at other times) show "gaps". However, this should only lead to gaps in the mixer and not to repeated sample fragments. 

**Additional context**
Add any other relevant information about the problem here.

**Your rtl_airband.conf file**
```
fft_size = 1024;

devices:
(
 
 { 
  type = "soapysdr";
  device_string = "driver=sdrplay,serial=200*******,biasT_ctrl=true"
  gain = "RFGR=7,IFGR=42";

  sample_rate = 3.0;
  centerfreq = 121.500;

  channels:
  (
    {
      freq = 121.025;
      outputs: (
        {
           type = "file";
           directory = "/home/pi/rtlsdr_airband_data/twr_mp3/";
           filename_template = "TWR";
           continuous = true;
           append = true;
        },
        {
           type = "rawfile";
           directory = "/home/pi/rtlsdr_airband_data/atco2";
           filename_template = "Tower";
           split_on_transmission = true;
        },
        {
           type = "mixer";
           name = "mixer";
           balance = -0.8;
           ampfactor = 1.0;
        }
      );
    },
    {
      freq = 121.900;
      outputs: (
        {
           type = "file";
           directory = "/home/pi/rtlsdr_airband_data/cld_mp3/";
           filename_template = "CLD";
	       continuous = true;
           append = true;
        }
      );
    }
  );
 },

 {
  type = "soapysdr";

  device_string = "driver=sdrplay,serial=200******,biasT_ctrl=false"
  gain = "RFGR=5,IFGR=40";

  sample_rate = 3.0;
  centerfreq = 126.150;

  channels:
  (
    {
      freq = 127.325;
      outputs: (
        {
           type = "file";
           directory = "/home/pi/rtlsdr_airband_data/arr_dep_mp3/";
           filename_template = "ARR_DEP";
    	   continuous = true;
           append = true;
        },
        {
           type = "rawfile";
           directory = "/home/pi/rtlsdr_airband_data/atco2";
           filename_template = "Approach";
           split_on_transmission = true;
        },
        {
           type = "mixer";
           name = "mixer";
           balance = 0.8;
           ampfactor = 1.0;
        }
      );
    },
    {
      freq = 125.125;
      outputs: (
        {
           type = "file";
           directory = "/home/pi/rtlsdr_airband_data/atis_mp3/";
           filename_template = "ATIS";
           continuous = true;
           append = true;
        },
        {
           disable=true;
           type = "mixer";
           name = "scanner";
           balance = 1.0;
           ampfactor = 1.0;
        }
      )
    }
  )
 }
);

mixers: {
  mixer: {
    outputs: (
         {
          type = "file";
          directory = "/home/pi/rtlsdr_airband_data/twr_arr_dep_mp3/";
          filename_template = "Stereo_L_Twr_R_Arr_Dep";
          continuous = true;
          append = true;
        },
        {
          type = "icecast";
          server = "aaa.bbb.ccc.ddd";
          port = 1234;
          mountpoint = "stream";
          username = "*******";
          password = "******";
        },
    );
  }
};

```

thowe-switzerland, 2021-07-30:
I must correct my statement that the "gaps" in the input do not correlate with the repetitions in the mixer. Apparently they do.

I listened to the direct recording of the unmixed input channel again well. And there is something missing: The pilot probably said "Swiss Ambulance" (which would be the beginning of his call sign) , recorded is only "S....... Ambulance". I suspect that the missing part ("wiss") corresponds to the repeated part in the mixer.

Here is the recording of the input channel with the (rather hard to recognize) gap.
[example-gap-section.zip](https://github.com/szpajder/RTLSDR-Airband/files/6909786/example-gap-section.zip)

charlie-foxtrot, 2021-07-30:
The mixer(s) always run in their own thread.  If you are not using `multiple_demod_threads` or `multiple_output_threads` then all of your demodulation and per-channel outputs will happen together in the same thread.

Sounds like a race condition between the demodulation / output thread and the mixer thread.

charlie-foxtrot, 2021-07-30:
@thowe-switzerland if you enable the stats file im curious if you're getting any overrun counts when this happens.

Audio is processed in 1/8 of a second batches and your file does look to have the same 0.125 second portion repeated  4 times.

FWIW the mixer thread has a mutex around the critical parts so nothing obviously wrong

thowe-switzerland, 2021-07-30:
@charlie-foxtrot 

Thanks for the hint and sorry for my error: The conf file in the description did not reflect the last change I made after our discussion. When this problem occurred today, the following global options were effketively set:
```
multiple_demod_threads = true;
multiple_output_threads = true;
```
A side note: I don't see any difference in the utilization of the cores if these options are set or not. But that may have to be the case.

thowe-switzerland, 2021-07-30:
**rtlsdr_airband_stats.txt** (no restart since the problem)
```
# HELP channel_activity_counter Loops of output_thread with frequency active.
# TYPE channel_activity_counter counter
channel_activity_counter{freq="121.025"}	8152
channel_activity_counter{freq="121.900"}	71
channel_activity_counter{freq="127.325"}	8182

# HELP channel_noise_level Squelch noise_level.
# TYPE channel_noise_level gauge
channel_noise_level{freq="121.025"}	0.033
channel_noise_level{freq="121.900"}	0.031
channel_noise_level{freq="127.325"}	0.024

# HELP channel_signal_level Squelch signal_level.
# TYPE channel_signal_level gauge
channel_signal_level{freq="121.025"}	0.038
channel_signal_level{freq="121.900"}	0.031
channel_signal_level{freq="127.325"}	0.025

# HELP channel_squelch_counter Squelch open_count.
# TYPE channel_squelch_counter counter
channel_squelch_counter{freq="121.025"}	338
channel_squelch_counter{freq="121.900"}	51
channel_squelch_counter{freq="127.325"}	276

# HELP channel_flappy_counter Squelch flappy_count.
# TYPE channel_flappy_counter counter
channel_flappy_counter{freq="121.025"}	66
channel_flappy_counter{freq="121.900"}	4
channel_flappy_counter{freq="127.325"}	8

# HELP buffer_overflow_count Number of times a device's buffer has overflowed.
# TYPE buffer_overflow_count counter
buffer_overflow_count{device="0"}	0
buffer_overflow_count{device="1"}	0

# HELP output_overrun_count Number of times a device or mixer output has overrun.
# TYPE output_overrun_count counter
output_overrun_count{device="0"}	46
output_overrun_count{device="1"}	25
output_overrun_count{mixer="0"}	49
output_overrun_count{mixer="1"}	53

# HELP input_overrun_count Number of times mixer input has overrun.
# TYPE input_overrun_count counter
input_overrun_count{mixer="0",input="0"}	10
input_overrun_count{mixer="0",input="1"}	9
input_overrun_count{mixer="1",input="0"}	11
input_overrun_count{mixer="1",input="1"}	11
input_overrun_count{mixer="1",input="2"}	10
```

thowe-switzerland, 2021-07-30:
**Additional information**

CPU temperature (40 degrees --> CPU should run at normal speed)
[image]

htop
[image]

top
[image]

charlie-foxtrot, 2021-07-31:
See if you can correlate when the issue occurs to an increase in one or more overflow counters.

I use Prometheus and Grafana to plot the metrics over time

thowe-switzerland, 2021-08-01:
The large number of overruns of the channels let's me suspect that writing to the SD Card (Samsung 16GB, class 10) might be a bottleneck.

I have now set up a 1GB RAM disk (tmpfs) and write the outputs to it. After about 16h of operation with RAM disk instead of SD card, hardly any buffer overruns were counted this way. Also I have the impression that the streams run evenly without repeaters or gaps.

Currently I would assume that writing to files on the SD card plays a significant role in the problem. That the problem is rather not CPU-related but IO-related.

```
# HELP channel_activity_counter Loops of output_thread with frequency active.
# TYPE channel_activity_counter counter
channel_activity_counter{freq="121.025"}	14275
channel_activity_counter{freq="121.900"}	342
channel_activity_counter{freq="127.325"}	14367
channel_activity_counter{freq="125.125"}	381118

# HELP channel_noise_level Squelch noise_level.
# TYPE channel_noise_level gauge
channel_noise_level{freq="121.025"}	0.050
channel_noise_level{freq="121.900"}	0.046
channel_noise_level{freq="127.325"}	0.027
channel_noise_level{freq="125.125"}	0.047

# HELP channel_signal_level Squelch signal_level.
# TYPE channel_signal_level gauge
channel_signal_level{freq="121.025"}	0.055
channel_signal_level{freq="121.900"}	0.047
channel_signal_level{freq="127.325"}	0.030
channel_signal_level{freq="125.125"}	25.071

# HELP channel_squelch_counter Squelch open_count.
# TYPE channel_squelch_counter counter
channel_squelch_counter{freq="121.025"}	450
channel_squelch_counter{freq="121.900"}	31
channel_squelch_counter{freq="127.325"}	429
channel_squelch_counter{freq="125.125"}	2251

# HELP channel_flappy_counter Squelch flappy_count.
# TYPE channel_flappy_counter counter
channel_flappy_counter{freq="121.025"}	33
channel_flappy_counter{freq="121.900"}	12
channel_flappy_counter{freq="127.325"}	1
channel_flappy_counter{freq="125.125"}	0

# HELP buffer_overflow_count Number of times a device's buffer has overflowed.
# TYPE buffer_overflow_count counter
buffer_overflow_count{device="0"}	0
buffer_overflow_count{device="1"}	0

# HELP output_overrun_count Number of times a device or mixer output has overrun.
# TYPE output_overrun_count counter
output_overrun_count{device="0"}	0
output_overrun_count{device="1"}	0
output_overrun_count{mixer="0"}	0
output_overrun_count{mixer="1"}	0

# HELP input_overrun_count Number of times mixer input has overrun.
# TYPE input_overrun_count counter
input_overrun_count{mixer="0",input="0"}	1
input_overrun_count{mixer="0",input="1"}	0
input_overrun_count{mixer="1",input="0"}	1
input_overrun_count{mixer="1",input="1"}	1
input_overrun_count{mixer="1",input="2"}	0
```

charlie-foxtrot, 2021-08-02:
Interesting…. I also have all my files being written to tempfs (including the stats file and logs). My reasoning wasn’t performance but to keep from thrashing the SD card

thowe-switzerland, 2021-08-02:
I was also concerned about the rapid wear of the SD card and plan to switch to an external SSD soon anyway.

In the last 24h the overrun counters remained completely unchanged. So at least my SD Card is causing IO problems, which also affect Icecast streams (gaps, repeats). Since the switch to ramdisk all problems seem to be resolved...

I will try tests with SD Cards from another manufacturer in a few days. Currently I have only remote access.

thowe-switzerland, 2021-08-11:
Status Update:

I have noticed the overruns when outputting to files on the SD card with two different SD cards of the same type:
SanDisk Ultra 16GB, Class 10, UHS-I
[image]

Although this is a solid SD card, it does not perform optimally [in the comparison](https://www.jeffgeerling.com/blog/2019/raspberry-pi-microsd-card-performance-comparison-2019), especially with random 4K I/O. However, a minimum of 1.5 MB/s should always be possible, which should be sufficient.

Since I now write the output files to a ramdisk, I no longer have any problems with timing.  No overruns. My problem is therefore solved.

The other question is, if there is an easy way to decouple the outputs, so that in case of timing problems in one channel (e.g. slow file on SD card) another channel (e.g. Icecast) would not be affected as well.

It is up to you if you want to close the ticket.

Thank you very much for your help.

closed 2021-08-21 by szpajder