rtl-airband/RTLSDR-Airband #269: [FEATURE] Append tag or frequency to filename when using scan mode

steve-horvath, 2021-08-12:
First, I love the software.  This is a great application for an RTL dongle.

**Is your feature request related to a problem? If so, please describe.**
When using rtl_airband with rdio-scanner, it would be nice to be able to separate the various single channel frequencies when using scan mode.
Currently, when using scan mode, all files recorded have a date and time stamp, but it would be ice to be able to filter those files by frequency (for example, isolate which files relate to my local airport tower, or ground frequencies).

**Describe the solution you'd like**
Appending the frequency, or an alpha tag, to the file, based on the frequency tuned would allow the files to be filtered to the service.  the frequency is logged in the syslog, such as:
Aug 12 11:47:18 trunk-recorder rtl_airband[25242]: Activity on 414.712 MHz 
When the file is recorded, so the information is present.

**Describe alternative solutions or features you've considered**
having an external process watch the log, parse out the frequency, and rename the file could be an option, but seems to be complicated.

**Additional context**
Add any other relevant information about the feature request here.

LoganSound, 2021-08-12:
Oh I second, this would be pretty handy.

charlie-foxtrot, 2021-09-03:
To make sure I follow, you're running in scan mode with creation of a new file per transmission and asking for the current frequency to be added to the name of the output file?

Should be trivial . . . can you provide the config you're using and a few example current filename -> requested filename?

steve-horvath, 2021-09-03:
as a hacky workaround, I have built up a python script that watches the directory, then parses the syslog to see the "Activity on" event.

for example, if I have a recording:
Log:  Activity on 150.695 MHz
Filename:  CONV_20210903_100443.mp3

I then use my script to rename/move it like:
CONV_20210903_100443_150.695_104.mp3

Where the date/time is the same, but I append the frequency, and a tag.  In Rdio Scanner, I have configured that 104 tag as a Talkgroup ID, so that the conventional scan system is appearing like a trunk system, with each frequency looking like a talkgroup.  In the Rdio Scanner web interface/app, I can then search for traffic on 150.695 by selecting Talkgroup ID 104, and I can include/exclude it in my scan list, so I can still record it, and optionally listen to it real-time.

My rtl_airband config is:
```
stats_filepath = "/tmp/rtl_airband_stats.txt";
lowpass=true;
localtime=true;
log_scan_activity = true;
devices:
({
  type = "rtlsdr";
  serial = "77771111153705700";
  gain = 35;
  correction = 0;
  mode = "scan";
  channels:
  (
    {
      freqs = ( 414.7125, 149.44, 150.575, 150.695, 148.885, 419.7625, 152.345, 154.07, 156.8, 154.205, 154.16, 414.0375, 160.662, 161.025, 160.544, 160.215, 155.7, 156.275, 160.875, 157.2, 161.8);
      modulation = "nfm";
      afc=1;
      outputs: (
        {
        type = "file";
        directory = "/home/shorvath/trunk_recordings/conv_raw/";
        filename_template = "CONV";
        include_freq = true;
        split_on_transmission = true;
        }
      );
    }
  );
 }
);
```

charlie-foxtrot, 2021-09-03:
How important is the tag (ie `_104`) in addition to the frequency?

Adding a flag that only takes effect in `scan` mode with `split_on_transmission = true` that appends the frequency in Hz to the filename would give you `CONV_20210903_100443_150695000.mp3`, does that work?

Another more flexible (but also slightly more involved) option would be to support `filename_template` as a list the same length as `freqs` when in `scan` mode with `split_on_transmission = true`.  This would let you set the template to be something like `CONV_150.695_104` and your output filename would be `CONV_150695000_104_20210903_100443.mp3`.

@szpajder any preference from your side?

szpajder, 2021-09-03:
I prefer the first option (ie. without tags)

Mapping frequencies to talk groups or the like - this can be easily done
externally.

charlie-foxtrot, 2021-09-05:
@steve-horvath can you test #273 in scan mode and see if that works for you?

steve-horvath, 2021-09-30:
> @steve-horvath can you test #273 in scan mode and see if that works for you?

This looks good.

Sorry it took so long to test this out, this works perfectly.  I am able to organize my conventional signals based on frequency when scanning multiple sources.