rtl-airband/RTLSDR-Airband #270: [BUG] TAGS not Visible when streaming with ICECAST2

tet024, 2021-08-13:
Hello, 

I have an issue that the tags aren't visible any longer when streaming with ICECAST. This was working before but suddenly they disappeared. 

      outputs: (

          # For local IceCast server
        {
          disable = false;    
          type = "icecast";
          server = "127.0.0.1";
          port = 8000;
          mountpoint = "stream.mp3";
          name = "Airport_MLA";
          genre = "ATC";
          username = "source";
          password = "xxxxxxxx";
          send_scan_freq_tags = true;
          description = "MLA Aggregated Feed";

any ideas on how to trouble shoot this issue please? 

Thanks, 
Et

szpajder, 2021-08-13:
Which rtl_airband version and libshout version are you using?

tet024, 2021-08-15:
RTLSDR-Airband version v3.2.1-2-gb4335d1
Icecast 2.4.4

szpajder, 2021-08-15:
Icecast server version is irrelevant. Please check the version of libshout
library on the machine where you run rtl_airband.

See also https://github.com/szpajder/RTLSDR-Airband/issues/203

tet024, 2021-08-16:
Good Morning, 

libshout - 2.4.3-1

Thanks, 
Et

szpajder, 2021-08-16:
In this case it's a duplicate of https://github.com/szpajder/RTLSDR-Airband/issues/203

closed 2021-08-16 by szpajder