rtl-airband/RTLSDR-Airband #274: [BUG] Compile Errors on RPi3 with Ubuntu 20.04

steve1515, 2021-09-13:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit): RTLSDR-Airband-3.2.1 Stable Release
  - `make` options used to build the program: $ make PLATFORM=rpiv2 NFM=1
  - Hardware platform (eg. x86_64, Raspberry Pi v4): Raspberry Pi 3 Model B
  - Operating system name and version: Ubuntu 20.04

**What happened?**
When compiling the 2 errors below occur.
```
ubuntu@ubuntu:~/rtlsdr-airband/RTLSDR-Airband-3.2.1$ make PLATFORM=rpiv2 NFM=1
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
g++: error: unrecognized command line option ‘-mfpu=neon-vfpv4’
g++: error: unrecognized command line option ‘-mfloat-abi=hard’
make: *** [<builtin>: rtl_airband.o] Error 1
```

**What you expected to happen?**
Compile without errors.

**Steps to Reproduce**
```
$ sudo apt install libmp3lame-dev libshout3-dev libconfig++-dev
$ sudo apt install libraspberrypi-dev
$ sudo apt install libfftw3-dev
$ sudo apt install librtlsdr-dev rtl-sdr
$ mkdir rtlsdr-airband
$ cd rtlsdr-airband/
$ wget https://github.com/szpajder/RTLSDR-Airband/archive/refs/tags/v3.2.1.tar.gz
$ tar xvzf v3.2.1.tar.gz
$ cd RTLSDR-Airband-3.2.1/
$ make PLATFORM=rpiv2 NFM=1

g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
g++: error: unrecognized command line option ‘-mfpu=neon-vfpv4’
g++: error: unrecognized command line option ‘-mfloat-abi=hard’
make: *** [<builtin>: rtl_airband.o] Error 1
```

steve1515, 2021-09-13:
After doing some research, the issue here seems to be that I have a 64-bit OS. I was able to use the `PLATFORM=armv8-generic` options to compile successfully.

A question remains though...
 `make help` seems to indicate that compiling this way will use main CPU for FFT as opposed to the `PLATFORM=rpiv2` option that indicates that the BCM VideoCore will be used for FFT. Is there a way to still use the VideoCore in 64-bit architectures? Will there be a performance penalty?

dlpwx, 2021-09-13:
I can't answer your question as to a workaround to enable use of VideoCore after compiling, though I'm curious to see the answer. Because my 32-bit RPI 4 also needed to be compiled with armv8-generic. However, I can say there are limitations in what the CPU can do. It just depends on your use case and rtl_airband config. e.g. fft_size, sample_rate, channels and devices. I've been CPU limited in a few things I wanted to try. These CPU limits will apply even more so to the RPI 3. That said, for certain uses, you can enable multithreaded flags. See the wiki for when both of those can be used.

multiple_demod_threads = true;
multiple_output_threads = true;

szpajder, 2021-09-15:
@steve1515 `-mfpu=neon-vfpv4` and `-mfloat-abi=hard` options are not available for aarch64 architecture because these features are always enabled on that arch. Just remove these options from the makefile and try again.

@dlpwx VideoCore in RPi4 is different than in earlier RPi generations, hence it is not supported in rtl_airband. `armv8-generic` is the only way to go on this platform.

steve1515, 2021-09-17:
@szpajder When I remove those two options, I get the following error:
```
cc1plus: error: unknown value ‘armv7-a’ for ‘-march’
cc1plus: note: valid arguments are: armv8-a armv8.1-a armv8.2-a armv8.3-a armv8.4-a armv8.5-a native; did you mean ‘armv8-a’?
```
If I then change the `-march` to either `-march=armv8-a+crc` or `-march=armv8-a`, I get the following error:
```
rtl_airband.cpp:22:2: error: #error Broadcom VideoCore support can only be enabled on ARM builds
   22 | #error Broadcom VideoCore support can only be enabled on ARM builds
      |  ^~~~~
make: *** [<builtin>: rtl_airband.o] Error 1
```

This might be something that needs to be fixed in the source code. It looks like ``__arm__`` is only defined in 32-bit architectures, so the way the code is written (line 22 of rtl_airband.cpp) will only allow VideoCore with 32-bit systems.

If I simply comment out that check on line 22 and compile with `-march=armv8-a+crc` or `-march=armv8-a`, I then get the following errors:
```
ubuntu@ubuntu:~/rtlsdr-airband/RTLSDR-Airband-3.2.1$ make PLATFORM=rpiv2 NFM=1
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR   -c -o input-common.o input-common.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR   -c -o input-helpers.o input-helpers.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR   -c -o input-file.o input-file.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR   -c -o output.o output.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR   -c -o config.o config.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR   -c -o util.o util.cpp
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR   -c -o mixer.o mixer.cpp
make -C hello_fft
make[1]: Entering directory '/home/ubuntu/rtlsdr-airband/RTLSDR-Airband-3.2.1/hello_fft'
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR -I..   -c -o mailbox.o mailbox.c
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR -I..   -c -o gpu_fft.o gpu_fft.c
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR -I..   -c -o gpu_fft_twiddles.o gpu_fft_twiddles.c
gpu_fft_twiddles.c: In function ‘float* twiddles_step_16(double, float*, double)’:
gpu_fft_twiddles.c:70:39: warning: unused parameter ‘two_pi’ [-Wunused-parameter]
   70 | static float *twiddles_step_16(double two_pi, float *out, double theta) {
      |                                ~~~~~~~^~~~~~
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR -I..   -c -o gpu_fft_shaders.o gpu_fft_shaders.c
g++ -O3 -g -Wall -Wextra -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a -ffast-math  -DNFM -DWITH_RTLSDR -I..   -c -o gpu_fft_base.o gpu_fft_base.c
gpu_fft_base.c: In function ‘unsigned int gpu_fft_base_exec_direct(GPU_FFT_BASE*, int)’:
gpu_fft_base.c:65:16: warning: comparison of integer expressions of different signedness: ‘unsigned int’ and ‘int’ [-Wsign-compare]
   65 |     for (q=0; q<num_qpus; q++) { // Launch shader(s)
      |               ~^~~~~~~~~
gpu_fft_base.c:72:50: warning: comparison of integer expressions of different signedness: ‘unsigned int’ and ‘int’ [-Wsign-compare]
[... cut ...]
rtl_airband_neon.s:77: Error: operand 1 must be an integer register -- `ldrb r12,[r1,#7]'
rtl_airband_neon.s:78: Error: unknown mnemonic `vstmia' -- `vstmia r0!,{q6-q7}'
rtl_airband_neon.s:79: Error: operand 1 must be an integer register -- `subs r4,r4,#1'
rtl_airband_neon.s:82: Error: unknown mnemonic `vpop' -- `vpop {d4-d15}'
rtl_airband_neon.s:83: Error: unknown mnemonic `pop' -- `pop {r4-r12,pc}'
make: *** [<builtin>: rtl_airband_neon.o] Error 1
```

Maybe it's not possible to use the VIdeoCore on 64-bit on a RPi 3?

szpajder, 2021-09-17:
Right, I forgot to mention that the assembly code for DSP operations which
is enabled together with VC support works only on 32-bit arches.

In this case I suggest switching to a 32-bit OS.

steve1515, 2021-09-19:
I just came to report that I've now tried with 32-bit Ubuntu and VC support. I'm now get about 25% CPU usage vs. the 50% I was getting without VC support in the 64-bit Ubuntu.