rtl-airband/RTLSDR-Airband #303: Build on Apple M1 Silicon (with diff to make it build)

sdadh01, 2021-12-20:
Along with the normal brew installs (brew install lame libshout libconfig fftw librtlsdr soapysdr pulseaudio) I needed to make the following changes (shown as a diff below) to get a build to work on Apple silicon:

```
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 3aaff68..9aa5709 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -96,7 +96,11 @@ set(CMAKE_REQUIRED_LIBRARIES ${CMAKE_REQUIRED_LIBRARIES_SAVE})

 option(NFM "Enable support for narrow FM channels" OFF)

-set(PLATFORM "native" CACHE STRING "Optimize the build for the given hardware platform")
+if (PLATFORM)
+  set(PLATFORM "'${PLATFORM}'" CACHE STRING "Optimize the build for the given hardware platform")
+else()
+  set(PLATFORM "native" CACHE STRING "Optimize the build for the given hardware platform")
+endif()

 option(RTLSDR "Enable RTL-SDR support" ON)
 set(WITH_RTLSDR FALSE)
@@ -189,11 +193,13 @@ elseif(PLATFORM STREQUAL "armv7-generic")
        add_definitions(-march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard)
 elseif(PLATFORM STREQUAL "armv8-generic")
        add_definitions(-march=armv8-a+crc -mtune=cortex-a53)
+elseif(PLATFORM STREQUAL "apple-m1")
+       add_definitions(-mcpu=apple-m1 -mtune=apple-m1)
 elseif(PLATFORM STREQUAL "native")
        add_definitions(-march=native -mtune=native)
 else()
        message(FATAL_ERROR "Unknown platform '${PLATFORM}'. Valid options are:
-       rpiv1, rpiv2, armv7-generic, armv8-generic, native")
+       rpiv1, rpiv2, armv7-generic, armv8-generic, apple-m1, native")
 endif()

 # Try using VC GPU if enabled. Fallback to fftw3f if disabled or if VC lib not found
```

With this in place I successfully did a  "cmake -D PLATFORM=apple-m1 ../" in the build directory I created. I noticed that without the bit of logic that set the PLATFORM if it was not already set the result was always 'native'.

Could this (or something close) be added to make the compilation work. I'd do a PR but I didn't know if you accepted them.

Note:  I haven't actually tested the resulting binary other than to do an 'rtl_airband -h' just to make sure it ran - I'll do that next weekend.

szpajder, 2021-12-20:
>
> I noticed that without the bit of logic that set the PLATFORM if it was
> not already set the result was always 'native'.
>
Yes, as expected, because this is the default which is expected to work in
most cases. The compiler shall autodetect the CPU then.

Can you describe what issues you encountered with PLATFORM=native?

sdadh01, 2021-12-20:
Yes, it's really not a native problem more of a MacOS clang issue. The MacOS clang does not take '-march' as an argument but instead '-mcpu'. There may be another way around the issue but the solution I took was to make PLATFORM overridable and use the PLATFORM if/elseif to give the compiler the arguments it expected.

Hope that helps!

szpajder, 2021-12-20:
Can you try the latest unstable branch?

It add a new `PLATFORM` value `default` which uses no special optimization options and is the default now. Additionally, when `-DPLATFORM=native` has been specified, cmake now checks if the compiler really supports `-march=native` and `-mtune=native` and spits out an error if it doesn't.

Please try both variants (just remember to delete `build/CMakeCache.txt` before changing cmake options).

sdadh01, 2021-12-25:
Sorry for the delay - actual day job work got in the way. I did  a clean clone of the unstable branch and tried both variants:

- a 'cmake ../' with no changes to PLATFORM. Compiled perfectly, no problems.
- a 'cmake -DPLATFORM=native ../' cmake check on the compiler works and spits out an error as expected.

All looks good and I have tried it with an RTL-SDR dongle for my local airport and I'm hearing the voices I'd expect to hear!
Thanks for fixing in unstable.

szpajder, 2021-12-26:
Released in v4.0.2

commit 085d62ac (2021-12-20): Change default PLATFORM from `native` to `default`

`default` does not use any special optimizations and therefore
produces a more portable binary. Additionally it allows building
on platforms where `-march=native` or `-mtune=native` option
is not available (notably clang on Apple M1)

Fixes #303
closed 2021-12-26 by szpajder