rtl-airband/RTLSDR-Airband #314: [BUG] File output is not always creating .tmp files when active

KS4EC, 2022-01-25:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit):
  **4.0.2** 
  - `make` options used to build the program:
 **-rpiv2 -DNFM=ON**
  - Hardware platform (eg. x86_64, Raspberry Pi v4):
 **Raspberry Pi 3b**
  - Operating system name and version: 
 **Raspbian GNU/Linux 10 (buster)**

**What happened?**
**When using the File output, files are immediately created with .mp3 not .tmp when they are active**

**What you expected to happen?**
**When using the file output per notes in 4.0.0 .tmp files are supposed to be created when a chan is active.**

**Steps to Reproduce**
**Intermittent, I have not been able to sort out why it works sometimes, but mostly not
only worked 1 time then stopped working.**

**Additional context**
**Add any other relevant information about the problem here.
I am using a ram drive for these files as I am using the file activity to drive a hardware status monitor for the feeds.
This happens whether the status monitor is running or not.**

**Your rtl_airband.conf file**
Remove passwords, server addresses and other private information.
-
devices = (

  { 
    disable = false;
    index = 0; 
    type = "rtlsdr"; 
    gain = 38;
    mode = "multichannel";
    centerfreq = 443.0125;
    correction = 1;
    channels: (
      {
         freq = 442.000;
         disable = false;
         modulation = "nfm";
         highpass = 250; 
         outputs = (
                     {
                      type = "icecast"; 
                      server = "***"; 
                      port = ***; 
                      mountpoint = "***"; 
                      username = "***"; 
                      password = "***"; 
                      name = "442.05 Wellington"; 
                      description = "Wellington Radio Club"; 
                      genre = "HAM";
                     },
                     { 
                      type = "file"; 
                      directory = "/var/tmpab/mp3"; 
                      filename_template = "00"; 
                      split_on_transmission = true; 
                      append = false;
                     }
                   );
      }
              )
    }
)

szpajder, 2022-01-25:
No problem here, it works 100% reliably for me.

How do you verify names of newly created files? I do `watch -n1 ls -l /the/mp3/directory` and I clearly see that when there is transmission, the last file in the list has a `.tmp` suffix which disappears when the transmission ends.

Also, what do you mean by "stopped working"? You start the program, it initially appends the suffix and then it stops appending? Or does it change across subsequent runs (ie. you start it and it either adds the suffix every time and forever or doesn't add it at all)? In the latter case please verify whether you are launching the correct binary. Perhaps you have an old 3.x.x binary laying around and you are starting it inadvertently instead of the 4.0.2 one. The program prints its version number on startup - either to syslog (default) or to standard error (when run with `-e -F` options).

KS4EC, 2022-01-25:
After a lot of typing below, I read your last line::::::: Perhaps you have an old 3.x.x binary laying around and you are starting it inadvertently instead of the 4.0.2 one. The program prints its version number on startup - either to syslog (default) or to standard error (when run with -e -F options).
I did that and you were right, I should learn to read from the bottom to the top!I was calling the rtl_airband in the home/pi/RTLSDR_Airband directory, but I see that the correct version is in the path and when I run it without the ./ it runs the right binary.Looks like I have a bunch of cleanup to do!
I'll leave the info below for your browsing pleasure and please let me knwo if you have a better idea of how to do wha tI am trying to do.
Thanks again, Another id 10 t error solved.
-------------------------------------------------
Original Reply :)
When I am testing I am doing it old school, go to the directory and repeatedly do an ls to see what is in there.
Here is the part of the Python script that checks for .tmp files build a status string to be displayed on the LCDNot that I think it matters but the files are saved in ramdisk mounted at /var/tmpab/mp3 so I dont kill the SD card :) 
---------------------
# get list of .tmp files
file_list = glob.glob(dir_name + '*.tmp')  
# reset chan status to off for all chanschan_status = [" ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " ", " "]chan_display = ""
# update status based on CH# file_name[15:17]for file_name in file_list:     chan_number = (file_name[15:17])     chan_status[int(chan_number)-1] = "*" # update chan status
# build the chan_display  for ch in chan_status:    chan_display = chan_display + ch
# Clear out all .mp3 filesfiles = os.listdir(dir_name)  for file in files:    if file.endswith(".mp3"):    os.remove(os.path.join(dir_name,file))
----------------------
As you can see I am not a pro python scripter! Google is my partner in this. :)
I set the filename_template with a 2 digit number that is the channel number for the status. This is then used as an index for the chan_status array and I put a "*" in there.
 By "stopped working" I mean that I worked for days developing a way to watch the file size to detect changes because I never saw .tmp files, then all of the sudden after a restart of airband it started making tmp files so I started over to detect tmp files, then after running fine I shut down the pi for the night and the next morning it was back to not making tmp files again and I have not been able to get it to generate tmp files since then.In short, when it fails (most of the time) it never creates the tmp, only the mp3's 
there were no other config file changes.
For the sake of testing I reformatted my config with 1 chan and broke out the 1 line I was using into multiple lines.
That is what I posted originally.Here is the single config line that worked then stopped.Note: freq is intentionally set to 442.000 instead of 442.05 so I don't interfere with the repeater with testing.This is a small part of a large config file that has all but this chan disabled.As I said above I am actually testing with the config file I posted for simplicity but the large file formatted like this was the one that was working.-------------------------channels = ({ freq = 442.000; disable = false; modulation = "nfm"; highpass = 250; outputs = ({ type = "icecast"; server = "***"; port = ***; mountpoint = "***"; username = "***"; password = "***"; name = "442.05 Wellington"; description = "Wellington Radio Club"; genre = "HAM";},{ type = "file"; directory = "/var/tmpab/mp3"; filename_template = "01"; split_on_transmission = true; append = false;});},--------------------------
Any Ideas???

KS4EC, 2022-01-25:
@szpajder  You can remove this thread as it is not a bug in the software, only the user :)

closed 2022-01-25 by KS4EC