rtl-airband/RTLSDR-Airband #315: 4.0.2 Mageia package build fails for i586

barjac, 2022-02-01:
We had no problems with 4.0.1 builds on all arches that we build for: x86_64, i586, armv7hl and aarch64.
We now have this issue with 4.0.2. It builds for x86_64 but not i586 (arm not yet tested as the build system stopped on the fail).

In a nutshell there are lots of: error: inlining failed in call to 'always_inline' '__m128 _mm_set_ps(float, float, float, float)': target specific option mismatch

Here is the full build log in question:
[4.0.2_i586_log.txt](https://github.com/szpajder/RTLSDR-Airband/files/7982475/4.0.2_i586_log.txt)

For comparison an x86_64 log is here:
[4.0.2_x86_64_log.txt](https://github.com/szpajder/RTLSDR-Airband/files/7982497/4.0.2_x86_64_log.txt)

I hope that helps, if you need more info please just ask.
Thanks,
Barry

barjac, 2022-02-01:
I have now tested builds again in all arches and the only issues are as above in i586, tested in two different build systems.

charlie-foxtrot, 2022-02-01:
It looks like you're `cmake` command is not setting `PLATFORM`.  The only change between 4.0.1 and 4.0.2 was to the `cmake` file to change the default `PLATFORM`.  Maybe try setting `PLATFORM` to `native`?

https://github.com/szpajder/RTLSDR-Airband/compare/v4.0.1...v4.0.2

Also, looking at your build output, you're setting environment variables `NFM` and `PULSE` but I'm not seeing how those make it into `cmake`, and the summery is showing NFM is disabled:

```
-- RTLSDR-Airband configuration summary:

-- - Build type:		RelWithDebInfo
-- - Operating system:		Linux
-- - SDR drivers:
--   - librtlsdr:		requested: ON, enabled: TRUE
--   - mirisdr:			requested: ON, enabled: FALSE
--   - soapysdr:		requested: ON, enabled: TRUE
-- - Other options:
--   - Platform:		default
--   - Broadcom VideoCore GPU:	FALSE
--   - NFM support:		OFF
--   - PulseAudio:		requested: ON, enabled: TRUE
--   - Profiling:		requested: OFF, enabled: FALSE
--   - Icecast TLS support:	TRUE
-- Configuring done
-- Generating done
```

barjac, 2022-02-02:
Many thanks for your rapid response.
Yes you are correct on all points.
All corrected and building OK for all our arches now, so closing as fixed.
Regards,
Barry

closed 2022-02-02 by barjac