rtl-airband/RTLSDR-Airband #335: Wiki Update - Scan mode

charlie-foxtrot, 2022-10-02:
Scan Mode section of the wiki is missing some of the optional parameters that take list inputs. Here are the missing parameters that support "list input" (ie scan mode):

- `lables`
- `squelch_threshold`
- `squelch_snr_threshold`

It looks like `tau` is not listed and supported in scan mode as well, but only as a single value (ie not different per frequency).

For parameters that take list inputs it would be helpful to show how to disable for some channels in the list.  Here is how to disable on a per-frequency for each (after PR #334):

- `squelch_threshold` - set to `0` to disable (ie use auto squelch) for one frequency
- `squelch_snr_threshold` - set to `-1.0` to use the default value for one frequency
- `notch` - values less than or equal to 0.0 will be ignored, set to `0` to disable for one frequency
- `notch_q` - set to `0.0` to use the default value for one frequency
- `bandwidth` - values less than or equal to 0.0 will be ignored, set to `0` to disable for one frequency

szpajder, 2022-10-03:
Wiki updated. Thanks for pointing this out.

As far as default value for `bandwidth` goes - a value of 0 gives "bandwidth value '0' invalid, ignoring" error (the comparison is less or equal 0 rather than less than 0). A value of 0 for `notch` results in "invalid parameter type".

charlie-foxtrot, 2023-04-23:
Error messages cleaned up in #371

closed 2023-04-23 by charlie-foxtrot