rtl-airband/RTLSDR-Airband #343: [FEATURE] Add dBFS metrics

OA6DXV, 2023-01-07:
Understanding the signal values ​​is a bit confusing and taking advantage of the fact that the formula to convert these values ​​to dBFS is found at the end of **util.cpp**  (level_to_dBFS) , it would be good to take advantage of this and give the option of being able to have these values ​​in the metrics file.
It would be enough to add the following code to **output.cpp** :

- Noise Level
```
static void output_dbfs_noise_levels(FILE *f) {
	fprintf(f, "# HELP dBFS_noise_level Squelch noise_level converted to dBFS.\n"
			"# TYPE dBFS_noise_level gauge\n");

	for (int i = 0; i < device_count; i++) {
		device_t* dev = devices + i;
		for (int j = 0; j < dev->channel_count; j++) {
			channel_t* channel = devices[i].channels + j;
			for (int k = 0; k < channel->freq_count; k++) {
				print_channel_metric(f, "dBFS_noise_level", channel->freqlist[k].frequency, channel->freqlist[k].label);
				fprintf(f, "\t%.3f\n", level_to_dBFS(channel->freqlist[k].squelch.noise_level()));
			}
		}
	}
	fprintf(f, "\n");
}
```

- Signal Level
```
static void output_dbfs_signal_levels(FILE *f) {
	fprintf(f, "# HELP dBFS_signal_level Squelch signal_level converted to dBFS.\n"
			"# TYPE dBFS_signal_level gauge\n");

	for (int i = 0; i < device_count; i++) {
		device_t* dev = devices + i;
		for (int j = 0; j < dev->channel_count; j++) {
			channel_t* channel = devices[i].channels + j;
			for (int k = 0; k < channel->freq_count; k++) {
				print_channel_metric(f, "dBFS_signal_level", channel->freqlist[k].frequency, channel->freqlist[k].label);
				fprintf(f, "\t%.3f\n", level_to_dBFS(channel->freqlist[k].squelch.signal_level()));
			}
		}
	}
	fprintf(f, "\n");
}
```
Here you can see an example of my Dashboard in Grafana using values ​​in dBFS, being able to even calculate the SNR, which with RAW values ​​is very complicated and with dBFS it can be done directly subtracting the signal and noise values:
_https://imgur.com/a/guUNy0M_

charlie-foxtrot, 2023-01-07:
@OA6DXV  Great idea, can you open a pull request to the ‘unstable’ branch with these changes?

longview, 2023-02-12:
This is a great addition, thanks!

I use Prometheus output picked up by telegraf and into influx, getting a SNR query is not immediately obvious so I'll note how I got that working here:
```
from(bucket: "yourbucket")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r["_measurement"] == "prometheus")
  |> filter(fn: (r) => r["_field"] == "channel_dbfs_noise_level" or r["_field"] == "channel_dbfs_signal_level")
  any frequency filters here
  |> group(columns: ["_measurement"])
  |> group(columns: ["freq"])
  |> pivot(rowKey: ["_time"], columnKey: ["_field"], valueColumn: "_value")
  |> map(fn: (r) => ({r with _value: r.channel_dbfs_signal_level - r.channel_dbfs_noise_level}))
  |> drop(columns: ["channel_dbfs_noise_level"])
  |> drop(columns: ["channel_dbfs_signal_level"])
  |> aggregateWindow(every: v.windowPeriod, fn: max, createEmpty: false)
  |> yield(name: "mean")
```

For my Grafana plots I also had to add the drop commands to get rid of the signal and noise figures from the data to clean it up.

jayant4001, 2023-02-17:
Thank you. This is great! I've been manually editing the metrics export function to export the values in dB. Adding this in the main branch would be awesome for the folks using Grafana dashboards.

OA6DXV, 2023-02-17:
Yup, thanks @charlie-foxtrot , now everyone can design grafana dashboards for signal and activity analysis like this one:
[https://i.imgur.com/y8fBlM7.png](url)
-73

charlie-foxtrot, 2023-04-21:
changes landed in #355

closed 2023-04-21 by charlie-foxtrot