rtl-airband/RTLSDR-Airband #348: [BUG] No sound in AM mod.

filejimo, 2023-01-10:
Hi,
   When using scan mode, AM modulation, with a walkie talkie to send the am signal. RTLSDR-Airband has saved the mp3 file. But there is no sound when playing the mp3. If the mode comes to NFM, everything is ok.

**Describe your environment**

  - RTLSDR-Airband 4.0.2
  - Hardware platform (eg. x86_64, Raspberry Pi v4):rtlsdr,arm64 dev board
  - Operating system name and version:  ubuntu 20.04 arm64

conf:
devices: (
  {
    type = "rtlsdr";
	mode="scan";
    index = 0;
    gain = 30;
   
    channels:
    (
      {
        freqs = (119.000);
		modulations = ( "am" );
        outputs: (
          {
            type = "file";
            directory = "/home/fe"
            filename_template="sound";
          }
        );
      }
     
    );
  }
);

hamav8tor, 2023-01-10:
More than likely you may be too close to the receiver and are simply overloading it. That would produce the illusion of no audio. Try getting further away from the receiver.

OA6DXV, 2023-01-10:
Try set gain to 0, maybe you are overloading the SDR

gabeschine, 2025-02-27:
Haven't seen an update on this in over a year. Closing. Please re-open if necessary.

closed 2025-02-27 by gabeschine