rtl-airband/RTLSDR-Airband #370: [FEATURE] Change scan mode to use "virtual devices"

charlie-foxtrot, 2023-04-21:
Currently scan mode is a configuration of a device where there is a single channel with a list of frequencies to scan through.

Disadvantages to this approach includes:
- device-level parameters (ie gain) so can't be changed per frequency
- can't set different outputs per frequency
- two frequencies within the SDR's bandwidth are processed serially instead of in parallel

Another approach is for scan mode to use "virtual devices" where only one device is fed from the physical SDR at a time.  This would allow the full set of configurations for scan mode

rough316, 2023-04-28:
Will you be switching to this mode for scan? Sounds like a lot better way to go.

charlie-foxtrot, 2023-04-28:
Yes but won’t have time to work on it for a while (put it here so I wouldn’t forget)

robo65536, 2023-06-19:
Hello! How about scan mode in "virtual devices"? I would like to test this.

> Yes but won’t have time to work on it for a while (put it here so I wouldn’t forget)