rtl-airband/RTLSDR-Airband #374: should i be setting bandwidth to avoid hearing other frequency's ? if so, what value

israelgoldstein, 2023-04-24:
Hi all, 

first of id like to let you know I've only been using your software for a few weeks, and already loving it,
it works great for my use case currently, I stream to icecast, and also save recording transmission as a mp3 file

the issue I am facing, I've noticed that some channels pick up audio from another frequency.
they are narrowband channels int he 482Mhz range

so I thought I can use "bandwidth" parameter to handle this, 
but it is not resolving my issue, I'm still hearing the transmissions from other channels/frequency 
also worth mentioning, i have setup the proper CTCSS(PL) for that channel

perhaps I don't know what value to give bandwidth ?
I've tried setting it to 2000, the audio wasn't clear
I've set it to 4000 it is clearer  
but still hearing other frequencys/channels 

does anyone have this successfully deployed and working - can perhaps assist me ?

israelgoldstein, 2023-04-25:
@charlie-foxtrot here is my config file (i have many more channels, but i trimmed it down before posting it 

fft_size = 512;
localtime = true;
log_scan_activity = true;
multiple_demod_threads = true;
devices: (
  {
    disable = false;
    type = "rtlsdr";
    serial = "50000001";
    gain = 38.5;
    centerfreq = 477.41;
    channels:
    (
      {
        #disable = true;
        freq = 476.4625;
        lowpass = 3200;
        highpass = 200;
        modulation = "nfm";
        notch = 110.9;
        outputs: (
          {
            #disable = true;
            type = "file";
            directory = "/home/pi/folder1";
            filename_template = "Channel1";
            split_on_transmission = true;
            include_freq = true;
          },
          {
            #disable = true;
            type = "icecast";
            server = "my.server.net";
            port = 5027;
            mountpoint = "My_Mount_01";
            name = "Channel 1";
            genre = "Live feed";
            username = "source";
            password = "MyPassword";
        }
        );
      },
      {
        #disable = true;
        freq = 476.5125;
        lowpass = 3200;
        highpass = 200;
        modulation = "nfm";
        notch = 123.0;
        outputs: (
          {
            #disable = true;
            type = "file";
            directory = "/home/pi/folder2";
            filename_template = "Channel2";
            split_on_transmission = true;
            include_freq = true;
          },
          {
            #disable = true;
            type = "icecast";
            server = "my.server.net";
            port = 5027;
            mountpoint = "My_Mount_02";
            name = "Channel 2";
            genre = "Live feed";
            username = "source";
            password = "MyPassword";;
        }
        );
      }
    );
  },
  {
    #disable = true;
    type = "rtlsdr";
    serial = "30000001";
    #serial = "80000003";
    gain = 39.0;
    centerfreq = 482.600;
    channels:
    (
      {
        disable = true;
        freq = 464.6625;
        lowpass = 3200;
        highpass = 200;
        modulation = "nfm";
        notch = 167.9;
        outputs: (
          {
            disable = true;
            type = "file";
            directory = "/home/pi/folder3";
            filename_template = "Channel3";
            split_on_transmission = true;
            include_freq = true;
          },
          {
            disable = true;
            type = "icecast";
            server = "my.server.net";
            port = 5027;
            mountpoint = "My_Mount_03";
            name = "Channel 3";
            genre = "Live feed";
            username = "source";
            password = "MyPassword";
        }
        );
      },
      {
        #disable = true;
        freq = 482.0187;
        lowpass = 3200;
        highpass = 200;
        bandwidth = 3500;
        modulation = "nfm";
        notch = 136.5;
        outputs: (
          {
            #disable = true;
            type = "file";
            directory = "/home/pi/folder4";
            filename_template = "Channel4";
            split_on_transmission = true;
            include_freq = true;
          },
          {
            #disable = true;
            type = "icecast";
            server = "my.server.net";
            port = 5027;
            mountpoint = "My_Mount_04";
            name = "Channel 4";
            genre = "Live feed";
            username = "source";
            password = "MyPassword";
        }
        );
      }
    );
  }
);

charlie-foxtrot, 2023-04-25:
try setting `fft_size` to 1024 at the top level then add `bandwidth = 5000;` to each channel.

charlie-foxtrot, 2023-04-25:
Also, depending on what SDRs you're using you may need to set `sample_rate = 2.4;`, see [here](https://github.com/charlie-foxtrot/RTLSDR-Airband/wiki/Tweaking-sampling-rate-and-FFT-size) for more information on both

israelgoldstein, 2023-04-25:
your suggested changes seems to have resolved - i set the bandwidth to 5000, and 

can you try to explain to me, what did the changes affect ?

I am using RTL-SDR SDR's

also, some channels give me squelch noise, should i open a separate issue about that ?
and id like to know if the software can handle these weird frequency, 428.21875 i recall old scanners couldn't handle it

charlie-foxtrot, 2023-04-25:
I'm glad that worked for you!

The link does good job of explaining sample rate and FFT.  The short version is the sample rate is how much signal bandwidth you have coming in and the FFT size is how much you slice that bandwidth up.  If the FFT and sample size are too large relative to your target signal then there is a possibility that extra signal can bring in neighboring channels.  Options are to decrease `sample_rate` and/or increase `fft_size`.  Setting the bandwidth further filters down the slice to just the portion with your intended channel.

FFT size must be a power of 2, so im guessing `2014` was a typo of `1024` 😄 

For the RTL-SDR SDR's I *think* the `sample_rate` should be set to `2.4` but I'm not finding a good source for that so may be wrong, but give it a try.

For "squelch noise", yes, please open another issue with a description and an audio file if possible.

And finally, `428.21875` should be fine.

closed 2023-04-25 by charlie-foxtrot