rtl-airband/RTLSDR-Airband #377: [BUG] v4.1.0 Build Fails on RPi 3 Ubuntu 20.04

steve1515, 2023-04-30:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit): `v4.1.0`
  - `make` options used to build the program: `$ cmake -DPLATFORM=rpiv2 -DNFM=ON -DRTLSDR=ON -DMIRISDR=OFF -DSOAPYSDR=OFF -DPULSEAUDIO=ON -DPROFILING=OFF ../`
  - Hardware platform (eg. x86_64, Raspberry Pi v4): `Raspberry Pi 3 Model B`
  - Operating system name and version: `Ubuntu 20.04`

**What happened?**
The build fails with:
```
In file included from /home/ubuntu/RTLSDR-Airband-4.1.0/src/hello_fft/mailbox.c:28:
/home/ubuntu/RTLSDR-Airband-4.1.0/src/hello_fft/../rtl_airband.h:33:10: fatal error: config.h: No such file or directory
   33 | #include "config.h"
```

**What you expected to happen?**
I expect the build to succeed.

**Steps to Reproduce**
Enter the following commands:
```
$ tar xvzf v4.1.0.tar.gz
$ cd RTLSDR-Airband-4.1.0/
$ mkdir build
$ cd build
$ cmake ../
$ cmake -DPLATFORM=rpiv2 -DNFM=ON -DRTLSDR=ON -DMIRISDR=OFF -DSOAPYSDR=OFF -DPULSEAUDIO=ON -DPROFILING=OFF ../
$ make
```

**Additional context**
This is the full cmake/make output while entering the above commands:
```
ubuntu@xxx:~/RTLSDR-Airband-4.1.0/build$ cmake ../
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to Release
-- Performing Test CXX_HAS_PTHREAD
-- Performing Test CXX_HAS_PTHREAD - Success
-- Performing Test CXX_HAS_FFAST_MATH
-- Performing Test CXX_HAS_FFAST_MATH - Success
-- Looking for sincosf
-- Looking for sincosf - found
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'libconfig++'
--   Found libconfig++, version 1.5
-- Found lame includes: /usr/include/lame/lame.h
-- Found lame library: /usr/lib/arm-linux-gnueabihf/libmp3lame.so
-- Checking for module 'shout'
--   Found shout, version 2.4.3
-- Looking for SHOUT_TLS_AUTO
-- Looking for SHOUT_TLS_AUTO - found
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN - found
-- Looking for SHOUT_TLS_RFC2818
-- Looking for SHOUT_TLS_RFC2818 - found
-- Looking for SHOUT_TLS_RFC2817
-- Looking for SHOUT_TLS_RFC2817 - found
-- Looking for SHOUT_TLS_DISABLED
-- Looking for SHOUT_TLS_DISABLED - found
-- Looking for shout_set_tls
-- Looking for shout_set_tls - found
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0
-- Found RTLSDR: /usr/lib/arm-linux-gnueabihf/librtlsdr.so
-- Checking for module 'libmirisdr'
--   No package 'libmirisdr' found
-- Could NOT find MiriSDR (missing: MIRISDR_LIBRARY MIRISDR_INCLUDE_DIR)
-- Checking for SoapySDR
-- Could NOT find SoapySDR (missing: SoapySDR_DIR)
--   SoapySDR not found
-- Checking for module 'libpulse'
--   Found libpulse, version 13.99.1
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.8
-- RTLSDR-Airband configuration summary:

-- - Build type:                Release
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: ON, enabled: FALSE
--   - soapysdr:                requested: ON, enabled: FALSE
-- - Other options:
--   - Platform:                default
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  FALSE
--   - NFM support:             OFF
--   - PulseAudio:              requested: ON, enabled: TRUE
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/RTLSDR-Airband-4.1.0/build
ubuntu@xxx:~/RTLSDR-Airband-4.1.0/build$ cmake -DPLATFORM=rpiv2 -DNFM=ON -DRTLSDR=ON -DMIRISDR=OFF -DSOAPYSDR=OFF -DPULSEAUDIO=ON -DPROFILING=OFF ../
-- Build type not specified: defaulting to Release
-- Found lame includes: /usr/include/lame/lame.h
-- Found lame library: /usr/lib/arm-linux-gnueabihf/libmp3lame.so
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/c++
-- Found BCM_VC: /usr/lib/arm-linux-gnueabihf/libbcm_host.so
-- RTLSDR-Airband configuration summary:

-- - Build type:                Release
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: OFF, enabled: FALSE
--   - soapysdr:                requested: OFF, enabled: FALSE
-- - Other options:
--   - Platform:                rpiv2
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  TRUE
--   - NFM support:             ON
--   - PulseAudio:              requested: ON, enabled: TRUE
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/RTLSDR-Airband-4.1.0/build
ubuntu@xxx:~/RTLSDR-Airband-4.1.0/build$ make
Scanning dependencies of target hello_fft
[  4%] Building CXX object src/hello_fft/CMakeFiles/hello_fft.dir/mailbox.c.o
In file included from /home/ubuntu/RTLSDR-Airband-4.1.0/src/hello_fft/mailbox.c:28:
/home/ubuntu/RTLSDR-Airband-4.1.0/src/hello_fft/../rtl_airband.h:33:10: fatal error: config.h: No such file or directory
   33 | #include "config.h"
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [src/hello_fft/CMakeFiles/hello_fft.dir/build.make:63: src/hello_fft/CMakeFiles/hello_fft.dir/mailbox.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:171: src/hello_fft/CMakeFiles/hello_fft.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

**Your rtl_airband.conf file**
(Not relevant in this case.)

charlie-foxtrot, 2023-05-01:
Reproduced and fixed build issues in #378

charlie-foxtrot, 2023-05-01:
@steve1515 if you're able to test #378 I would appreciate it, I dont currently have a rpi3 to run on

steve1515, 2023-05-01:
Tested and working.

charlie-foxtrot, 2023-05-02:
Thanks for the catch and help testing, fix merged and released in v4.1.1

closed 2023-05-02 by charlie-foxtrot