rtl-airband/RTLSDR-Airband #381: [BUG] building against libshout 2.4.6 fails with calls to deprecated functions

salfter, 2023-05-03:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit): 4.1.1
  - `make` options used to build the program: 
```
-- RTLSDR-Airband configuration summary:

-- - Build type:                RelWithDebInfo
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: ON, enabled: FALSE
--   - soapysdr:                requested: ON, enabled: FALSE
-- - Other options:
--   - Platform:                default
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  FALSE
--   - NFM support:             ON
--   - PulseAudio:              requested: ON, enabled: TRUE
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- <<< Gentoo configuration >>>
Build type      RelWithDebInfo
Install path    /usr
Compiler flags:
C
C++             -pipe -O2 -march=native -Wall -Wextra
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed
Module          -Wl,-O1 -Wl,--as-needed
Shared          -Wl,-O1 -Wl,--as-needed
```
  - Hardware platform (eg. x86_64, Raspberry Pi v4): amd64
  - Operating system name and version: Gentoo Linux

**What happened?**

With libshout 2.4.6 installed, compilation fails:

```
FAILED: src/CMakeFiles/rtl_airband_base.dir/output.cpp.o
/usr/bin/x86_64-pc-linux-gnu-g++ -D_FILE_OFFSET_BITS=64 -D_POSIX_C_SOURCE=200112L -I/var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1_build/src  -pipe -O2 -march=native -Wall -Wextra -pthread -ffast-math -std=c++11 -Werror -MD -MT src/CMakeFiles/rtl_airband_base.dir/output.cpp.o -MF src/CMakeFiles/rtl_airband_base.dir/output.cpp.o.d -o src/CMakeFiles/rtl_airband_base.dir/output.cpp.o -c /var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1/src/output.cpp
/var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1/src/output.cpp: In function ‘void shout_setup(icecast_data*, mix_modes)’:
/var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1/src/output.cpp:81:29: error: ‘int shout_set_format(shout_t*, unsigned int)’ is deprecated: Use shout_set_content_format() [-Werror=deprecated-declarations]
   81 |         if (shout_set_format(shouttemp, SHOUT_FORMAT_MP3) != SHOUTERR_SUCCESS){
      |             ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/shout/shout.h:10,
                 from /var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1/src/output.cpp:30:
/usr/include/x86_64-pc-linux-gnu/shout/shout.h:361:5: note: declared here
  361 | int shout_set_format(shout_t *self, unsigned int format) SHOUT_ATTR_F_DEPRECATED("Use shout_set_content_format()"); // obsolete
      |     ^~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1/src/output.cpp:84:43: error: ‘int shout_set_name(shout_t*, const char*)’ is deprecated: use shout_set_meta() with SHOUT_META_NAME [-Werror=deprecated-declarations]
   84 |         if(icecast->name && shout_set_name(shouttemp, icecast->name) != SHOUTERR_SUCCESS) {
      |                             ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/x86_64-pc-linux-gnu/shout/shout.h:348:5: note: declared here
  348 | int shout_set_name(shout_t *self, const char *name) SHOUT_ATTR_F_DEPRECATED("use shout_set_meta() with SHOUT_META_NAME"); // obsolete
      |     ^~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1/src/output.cpp:87:45: error: ‘int shout_set_genre(shout_t*, const char*)’ is deprecated: use shout_set_meta() with SHOUT_META_GENRE [-Werror=deprecated-declarations]
   87 |         if(icecast->genre && shout_set_genre(shouttemp, icecast->genre) != SHOUTERR_SUCCESS) {
      |                              ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/x86_64-pc-linux-gnu/shout/shout.h:354:5: note: declared here
  354 | int shout_set_genre(shout_t *self, const char *genre) SHOUT_ATTR_F_DEPRECATED("use shout_set_meta() with SHOUT_META_GENRE"); // obsolete
      |     ^~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1/src/output.cpp:90:57: error: ‘int shout_set_description(shout_t*, const char*)’ is deprecated: use shout_set_meta() with SHOUT_META_DESCRIPTION [-Werror=deprecated-declarations]
   90 |         if(icecast->description && shout_set_description(shouttemp, icecast->description) != SHOUTERR_SUCCESS) {
      |                                    ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/x86_64-pc-linux-gnu/shout/shout.h:357:5: note: declared here
  357 | int shout_set_description(shout_t *self, const char *description) SHOUT_ATTR_F_DEPRECATED("use shout_set_meta() with SHOUT_META_DESCRIPTION"); // obsolete
      |     ^~~~~~~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1/src/output.cpp: In function ‘void process_outputs(channel_t*, int)’:
/var/tmp/portage/net-wireless/rtlsdr-airband-4.1.1/work/RTLSDR-Airband-4.1.1/src/output.cpp:479:55: error: ‘int shout_set_metadata(shout_t*, shout_metadata_t*)’ is deprecated: Use shout_set_metadata_utf8() [-Werror=deprecated-declarations]
  479 |                                 if (shout_set_metadata(icecast->shout, meta) != SHOUTERR_SUCCESS) {
      |                                     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/usr/include/x86_64-pc-linux-gnu/shout/shout.h:367:5: note: declared here
  367 | int shout_set_metadata(shout_t *self, shout_metadata_t *metadata) SHOUT_ATTR_F_WARN_UNUSED_RESULT SHOUT_ATTR_F_DEPRECATED("Use shout_set_metadata_utf8()");
      |     ^~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
```

If I roll libshout back to 2.4.5, rtlsdr-airband builds successfully.

**What you expected to happen?**

successful build

**Steps to Reproduce**

```sudo emerge -1 =libshout-2.4.6 && sudo emerge rtlsdr-airband```: fails
```sudo emerge -1 =libshout-2.4.5 && sudo emerge rtlsdr-airband```: succeeds

(Note that libshout 2.4.6 is the current version in Portage.  I added libshout 2.4.5 to my overlay (https://gitlab.com/salfter/portage) for testing; an ebuild for rtlsdr-airband is also in my overlay.)

charlie-foxtrot, 2023-05-04:
@salfter can you give #382 a try?  Do you have a way to test the metadata tags in addition to compiling?

salfter, 2023-05-04:
The patch fixes the compile problem.  I don't have a way to test the tags, though...it's functionality I haven't been using.

closed 2023-05-04 by salfter