rtl-airband/RTLSDR-Airband #383: [audio is distorted - maybe due to narrowband ?]

israelgoldstein, 2023-05-07:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit): 4.1.0
  - `make` options used to build the program: NFM ON
  - Hardware platform (eg. x86_64, Raspberry Pi v4):thinkcentre server 
  - Operating system name and version: ubuntu 

**What happened?**
The wav files that's being generated sound distorted. 

**What you expected to happen?**
I expect it to sound clear.

**Steps to Reproduce**

**Additional context**
Add any other relevant information about the problem here.

**Your rtl_airband.conf file**
Remove passwords, server addresses and other private information.

charlie-foxtrot, 2023-05-09:
Is this a new problem or something that has always been there?

If you run with `-f` what are the signal and noise levels like?

Is this specific to these channels or does it happen with others as well?

You aren't setting a `correction` value for the SDR, have you confirmed there is no tuning error?

I don't have a specific service for uploading audio files but there are some free ones out there where you can share a file with a public link (google drive, drop box, etc)

israelgoldstein, 2023-05-16:
this has been ongoing for a while, 

attached is the output:

RTLSDR-Airband version 4.0.3 starting
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 2300000.022848 Hz
[R82XX] PLL not locked!
Device #1: gain set to 1.40 dB
RTLSDR device 1 initialized
Allocating 10 zero-copy buffers
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Device #2: gain set to 3.70 dB
RTLSDR device 2 initialized
Allocating 10 zero-copy buffers
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Device #0: gain set to 14.40 dB
RTLSDR device 0 initialized
Allocating 10 zero-copy buffers
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Device #3: gain set to 16.60 dB
RTLSDR device 3 initialized
Allocating 10 zero-copy buffers

 476.562   476.438   476.337   476.587   476.387   476.312   476.637   476.362   476.887   476.537   476.837   476.912   476.487   476.962   476.662   476.462   476.512   476.938   476.862   477.012   476.412   476.987   476.737   476.787   476.762   476.688   477.137   477.087   477.037   477.112   476.712   476.612   477.062   -50/-51  -51/-52   -53/-53   -27/-51*  -36/-51*  -45/-52   -50/-51   -53/-53   -49/-50*  -50/-51   -32/-50*  -50/-51   -52/-52   -33/-50*  -50/-51   -51/-52   -51/  -51/-51  -52/-52   -52/-53   -27/-51*  -35/-51*  -45/-52   -50/-51   -52/-53   -50/-50   -52/-51   -31/-50*  -50/-51   -52/-52   -33/-50*  -50/-51   -51/-52   -50/  -50/-51  -52/-52   -53/-53   -27/-50*  -37/-51*  -46/-52   -50/-51   -52/-53   -50/-50   -50/-51   -32/-50*  -50/-51   -50/-52   -33/-50*  -50/-51   -51/-52   -51/  -50/-51  -51/-52   -52/-53   -27/-50*  -36/-51*  -46/-52   -49/-51   -52/-53   -50/-50   -50/-51   -31/-50*  -49/-51   -51/-52   -33/-50*  -50/-51   -51/-52   -50/  -50/-51  -52/-52   -53/-53   -27/-50*  -36/-51*  -45/-52   -51/-51   -52/-53   -50/-50   -51/-51   -49/-50*  -50/-51   -51/-52   -33/-50*  -49/-51   -51/-52   -51/  -50/-51  -51/-52   -52/-53   -27/-50*  -37/-51*  -46/-52   -49/-51   -52/-53   -50/-50   -51/-51   -50/-50   -49/-51   -50/-52   -33/-50*  -50/-51   -51/-52   -50/  -51/-51  -51/-52   -52/-53   -27/-50*  -35/-51*  -45/-52   -50/-51   -52/-53   -49/-50   -50/-51   -49/-50   -50/-51   -51/-52   -33/-50*  -50/-51   -52/-52   -51/  -50/-51  -52/-52   -52/-53   -48/-50*  -37/-51*  -46/-52   -50/-51   -52/-53   -49/-50   -50/-51   -49/-50   -50/-51   -50/-52   -33/-50*  -50/-51   -51/-52   -50/ -50/-51   -51/-52   -53/-53   -28/-51*  -35/-51*  -46/-52   -50/-51   -52/-53   -37/-50*  -50/-51   -31/-50*  -49/-51   -51/-52   -33/-50*  -51/-51   -50/-52   -50/  -51/-51  -52/-52   -52/-53   -27/-51*  -37/-51*  -45/-52   -50/-51   -52/-53   -37/-50*  -50/-51   -32/-50*  -50/-51   -51/-52   -33/-50*  -50/-51   -51/-52   -51/  -50/-51  -51/-52   -52/-53   -27/-51*  -35/-51*  -45/-52   -50/-51   -51/-53   -37/-50*  -51/-51   -31/-50*  -50/-51   -51/-52   -33/-50*  -50/-51   -51/-52   -51/  -50/-51  -52/-52   -52/-53   -27/-51*  -36/-51*  -45/-52   -50/-51   -53/-53   -36/-50*  -50/-51   -31/-50*  -49/-51   -51/-52   -33/-50*  -50/-51   -51/-52   -51/ -49/-51  
-------------------------------------------------------------------------------

 464.662  

 -50/-52  
 -51/-52  
 -52/-52  
 -51/-52  
 -50/-52  
 -51/-52  
 -51/-52  
 -51/-52  
 -51/-52  
 -50/-52  
 -51/-52  
 -50/-52  

-------------------------------------------------------------------------------

 470.837   470.812   470.688   470.712   470.862   470.887  

 -50/-51  -50/-51   -25/-51*  -49/-51   -50/-51   -13/-48* 
 -49/-51  -50/-51   -25/-51*  -51/-51   -50/-51   -15/-48* 
 -49/-50  -50/-51   -25/-51*  -50/-51   -50/-51   -11/-48* 
 -50/-50  -50/-51   -26/-51*  -50/-51   -50/-51   -13/-48* 
 -49/-51  -50/-51   -24/-51*  -49/-51   -50/-51   -13/-48* 
 -50/-51  -49/-51   -25/-51*  -50/-51   -50/-51   -12/-48* 
 -50/-51  -50/-51   -25/-51*  -50/-51   -50/-51   -13/-48* 
 -30/-51* -50/-51   -25/-51*  -50/-51   -51/-51   -12/-48* 
 -29/-51* -50/-51   -25/-51*  -51/-51   -50/-51   -12/-48* 
 -50/-51  -50/-51   -24/-51*  -50/-51   -50/-51   -12/-48* 
 -50/-51  -50/-51   -25/-51*  -50/-51   -50/-51   -15/-48* 
 -49/-51  -49/-51   -24/-51*  -50/-51   -50/-51   -11/-48* 

 -15/-52* -50/-51   -50/-52   -51/-53   -51/-52   -49/-51   -51/-51   ---------

it seems to happen with all channels received by this device

how do i know my correction? the SDR advertises as having a good click chip, so it doesn't drift off. 
but i see some channels are off, for example 470.688 when its supposed to be 470.687

charlie-foxtrot, 2023-05-25:
I use SDR++ (https://www.sdrpp.org) to visualize both gain and offset adjustment.  If you can find an always transmitting signal at a known frequency (ie NOAA weather in the US) you can use that to determine the correction.  If your correction is +/- 2 or less then that likely is not causing your problem.

Next in SDR++ (or equivalent) tune to the center frequency you're setting and watch the signals for a while as you change  the gain.  You may have a strong signal on a neighboring frequency that is causing issues.

The config you posted had just 2 frequencies `463.0625` and `463.1875` while your output from running `-f` didn't have those frequencies but had many more.  I would recommend isolating to a few channels, getting those to work, then adding more.

charlie-foxtrot, 2023-06-09:
@israelgoldstein are you still seeing this issue?  If not, please close

israelgoldstein, 2023-06-13:
Hi, yes my audio is still distorted, i cant figure out why.  i didn't get to try sdr++

israelgoldstein, 2023-06-13:
i was able to upload a sample file to: https://file.io/sbO8bxowndai

israelgoldstein, 2023-07-05:
Attached is my config file for this setup

all channels user Bandwidth set to 12500 and also streaming to icecast 

`fft_size = 1024;
localtime = true;
log_scan_activity = true;
multiple_demod_threads = true;
devices: (
  {
    #disable = true;
    type = "rtlsdr";
    serial = "70000701";
    gain = 39.0;
    centerfreq = 477.300;
    sample_rate = 2.4;
    channels:
    (
      #Channels Go here
    );
  },
  {
    #disable = true;
    type = "rtlsdr";
    serial = "70000702";
    gain = 35.0;
    centerfreq = 482.900;
    sample_rate = 2.4;
    channels:
    (
      #Channels Go Here
    );
  },
  {
    #disable = true;
    type = "rtlsdr";
    serial = "70000703";
    gain = 25.0;
    centerfreq = 470.100;
    sample_rate = 2.4;
    channels:
    (
      #Channels Go Here
    );
  }
);
`