rtl-airband/RTLSDR-Airband #385: [Failed to open SoapySDR device]

israelgoldstein, 2023-05-21:
Using RTLSDR-Airband-4.1.1

this is the configuration for the AirSPY Mini im trying to use
   #disable = true;
    #type = "rtlsdr";
    #serial = "40000401";
    type = "soapysdr";
    device_string = "driver=airspy, serial=F8A868FF296E4E47";
    gain = "LNA=13,MIX=10,VGA=10";
    sample_rate = 6.0;
    #gain = 15.0;
    #sample_rate = 2.3;
    centerfreq = 120.000;

when i start rtl_airband, 
`rtl_airband -fe`

im getting the following error
`Failed to open SoapySDR device 'driver=airspy, serial=F8A868FF296E4E47': SoapySDR::Device::make() no match`

if i remove the AirSPY Mini, and use an rtlsdr, it loads fine

when i run 
`SoapySDRUtil --info`

> 
> Lib Version: v0.8.1-gd2496c50
> API Version: v0.8.200
> ABI Version: v0.8-3
> Install root: /usr/local
> Search path:  /usr/local/lib/SoapySDR/modules0.8-3
> Module found: /usr/local/lib/SoapySDR/modules0.8-3/librtlsdrSupport.so (0.3.3-40d0708)
> Available factories... rtlsdr
> Available converters...
>  -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
>  -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
>  -  CS32 -> [CS32]
>  -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
>  -  CU16 -> [CF32, CS16, CS8]
>  -   CU8 -> [CF32, CS16, CS8]
>  -   F32 -> [F32, S16, S8, U16, U8]
>  -   S16 -> [F32, S16, S8, U16, U8]
>  -   S32 -> [S32]
>  -    S8 -> [F32, S16, S8, U16, U8]
>  -   U16 -> [F32, S16, S8]
>  -    U8 -> [F32, S16, S8]

charlie-foxtrot, 2023-05-25:
Did this work with prior versions of RTLSDR-Airband?

I haven't tried `soapysdr` but comparing what you have to the wiki here https://github.com/charlie-foxtrot/RTLSDR-Airband/wiki/Configuring-SoapySDR-devices one difference I notice is your `device_string` has a space after the comma, perhaps thats throwing something goff?

Your `serial` value is much longer than the examples and some examples use `hardware`, have you tried different variations of those?

The Airspy example uses `device_id=0`, maybe thats worth a try as well?

charlie-foxtrot, 2023-05-29:
Looking closer at your output of `SoapySDRUtil --info` it lists `rtlsdr` but not `airspy`:

```
...
Available factories... rtlsdr
...
```

Using a rpi3, after installing the following:
```
sudo apt-get install -y soapysdr-module-airspy soapysdr-tools libsoapysdr-dev libairspy-dev airspy
```

I see:
```
$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.2-2
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr
Search path:  /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.7
Search path:  /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.7               (missing)
Search path:  /usr/local/lib/SoapySDR/modules0.7                                   (missing)
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.7/libairspySupport.so (0.1.2)
Available factories... airspy
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

then I can do find:
```
$ sudo SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device_id = 0
  driver = airspy
  label = AIRSPY [XXXXXXXXX]
  serial = XXXXXXXXX
```

and probe (NOTE: for me I needed to use `sudo` for the probe):
```
$ sudo SoapySDRUtil --probe="driver=airspy,device_id=0"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=airspy,device_id=0

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=Airspy
  hardware=Airspy
  device_id=0
  origin=https://github.com/pothosware/SoapyAirspy

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Bias tee - Enable the 4.5v DC Bias tee to power SpyVerter / LNA / etc. via antenna connection.
       [key=biastee, default=false, type=bool]
     * Bit Pack - Enable packing 4 12-bit samples into 3 16-bit words for 25% less USB trafic.
       [key=bitpack, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Full gain range: [0, 45] dB
    LNA gain range: [0, 15] dB
    MIX gain range: [0, 15] dB
    VGA gain range: [0, 15] dB
  Full freq range: [24, 1800] MHz
    RF freq range: [24, 1800] MHz
  Sample rates: 6, 3 MSps
```

charlie-foxtrot, 2023-06-09:
Assuming this is resolved as other issues have been opened indicating some recordings are coming through.  If this is still an issue please re-open.

closed 2023-06-09 by charlie-foxtrot

israelgoldstein, 2023-07-06:
i still have this issue on a new install 

im lacking info to get soapysdr working

`
root@SDRPC303:~/RTLSDR-Airband/build# SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-gbb33b2d2
API Version: v0.8.200
ABI Version: v0.8-3
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8-3 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
root@SDRPC303:~/RTLSDR-Airband/build# `

charlie-foxtrot, 2023-07-06:
The output you just posted shows:

```
...
Available factories... No factories found!
...
```

Where as my output shows:

```
...
Available factories... airspy
...
```

Did you install all the packages from my comment from May 29th?

```
sudo apt-get install -y soapysdr-module-airspy soapysdr-tools libsoapysdr-dev libairspy-dev airspy
```

israelgoldstein, 2023-07-06:
i tried and it didnt help

so this finally helped

`git clone https://github.com/pothosware/SoapyAirspy.git
cd SoapyAirspy
mkdir build
cd build
cmake ..
make
sudo make install`

israelgoldstein, 2023-08-28:
when i check info, i see it find airspy

root@SDRPC001:~# SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-gbb33b2d2
API Version: v0.8.200
ABI Version: v0.8-3
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8-3
Module found: /usr/local/lib/SoapySDR/modules0.8-3/libairspySupport.so (0.2.0-411f73e)
Available factories... airspy
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

i have 4 airspys plugged in

root@SDRPC001:~# airspy_info
airspy_lib_version: 1.0.11

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc10-6-g4008185 2020-05-08
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0xF8A868FF29679947
Supported sample rates:
        6.000000 MSPS
        3.000000 MSPS
Close board 1

Found AirSpy board 2
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc10-6-g4008185 2020-05-08
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x744C60C8215B644F
Supported sample rates:
        6.000000 MSPS
        3.000000 MSPS
Close board 2

Found AirSpy board 3
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc10-6-g4008185 2020-05-08
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x744C60C8214D6E4F
Supported sample rates:
        6.000000 MSPS
        3.000000 MSPS
Close board 3

Found AirSpy board 4
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc10-6-g4008185 2020-05-08
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x62CC68FF21865317
Supported sample rates:
        6.000000 MSPS
        3.000000 MSPS
Close board 4

but when i start rtl application, it  cant open the 4th airspy ?
root@SDRPC001:~# sudo rtl_airband -fe
SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': using native sample format 'CS16' (fullScale=32767.0)
SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': using native sample format 'CS16' (fullScale=32767.0)
SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': using native sample format 'CS16' (fullScale=32767.0)
SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': using native sample format 'CS16' (fullScale=32767.0)
RTLSDR-Airband version 4.1.1 starting
SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': sample rate set to 6000000 sps
SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': AGC off (requested: off)
SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': gain 'LNA' set to 5.0 dB
SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': gain 'MIX' set to 5.0 dB
SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': gain 'VGA' set to 15.0 dB
SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947' initialized
[INFO] Using format CS16.
SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947' started
SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': sample rate set to 6000000 sps
SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': AGC off (requested: off)
SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': gain 'LNA' set to 5.0 dB
SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': gain 'MIX' set to 5.0 dB
SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': gain 'VGA' set to 15.0 dB
SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317' initialized
[INFO] Using format CS16.
SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317' started
SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': sample rate set to 6000000 sps
SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': AGC off (requested: off)
SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': gain 'LNA' set to 5.0 dB
SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': gain 'MIX' set to 5.0 dB
SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': gain 'VGA' set to 15.0 dB
SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F' initialized
[INFO] Using format CS16.
SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F' started
Failed to open SoapySDR device 'driver=airspy,serial=0x744C60C8214D6E4F': Unable to open AirSpy device with serial 744c60c8214d6e4f
root@SDRPC001:~#