rtl-airband/RTLSDR-Airband #389: [BUG] cant use DPL?

israelgoldstein, 2023-06-08:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit): RTLSDR-Airband version 4.1.1

**What happened?**
Jun 08 12:16:12 SDRPC001 rtl_airband[228221]: Configuration error: devices.[0] channels.[3]: notch should be an float or a list of floats with at least 1 elements

**Your rtl_airband.conf file**
{
        #disable = true;
        freq = 151.0050;
        lowpass = 3200;
        highpass = 200;
        bandwidth = 5000;
        modulation = "nfm";
        notch = 465;
        outputs: (
          {
            #disable = true;
            type = "file";
            directory = "/home/pi/scripts/rtlairband_recordings/Conv_01/2001/2001004";
            filename_template = "Police 10";
            split_on_transmission = true;
            include_freq = true;
          }
        );
      }

charlie-foxtrot, 2023-06-09:
change `notch = 465;` to `notch = 465.0;` to make it a float.  I'll leave this issue open to allow `int`.

NOTE: CTCSS / PL is supported, the digital versions (ie DPL is not)

israelgoldstein, 2023-06-09:
465 is the DPL, not PL

so DPL is not supported?
what would it take to support DPL?

charlie-foxtrot, 2023-06-09:
Correct, DPL is not currently supported.

Adding support would require coding up a decoder, wiring the decoder into `squelch.cpp`, then adding configuration option(s).

If this is something you're interested in doing please open a PR against the `unstable` branch.  I don't have any DPL signals locally so wont be able to test so please include unit and integration tests.

closed 2023-09-15 by charlie-foxtrot