rtl-airband/RTLSDR-Airband #391: [BUG] readStream failed: TIMEOUT

israelgoldstein, 2023-06-27:
when I run this application, i get an error upon bootup

if i Disable from the config file the last AirSpy, then it starts up fine.. 

Jun 26 23:07:42 SDRPC001 systemd[1]: Started SDR AM/NFM demodulator.
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': using native sample format 'CS16' (fullScale=32767.0)
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': using native sample format 'CS16' (fullScale=32767.0)
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8215B644F': using native sample format 'CS16' (fullScale=32767.0)
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': using native sample format 'CS16' (fullScale=32767.0)
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: RTLSDR-Airband version 4.1.1 starting
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': sample rate set to 6000000 sps
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': AGC off (requested: off)
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': gain 'LNA' set to 5.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': gain 'MIX' set to 5.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947': gain 'VGA' set to 15.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947' initialized
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: [INFO] Using format CS16.
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0xF8A868FF29679947' started
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': sample rate set to 6000000 sps
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': AGC off (requested: off)
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': gain 'LNA' set to 5.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': gain 'MIX' set to 5.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317': gain 'VGA' set to 15.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317' initialized
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: [INFO] Using format CS16.
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x62CC68FF21865317' started
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8215B644F': sample rate set to 6000000 sps
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8215B644F': AGC off (requested: off)
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8215B644F': gain 'LNA' set to 5.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8215B644F': gain 'MIX' set to 5.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8215B644F': gain 'VGA' set to 15.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8215B644F' initialized
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: [INFO] Using format CS16.
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8215B644F' started
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': sample rate set to 6000000 sps
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': AGC off (requested: off)
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': gain 'LNA' set to 5.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': gain 'MIX' set to 5.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F': gain 'VGA' set to 15.0 dB
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F' initialized
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: [INFO] Using format CS16.
Jun 26 23:07:42 SDRPC001 rtl_airband[56951]: SoapySDR: device 'driver=airspy,serial=0x744C60C8214D6E4F' started
Jun 26 23:07:43 SDRPC001 rtl_airband[56951]: SoapySDR device 'driver=airspy,serial=0x744C60C8214D6E4F': readStream failed: TIMEOUT
Jun 26 23:07:44 SDRPC001 rtl_airband[56951]: SoapySDR device 'driver=airspy,serial=0x744C60C8214D6E4F': readStream failed: TIMEOUT
Jun 26 23:07:45 SDRPC001 rtl_airband[56951]: SoapySDR device 'driver=airspy,serial=0x744C60C8214D6E4F': readStream failed: TIMEOUT
^C

charlie-foxtrot, 2023-07-03:
Can you provide your `conf` file?  What hardware are you using?  It's possible you're overloading the USB bus, are you connecting them to a USB 3.0 hub?

israelgoldstein, 2023-07-03:
my conf file is FULL of channels - which i wouldn't want to post here - 

but the config i know is good, with just 3 AirSpys its good, with the 4th one its giving an issue
and if i disable the 3rd one, the 4th one will run fine.. so its an issue when i have 4 Active at a time

all my dongles on this PC im using all 4 AirSpy Mini's
plugged into the computer directly, no SUB Hubs

how do i check if the USB is being overloaded?

charlie-foxtrot, 2023-07-03:
There are configuration parameters beyond the specific channels / frequencies / names that may be important for debugging.  Can you strip the channel specifics from the config file and share that?

I would suggest playing with the bandwidth and / or the output format to see if you can get all 4 running at the same time.  6000000 samples per second x 16 bit samples x 2 for complex readings x 4 dongles = a lot of data.

`usbtop` is a utility that will show how much USB bandwidth is being used, running that while stepping through 1-4 active dongles may be informative.

I would recommend splitting your config into 4 pieces, one config file per dongle, and run multiple instances of `rtl-airband` simultaneously.  If you can run 4 independent instances of `rtl_airband`, each with their own dongle but you can't run one instance with 4 dongles then the next step would be to look at `rtl_airband`.  If on the other hand you can't get 4 independent instances running, then the next step would be to look at the OS, USB bus, hardware, etc.

israelgoldstein, 2023-07-05:
After removing my Channels, here is the config file

FYI with the channels still in the config, this will run, as the last dongle was disabled,

```
fft_size = 2048;
localtime = true;
log_scan_activity = true;
multiple_demod_threads = true;
devices: (
  {
    #disable = true;
    type = "soapysdr";
    device_string = "driver=airspy,serial=0xF8A868FF29679947";
    gain = "LNA=5,MIX=5,VGA=15";
    sample_rate = 6.0;
    centerfreq = 153.000;
    channels:
    (
      #Channels Go Here
    );
  },
  {
    #disable = true;
    type = "soapysdr";
    device_string = "driver=airspy,serial=0x62CC68FF21865317";
    gain = "LNA=5,MIX=5,VGA=15";
    sample_rate = 6.0;
    centerfreq = 157.200;
    channels:
    (
        #Channels Go Here
    );
  },
  {
    #disable = true;
    type = "soapysdr";
    device_string = "driver=airspy,serial=0x744C60C8214D6E4F";
    gain = "LNA=5,MIX=5,VGA=15";
    sample_rate = 6.0;
    centerfreq = 461.500;
    channels:
    (
      #Channels Go Here
    );
  },
  {
    disable = true;
    type = "soapysdr";
    device_string = "driver=airspy,serial=0x744C60C8214D6E4F";
    gain = "LNA=5,MIX=5,VGA=15";
    sample_rate = 6.0;
    centerfreq = 453.500;
    channels:
    (
      #Channels Go Here
    );
  }
);
```

here is a "sample" of what my Channels configuration look like 

```
{
        #disable = true;
        freq = 151.0125;
        lowpass = 3200;
        highpass = 200;
        bandwidth = 5000;
        modulation = "nfm";
        notch = 146.2;
        ctcss = 146.2;
        outputs: (
          {
            #disable = true;
            type = "file";
            directory = "/path/torecordings/Channel1";
            filename_template = "Channel1";
            split_on_transmission = true;
            include_freq = true;
          }
        );
      }
```

israelgoldstein, 2023-08-29:
finally installed usbtop

as I just revisited this issue, and i still get errors of

Aug 28 22:16:35 SDRPC001 rtl_airband[181131]: SoapySDR device 'driver=airspy,serial=0x744C60C8215B644F': readStream failed: TIMEOUT
Aug 28 22:16:36 SDRPC001 rtl_airband[181131]: SoapySDR device 'driver=airspy,serial=0x744C60C8215B644F': readStream failed: TIMEOUT
Aug 28 22:16:37 SDRPC001 rtl_airband[181131]: SoapySDR device 'driver=airspy,serial=0x744C60C8215B644F': readStream failed: TIMEOUT
Aug 28 22:16:38 SDRPC001 rtl_airband[181131]: SoapySDR device 'driver=airspy,serial=0x744C60C8215B644F': readStream failed: TIMEOUT
Aug 28 22:16:39 SDRPC001 rtl_airband[181131]: SoapySDR device 'driver=airspy,serial=0x744C60C8215B644F': readStream failed: TIMEOUT
Aug 28 22:16:40 SDRPC001 rtl_airband[181131]: SoapySDR device 'driver=airspy,serial=0x744C60C8215B644F': readStream failed: TIMEOUT
Aug 28 22:16:41 SDRPC001 rtl_airband[181131]: SoapySDR device 'driver=airspy,serial=0x744C60C8215B644F': readStream failed: TIMEOUT
Aug 28 22:16:42 SDRPC001 rtl_airband[181131]: SoapySDR device 'driver=airspy,serial=0x744C60C8215B644F': readStream failed: TIMEOUT

so here is my usbtop output

Bus ID 0 (Raw USB traffic, all USB buses)       To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               5.79 KiB/s      23700.41 KiB/s
  Device ID   3 :                               5.78 KiB/s      23700.91 KiB/s
  Device ID   4 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   5 :                               5.78 KiB/s      23699.50 KiB/s

israelgoldstein, 2023-11-01:
i just tried changing 2 of the AirSpy's to 3.0 Sample Rate instead of 6.0, and i was still getting the same behavior

israelgoldstein, 2023-11-01:
Bus ID 0 (Raw USB traffic, all USB buses)       To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               5.79 KiB/s      23699.76 KiB/s
  Device ID   3 :                               2.92 KiB/s      11981.80 KiB/s
  Device ID   4 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   5 :                               5.79 KiB/s      23700.85 KiB/s
Bus ID 1 (Raw USB traffic, bus number 1)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               0.00 KiB/s      0.00 KiB/s
Bus ID 2 (Raw USB traffic, bus number 2)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               0.00 KiB/s      0.00 KiB/s
Bus ID 3 (Raw USB traffic, bus number 3)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               5.79 KiB/s      23699.97 KiB/s
  Device ID   3 :                               2.92 KiB/s      11981.74 KiB/s
  Device ID   4 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   5 :                               5.79 KiB/s      23700.85 KiB/s
Bus ID 4 (Raw USB traffic, bus number 4)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s

israelgoldstein, 2023-11-01:
Bus ID 0 (Raw USB traffic, all USB buses)       To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               5.78 KiB/s      23701.24 KiB/s
  Device ID   3 :                               2.92 KiB/s      11981.71 KiB/s
  Device ID   4 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   5 :                               5.78 KiB/s      23699.92 KiB/s
Bus ID 1 (Raw USB traffic, bus number 1)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               0.00 KiB/s      0.00 KiB/s
Bus ID 2 (Raw USB traffic, bus number 2)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               0.00 KiB/s      0.00 KiB/s
Bus ID 3 (Raw USB traffic, bus number 3)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               5.78 KiB/s      23701.09 KiB/s
  Device ID   3 :                               2.92 KiB/s      11981.75 KiB/s
  Device ID   4 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   5 :                               5.78 KiB/s      23700.12 KiB/s
Bus ID 4 (Raw USB traffic, bus number 4)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s

Bus ID 0 (Raw USB traffic, all USB buses)       To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               5.79 KiB/s      23699.76 KiB/s
  Device ID   3 :                               2.92 KiB/s      11981.80 KiB/s
  Device ID   4 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   5 :                               5.79 KiB/s      23700.85 KiB/s
Bus ID 1 (Raw USB traffic, bus number 1)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               0.00 KiB/s      0.00 KiB/s
Bus ID 2 (Raw USB traffic, bus number 2)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               0.00 KiB/s      0.00 KiB/s
Bus ID 3 (Raw USB traffic, bus number 3)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   2 :                               5.79 KiB/s      23699.97 KiB/s
  Device ID   3 :                               2.92 KiB/s      11981.74 KiB/s
  Device ID   4 :                               0.00 KiB/s      0.00 KiB/s
  Device ID   5 :                               5.79 KiB/s      23700.85 KiB/s
Bus ID 4 (Raw USB traffic, bus number 4)        To device       From device
  Device ID   1 :                               0.00 KiB/s      0.00 KiB/s

gabeschine, 2025-02-27:
@israelgoldstein checking in. Are you still experiencing the same behavior?