rtl-airband/RTLSDR-Airband #392: [FEATURE] SRT Support

memchk, 2023-07-02:
**Is your feature request related to a problem? If so, please describe.**
I use RTLSDR-Airband to provide low-latency access to my airparks CTAF frequency. For this, I run the software dockerized along with liquidsoap to produce a web-consumable HLS stream. Currently this is painful as liquidsoap has effectively deprecated raw UDP support. I use `srt-live-transmit` to convert the raw UDP to SRT locally, then forward to the liquidsoap server.

**Describe the solution you'd like**
Implement the [SRT](https://github.com/Haivision/srt) protocol which is natively supported by both liquidsoap and ffmpeg. This gives a better transport than raw UDP (FEC, some ability to handle packet loss, etc.) without greatly complicating the encoding as we can still send raw PCM data.

**Describe alternative solutions or features you've considered**
We could instead support RTP/RTMP, but SRT has made an effort to be relatively easily integrated, and seems to be the emerging de-facto standard for audio/video transport.

memchk, 2023-07-02:
I would be happy to contribute support for this, but I wanted to open an issue to discuss alternatives and ensure that this feature is wanted by upstream first.

OA6DXV, 2023-07-04:
@memchk 
Thank you so much! srt-live-stream was just what I was looking for for so long to be able to work with steam UDP. Although for some reason when working with the raw s32le audio format (exporting by rtl_airband), sending it by srt and playing it from another computer with ffplay, the audio sounds choppy. 

The solution that I finally found was to use ffmpeg on the local machine (to be able to use the local udp stream) and convert it back to f32le (I don't understand what happens since the input is the same as the output) but in this way, when sending it by srt the audio is perfect!

**RTL_Airband udp_stream output port: 8889**
 ffmpeg -f s32le -ar 16000 -ac 2 -i udp://127.0.0.1:8889 -b:a 128k -f s32le udp://0.0.0.0:8891

**Take ffmpeg output on port 8891 and send via srt on port 8890**
srt-live-transmit udp://127.0.0.1:8891 -o srt://0.0.0.0:8890?mode="listener"

**and using ffplay on my windows pc** 
ffplay.exe -ac 1 -ar 8000 -analyzeduration 0 -probesize 32 -f f32le -i srt://10.0.5.20:8890

I don't know why I have to specify that the stream is a single channel and with a bitrate of 8k, when ffmpeg is using 2 channels and a bitrate of 16k, it's the only way I managed to make it sound perfect.
I can't use the SRT stream directly from VLC because it doesn't recognize the audio in f32le format

jayant4001, 2023-07-06:
Works like a charm using ffmpeg and srt-live-transmit. Thanks very much @OA6DXV!

yegors, 2025-06-29:
I forked the project and added highly experimental SRT support: https://github.com/yegors/RTLSDR-Airband 

You will need `libsrt`  (I used `libsrt1.4-openssl`) for this to compile, the build instructions are the same as the original. 

```
-- RTLSDR-Airband configuration summary:

-- - Version string:            b4fe4e8
-- - Build type:                Release
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: ON, enabled: FALSE
--   - soapysdr:                requested: ON, enabled: TRUE
-- - Other options:
--   - Platform:                native
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  FALSE
--   - NFM support:             OFF
--   - PulseAudio:              requested: ON, enabled: TRUE
--   - SRT output:              requested: ON, enabled: TRUE                     <------ new stuff
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/RTLSDR-Airband/build
```

Then use new type=srt which will stream raw pcm

```
outputs: (
          {
            type = "srt";
            listen_address = "0.0.0.0";
            listen_port = 8890;
          }
        );
```

You can consume it using command: `ffplay -ac 1 -ar 8000 -analyzeduration 0 -probesize 32 -f f32le srt://<host>:8890`

You can optionally specify `format='mp3'` in the output block,  which will encode the stream as an mp3, which means it will work in VLC directly, albeit with a ~1s delay  so I'm gonna try to fix this. 

I'm not a C++ dev, and this entire thing was AI coded, so use at work risk. :)