rtl-airband/RTLSDR-Airband #393: [BUG] mirisdr_set error building

kiricheglov, 2023-07-04:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit): 4.1.1
  - `make` options used to build the program: cmake && sudo make install
  - Hardware platform (eg. x86_64, Raspberry Pi v4): RPi v4
  - Operating system name and version: Debian 11

**What happened?**
i give error while airband is building:
/home/rootmalina/RTLSDR-Airband/src/input-mirisdr.cpp: In function ‘int mirisdr_init(input_t*)’:
/home/rootmalina/RTLSDR-Airband/src/input-mirisdr.cpp:106:10:error: ‘mirisdr_set_transfer’ was not declared in this scope
  106 |  int r = mirisdr_set_transfer(miri, (char *)"BULK");
      |          ^~~~~~~~~~~~~~~~~~~~
/home/rootmalina/RTLSDR-Airband/src/input-mirisdr.cpp:135:6: error: ‘mirisdr_set_sample_format’ was not declared in this scope; did you mean ‘mirisdr_set_sample_rate’?
  135 |  r = mirisdr_set_sample_format(miri, (char *)"504_S8");
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
      |      mirisdr_set_sample_rate
make[2]: *** [src/CMakeFiles/rtl_airband_base.dir/build.make:284: src/CMakeFiles/rtl_airband_base.dir/input-mirisdr.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:142: src/CMakeFiles/rtl_airband_base.dir/all] Error 2
make: *** [Makefile:149: all] Error 2

charlie-foxtrot, 2023-07-04:
Are you trying to build `mirisdr` support?  If not you can disable it by adding `-DMIRISDR=OFF` to the cmake command:

```
mkdir build && cd build && cmake -DMIRISDR=OFF ../
```

If you are trying to build `mirisdr` support please include the cmake output and details on what version of MiriSDR library was found

kiricheglov, 2023-07-05:
Thanks, its work for me:)
but i have problem: mp3 file is not created, i run "sudo rtl_airband -f -c rtl_airband.conf" and before minute stop, use shortcut "ctrl-c"

charlie-foxtrot, 2023-07-05:
Glad that worked for you, I'm going to close this issue.

For the MP3 not coming out, depending on configuration output files may only be generated if there is activity detected.  Look at the Wiki page for running the program, run in foreground to adjust your gain, and verify you are tuned to channels with signal.  If there is still a problem you can open a new ticket

closed 2023-07-05 by charlie-foxtrot