rtl-airband/RTLSDR-Airband #394: [BUG]

kiricheglov, 2023-07-11:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit): 4.1.1
  - `make` options used to build the program:
  - Hardware platform (eg. x86_64, Raspberry Pi v4): RPi v4
  - Operating system name and version: 
Debian 11, kernel 6.1.21
**What happened?**
I wanted to record the sound of a radio at a frequency of 102.9 MHz with modulation NFM, but the script instead of the speaker's voice records incomprehensible noise, the frequency and modulation are exactly correct
**What you expected to happen?**
I expected in .mp3 file hear the speaker's voice
**Steps to Reproduce**
sudo rtl_airband -f -c rtl_airband.conf
**Additional context**
Add any other relevant information about the problem here.

**Your rtl_airband.conf file**
Remove passwords, server addresses and other private information.

devices: (
  {
  type = "rtlsdr";
  index = 0;
  gain = 38.6
  centerfreq = 125.0
    channels: (
      {
        freq = 124.0;
        modulation = "am";
        outputs: (
          {
            type = "file";
            directory = "/home/rootmalina";
            filename_template = "channel_1";
            append = true;
          }
        );
      },
      {
        freq = 102.9;
        modulation = "nfm";
        outputs: (
          {
            type = "file";
            directory = "/home/rootmalina";
            filename_template = "channel_2";
            append = true;
          }
        );
      },
      {
        freq = 129.5;
        modulation = "am";
        outputs: (
          {
            type = "file";
            directory = "/home/rootmalina";
            filename_template = "channel_3";
            append = true;
          }
        );
      },
      {
        freq = 133.1;
        modulation = "am";
        outputs: (
          {
            type = "file";
            directory = "/home/rootmalina";
            filename_template = "channel_4";
            append = true;
          }
        );
      },
      {
        freq = 139.3;
        modulation = "am";
        outputs: (
          {
            type = "file";
            directory = "/home/rootmalina";
            filename_template = "channel_5";
            append = true;
          }
        );
      },
      {
        freq = 124.45;
        modulation = "am";
        outputs: (
          {
            type = "file";
            directory = "/home/rootmalina";
            filename_template = "channel_6";
            append = true;
          }
        );
      },
      {
        freq = 130.75;
        modulation = "am";
        outputs: (
          {
            type = "file";
            directory = "/home/rootmalina";
            filename_template = "channel_7";
            append = true;
          }
        );
      }
    );
  }
);

SullivanChrisJ, 2024-03-25:
102.9 MHz would be a commercial FM station, with a bandwidth of about 180 kHz. That isn't NFM (narrowband FM) which is more like 16 kHz.

gabeschine, 2025-02-27:
Haven't seen an update on this in almost year. Closing. Please re-open if necessary.

closed 2025-02-27 by gabeschine