rtl-airband/RTLSDR-Airband #402: [FEATURE] Minimum length in recording and max noise time

Klammraute, 2023-08-12:
Hi, thank you for your work, its great!
Could you please implement a minimum lenght option for recording?
have a lot of recordings shorter than 1 second. 
maybe something like:

"max_noise_time_ms": 500 
"min_time_ms": 2000

thanks

charlie-foxtrot, 2023-09-15:
Looks like this may be included in #411

Klammraute, 2023-09-17:
that are really good news - Can you tell me approximately when you release the new version? 
Thanks for all your work!

charlie-foxtrot, 2023-09-17:
at the moment release are manual and as-needed.  there are a few open PRs that I'm trying to land and do a release after that.

are you seeing the lots of recordings under 1 second using `split_on_transmission` or are they the hour-long recording files?

jammedjackal, 2024-07-24:
Hi All,

Is it possible to introduce this feature? It would be a life-saver for me. I'm experiencing a lot of very short calls due to random interferences, and elimination of them would be a highly appreciated functionality.