rtl-airband/RTLSDR-Airband #403: [FEATURE]

iu2frl, 2023-08-23:
**Is your feature request related to a problem? If so, please describe.**
This project works really good for AM or FM, would be nice to be able to record also SSB signals, this can be particularly useful to decode digital modes on amateur bands

**Describe the solution you'd like**
Software can receive USB or LSB channels and store them as WAV file. Bandwidth should be set by user (typical 2.7KHz)

**Describe alternative solutions or features you've considered**
Live decode of full IQ stream but it is extremely CPU costly

nox3000, 2024-05-11:
The single-sideband decoding feature is really needed, thank you!!!

SullivanChrisJ, 2025-02-20:
I would like to see this as well, although the chunked FFT approach which is the strength of Airband would mean that a second layer would be needed for closely spaced channels and for selecting the suppressed carrier frequency. I would settle for RAW output to a socket so that an external decoder (for SSB and digital including DMR, D-STAR, Fusion, etc.) could be attached.