rtl-airband/RTLSDR-Airband #410: [BUG] MP3 headers are malformed

D0han, 2023-09-05:
**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit): v4.1.1
  - `make` options used to build the program: cmake ../ && make
  - Hardware platform (eg. x86_64, Raspberry Pi v4): Raspberry Pi 3 Model B Rev 1.2
  - Operating system name and version: Debian GNU/Linux 11 (bullseye)

**What happened?**
Created MP3 files are malformed. In most cases audio is still usable, not sure if that is always the case.

**What you expected to happen?**
Fully standard and proper MP3 file

**Steps to Reproduce**
Use Airband to save radio traffic to mp3

**Additional context**
Code
```python
import librosa
audio_data, sample_rate = librosa.load("airband-created-file.mp3", mono=True)
```
output example
```
Note: Illegal Audio-MPEG-Header 0x4cb44fd5 at offset 119664.
Note: Trying to resync...
Note: Skipped 8 bytes in input.
Note: Illegal Audio-MPEG-Header 0x2c66b66d at offset 182168.
Note: Trying to resync...
Note: Skipped 23 bytes in input.
Note: Illegal Audio-MPEG-Header 0x55555555 at offset 186151.
Note: Trying to resync...
Note: Skipped 24 bytes in input.
```
Exact output varies.  
This is not a new issue, saw the same in old versions from 6-12 months ago.

**Your rtl_airband.conf file**
```
devices:
({
  type = "rtlsdr";
  index = 0;
  serial = "00000001";
  gain = 35;
  mode = "multichannel";
  multiple_demod_threads = true;
  centerfreq = 149.5;
  correction = 36;
  channels:
  (
    {
      freq = 148.825000;
      modulation = "nfm";
      outputs: (
        {
        type = "file";
        directory = "/home/airband/saved";
        filename_template = "ASD";
        split_on_transmission = true;
        include_freq = true;
        }
      );
    }
  );
 }
);
```

charlie-foxtrot, 2023-09-15:
Yes, I have noticed this as well but haven't looked into it because the audio seems fine / hasn't been a priority to look into.

Is this causing issues?  If so, I wonder if the LAME parameter changes in #411 have any impact

FWIW I ran into this using both `ffmpeg` as well as trying to open files in Audacity (I used the work around [here](https://manual.audacityteam.org/man/opening_malformed_mp3_files.html))

D0han, 2023-09-15:
Not sure if that causes data loss or not, i do not have parallel samples from another source to compare.  
If you can do first review pass on #411 to confirm the code is doing what we expect, i can try deploying it to see if that fixes the mp3 structure.

dseven, 2023-10-21:
I've been grappling with an issue that may or may not be related to this. RTLSDR-Airband is capturing ATC traffic to mp3 files that rdio-scanner picks up (via dirwatch). The problem is that some of the recordings stop playback via rdio-scanner part-way through, and some don't play at all. Perhaps interestingly, the ones that stop in the middle always seems to do so at the point where there's an "over" in the conversation - i.e. ATC stops talking, then the aircraft starts, or vice-versa. If I download the mp3 from rdio-scanner, and play it with any other player, all of the audio is there. If I import the mp3 into audacity, re-export it as mp3, and drop the result in the dirwatch directory, rdio-scanner plays it OK. I tried applying the proposed "MP3 Quality" changes from #414, but that didn't fix it. I thought it might be something to do with VBR and periods of silence, but apparently not. My solution, for now, is to have rdio-scanner transcode the mp3 to m4a, which seems to work.

D0han, 2023-11-03:
I checked out recordings, and a lot of them cuts off words. Having said that, i do not have any concrete arguments to tie it up with malformed MP3 headers.

coatebra, 2023-12-05:
Most of recorded mp3 files are not playable directly from a web server using html5 <audio> component. When I tried to convert mp3 to ogg using ffmpeg, I got a lot of errors: 

Header missing=N/A speed=   0x    
Error while decoding stream #0:0: Invalid data found when processing input

zigsphere, 2024-05-03:
Following - I really think this is causing many of the issue Im having. Since the headers are missing, icecast and perhaps the client is unaware what media format the audio is. With that being the case, there is a delay in the feed. 

Can this be fixed?

krbaker, 2024-06-21:
I'm also seeing this and curious to see where it goes.  I'm doing speech-to-text on Marine VHF and I upload the results to a Slack channel along with the audio.  90% of the time Slack doesn't recognize this as an audio file.

ffplay gets results like: 
[mp3 @ 0x75bbc4000cc0] Format mp3 detected only with low score of 1, misdetection possible!
[mp3 @ 0x75bbc4000cc0] Skipping 19 bytes of junk at 0.
[mp3 @ 0x75bbc4000cc0] Estimating duration from bitrate, this may be inaccurate
Input #0, mp3, from 'vhf-78a/recording_20240614_125529.mp3':
  Duration: 00:00:10.56, start: 0.000000, bitrate: 8 kb/s
  Stream #0:0: Audio: mp3, 8000 Hz, mono, fltp, 8 kb/s
[mp3float @ 0x75bbc4006a80] Header missing    0KB sq=    0B f=0/0   
[mp3float @ 0x75bbc4006a80] Header missing    0KB sq=    0B f=0/0   
[mp3float @ 0x75bbc4006a80] Header missing    0KB sq=    0B f=0/0   
   4.68 M-A:  0.000 fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0   

I'm in file split mode but it still seems to match to moments of discontinuity (changing speakers where squelch doesn't actually have time to close)

rough316, 2024-07-13:
Ugh, this just bit me too. I was trying to get mp3 for each channel in scan mode so I can process those and get a transcription... Malformed header on every mp3.

D0han, 2024-07-13:
https://github.com/charlie-foxtrot/RTLSDR-Airband/pull/411 was abandoned to the point of branch deletion.  
@charlie-foxtrot seems to not use github at all since February, which is abnormal in comparison to previous months. I think that this project lost its maintainer.

There are few forks visible in https://github.com/charlie-foxtrot/RTLSDR-Airband/network but i did not check if anyone did pick this up already in full capacity.

charlie-foxtrot, 2024-07-13:
I have unfortunately had to take a step back. I do intend to return but things are taking longer than I expected.

D0han, 2024-07-14:
Glad you're still here

krbaker, 2024-08-05:
@charlie-foxtrot I hope all is well, great to know you'll be back :-)

ljkenny, 2024-08-20:
A little more info after tripping over this myself.

Google Drive can only directly play a sub-set (the non-LAME3 ones) of the saved files.

Here is a `file` and `hexdump` of each file:

```
$ for f in `find . | grep mp3`; do file ${f}; hexdump -C ${f} | head -n1; echo; done
./mcr_20240819_16.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 55 55 55 55 55 55 55  |LAME3.100UUUUUUU|

./mcr_20240819_17.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 aa aa aa aa aa aa aa  |LAME3.100.......|

./mcr_20240819_18.mp3: MPEG ADTS, layer III,  v2.5,   8 kbps, 8 kHz, Monaural
00000000  ff e3 18 c4 00 00 00 03  48 01 40 00 00 ff e3 bb  |........H.@.....|

./mcr_20240819_19.mp3: MPEG ADTS, layer III,  v2.5,   8 kbps, 8 kHz, Monaural
00000000  ff e3 18 c4 00 00 00 03  fc 01 40 00 00 66 86 62  |..........@..f.b|

./mcr_20240819_20.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 55 55 55 55 55 55 55  |LAME3.100UUUUUUU|

./mcr_20240819_21.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 aa aa aa aa aa aa aa  |LAME3.100.......|

./mcr_20240819_22.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 55 55 55 55 55 55 55  |LAME3.100UUUUUUU|

./mcr_20240819_23.mp3: MPEG ADTS, layer III,  v2.5,   8 kbps, 8 kHz, Monaural
00000000  ff e3 18 c4 00 00 00 03  48 01 40 00 00 03 fe 3f  |........H.@....?|

./mcr_20240820_00.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 aa aa aa aa aa aa aa  |LAME3.100.......|

./mcr_20240820_01.mp3: MPEG ADTS, layer III,  v2.5,   8 kbps, 8 kHz, Monaural
00000000  ff e3 18 c4 00 00 00 03  48 01 40 00 00 ff ff ff  |........H.@.....|

./mcr_20240820_02.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 aa aa aa aa aa aa aa  |LAME3.100.......|

./mcr_20240820_03.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 aa aa aa aa aa aa aa  |LAME3.100.......|

./mcr_20240820_04.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 55 55 55 55 55 55 55  |LAME3.100UUUUUUU|

./mcr_20240820_05.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 55 55 55 55 55 55 55  |LAME3.100UUUUUUU|

./mcr_20240820_06.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 aa aa aa aa aa aa aa  |LAME3.100.......|

./mcr_20240820_07.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 55 55 55 55 55 55 55  |LAME3.100UUUUUUU|

./mcr_20240820_08.mp3: data
00000000  4c 41 4d 45 33 2e 31 30  30 55 55 55 55 55 55 55  |LAME3.100UUUUUUU|
```

The ones identified as MPEG play fine naively in Google Drive.

However the LAME3 ones are not identified as playable and `file` only IDs them as data.

Downloading the LAME3 ones and playing them on an Android device works fine, however.

SullivanChrisJ, 2025-02-20:
PySoundFile complains about every file.
```
Note: Skipped 26 bytes in input.
Note: Illegal Audio-MPEG-Header 0xaaaaaaaa at offset 29539.
Note: Trying to resync...
```
and occasionally a message like this one
```
[src/libmpg123/layer3.c:INT123_do_layer3():1774] error: part2_3_length (1408) too large for available bit count (1048)
```
And it sometime decodes the file as having no audio above -50dB even though Audacity shows peaks at -dB. I haven't seen any problems with Audacity though.

PyAv chokes on the file
```
>>> audio, sample = read_mp3.read_mono_mp3_to_float32_array("/media/ve3nrt/2025/02/VE3YRA_TX_20250214_113506.mp3")
Traceback (most recent call last):
  File "<python-input-2>", line 1, in <module>
    audio, sample = read_mp3.read_mono_mp3_to_float32_array("/media/ve3nrt/2025/02/VE3YRA_TX_20250214_113506.mp3")
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chris/projects/dragnet/src/dragnet/read_mp3.py", line 33, in read_mono_mp3_to_float32_array
    for frame in packet.decode():
                 ~~~~~~~~~~~~~^^
  File "av/packet.pyx", line 81, in av.packet.Packet.decode
  File "av/audio/stream.pyx", line 34, in av.audio.stream.AudioStream.decode
  File "av/audio/stream.pyx", line 43, in av.audio.stream.AudioStream.decode
  File "av/codec/context.pyx", line 448, in av.codec.context.CodecContext.decode
  File "av/codec/context.pyx", line 340, in av.codec.context.CodecContext._send_packet_and_recv
  File "av/error.pyx", line 428, in av.error.err_check
av.error.InvalidDataError: [Errno 1094995529] Invalid data found when processing input
```

While ffprobe is also unhappy.
```
[mp3 @ 0x5737f240fa40] Format mp3 detected only with low score of 25, misdetection possible!
[mp3 @ 0x5737f240fa40] Skipping 19 bytes of junk at 0.
[mp3 @ 0x5737f240fa40] Estimating duration from bitrate, this may be inaccurate
Input #0, mp3, from '/media/ve3nrt/2025/02/VE3YRA_TX_20250214_113506.mp3':
  Duration: 00:00:04.22, start: 0.000000, bitrate: 14 kb/s
  Stream #0:0: Audio: mp3 (mp3float), 8000 Hz, mono, fltp, 14 kb/s
```

Most files decode OK but occasionally one doesn't. This seems to correlate to when another channel is active producing RAW output and while I can't say for sure it is the cause it appears to happen more often than chance.

All files are FM decodes. If you need some further testing I can try to do it. I have over 100,000 files to test from. 

I'm going to try the MAD MP3 library I think. I'm very disappointed that libavcodec didn't work. MAD might work because Audacity does, but it is concerning that others have had issues with it.

The receiver is on a Raspberry Pi, the libraries are all a bit out of date, so maybe that has something to do with it, but I need to get the existing files working as well as thinking about the future.

SullivanChrisJ, 2025-02-22:
I had no luck with MAD, but got PyAV to work by ignoring errors and inserting silence where the glitches are, which are all 576 bytes in my mono 8kHz mp3 files.  The Python program I'm using to convert the files to NumPy arrays outputs messages like this.
```
Packet 3 decode error: [Errno 1094995529] Invalid data found when processing input forcing 576 zero samples at position 0.216
Packet 4 decode error: [Errno 1094995529] Invalid data found when processing input forcing 576 zero samples at position 0.288
Packet 54 decode error: [Errno 1094995529] Invalid data found when processing input forcing 576 zero samples at position 3.888
```
The first two errors are almost always at the same position (it is in seconds). The others are usually spaced out a bit.

gabeschine, 2025-02-22:
@SullivanChrisJ thank you for the investigation. I recently also encountered what I assume is related to this issue, where VLC was able to play the mp3 but the "time remaining" counter was inaccurate and kept changing / jumping.

Is there any chance you are willing to give https://github.com/charlie-foxtrot/RTLSDR-Airband/pull/494 a test-run to see if it resolves any of the issues you are seeing? It's a bit of a shot in the dark, but the PR author reports that inserting silence helped in their case.

There's a strong possibility that the way we are using the LAME library is incorrect and results in invalid data. I'm not an expert with LAME by any means, but if someone here is it might be worth a quick survey of usage.

SullivanChrisJ, 2025-02-22:
@gabeschine I'm not sure what you're asking for. Do you mean that the first four octets be removed and 0xFF be inserted in its place? I can do that but would like to make sure that's what you want.

Chris

gabeschine, 2025-02-23:
@SullivanChrisJ no sorry for being unclear. I was just asking if you'd try patching in that PR to see if it fixes some of the invalid MP3 encoding, but I just saw the author's response that they don't expect it to help beyond the issue of the initial headers being malformed.

What would be the most helpful to you in your further debugging of the later artifacts?

SullivanChrisJ, 2025-02-24:
@gabeschine So far I haven't thought of any easy way to dig into it. Maybe looking at the code path in the decoder library that causes the error to see what conditions trigger the errors. That sounds like a full day's work to me. The semi-random glitchy nature of the problem makes me think of buffer overruns and other hard to debug problems, but the relative consistency of it occurring at packets 3 and 4 might be good place to start. Another thing would be to work through a dump of the file, or analyze it with an off-the-shelf or  custom program, to try to figure out where it deviates from the spec. 

LAME seems to be widely used and it would be surprising if it is problem with the library itself, although it may be an obscure outlier issue that no-one else has exercised.

krbaker, 2025-02-24:
@SullivanChrisJ I think the way this code is using the LAME encoder isn't valid.  If I'm reading the code here correctly it is using a different instance of the LAME encoder on the same stream (when it inserts silence and tones).  If I'm reading the encoder correctly it has stored state that is required for the next frame(s).  Appending one encoders output to anothers mid stream doesn't work ( The 'why can't I copy-paste mp3s' section of https://lame.sourceforge.io/tech-FAQ.txt has a lot of detail that seems relevant. )

If I'm not mistaken the code creates a brand new instance of lame for encoding silence and the inserted tones. If I read correctly lame_encode_flush_nogap (used to close the files) can make a set of validly concatenate-able files but only if you continue with that instance
-- some PCM data can/is often still in the lame buffers after the flush nogap call https://android.googlesource.com/platform/external/lame/+/master/include/lame.h

it was a while back when I did this digging, so apologias if I have it wrong in my head.

SullivanChrisJ, 2025-02-25:
@krbaker thank you for the tip. I have some other things to get done before I can get back to it. I took a quick look at that code but not knowing LAME it wasn't something I'd notice, so I'm glad for your insight.

SullivanChrisJ, 2025-02-25:
@gabeschine Wonderful! I'm running it with 10 channels (4 MP3, 4 IQ, 2 socket) 24 hours a day, and can test a full day's recordings (about 4-5 hours and 500-1000 mp3 files) with a single command, so I'd be delighted to test it.

gabeschine, 2025-02-27:
Would you please test this PR: https://github.com/charlie-foxtrot/RTLSDR-Airband/pull/499 ?

I've done only the smallest amount of testing so far, and will be deploying it to my scanner box today - just wanted to get it out here for you to bang on as well.

D0han, 2025-03-13:
I have updated my install to 5.1.0, and while i can confirm that there is very much difference in terms of mp3 header errors, the cutoff on words i mentioned in one of the comments still occurs. No idea if that is a bug with how mp3 is created, detection of transmission or what else.

gabeschine, 2025-03-24:
> I have updated my install to 5.1.0, and while i can confirm that there is very much difference in terms of mp3 header errors, the cutoff on words i mentioned in one of the comments still occurs. No idea if that is a bug with how mp3 is created, detection of transmission or what else.

Please file a new issue and include as much information as you can about your configuration, what you've observed, and so on.