rtl-airband/RTLSDR-Airband #417: [BUG] unstable requires libshout version not available in APT on Raspberry PI (buster 10)

rough316, 2023-09-25:
TLDR: Newer version of libshout needed with unstable, had to build from source

**Describe your environment**

  - RTLSDR-Airband version you are using (stable release number or branch/commit): unstable
  - `make` options used to build the program: cmake ../ -DPLATFORM=rpiv2 -DNFM=ON 
  - Hardware platform (eg. x86_64, Raspberry Pi v4): Raspberry pi 3B 
  - Operating system name and version:  Raspbian / Buster 10

**What happened?**

Sep 24 11:46:03 raspberrypi rtl_airband[25004]: /usr/local/bin/rtl_airband: symbol lookup error: /usr/local/bin/rtl_airband: undefined symbol: shout_set_content_format

**What you expected to happen?**

No error

**Steps to Reproduce**

Build unstable on pi 3B / buster10

**Additional context**

Here is my OS version:

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

charlie-foxtrot, 2023-09-30:
Confirmed `libshout3-dev` version `2.4.1-2` is the latest available via `apt`.  Requirement was set as `v2.4.3` or later as part of #382, will looks to support at least `2.4.1` or later

charlie-foxtrot, 2023-09-30:
@rough316 can you try #422?

gabeschine, 2025-02-27:
Haven't seen an update on this in over a year. Closing.

closed 2025-02-27 by gabeschine