rtl-airband/RTLSDR-Airband #424: [compile error] error: ‘int shout_set_format(shout_t*, unsigned int)’ is deprecated

yo2llq, 2023-10-13:
**Rpi3B+/raspiOS lite 32bit**

  - RTLSDR-Airband version 4.1.1:
  - `make` options used to build the program: cmake -DPLATFORM=rpiv2 ../
  
**Compilation error**
`dan@rpi-adsb:~/rtlsdrairband/RTLSDR-Airband-4.1.1/build $ make
[  4%] Generating version.cpp, _version.cpp
[  8%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/config.cpp.o
[ 13%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-common.cpp.o
[ 17%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-file.cpp.o
[ 21%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-helpers.cpp.o
[ 26%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/mixer.cpp.o
[ 30%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/output.cpp.o
/home/dan/rtlsdrairband/RTLSDR-Airband-4.1.1/src/output.cpp: In function ‘void shout_setup(icecast_data*, mix_modes)’:
/home/dan/rtlsdrairband/RTLSDR-Airband-4.1.1/src/output.cpp:81:29: error: ‘int shout_set_format(shout_t*, unsigned int)’ is deprecated: Use shout_set_content_format() [-Werror=deprecated-declarations]
   81 |         if (shout_set_format(shouttemp, SHOUT_FORMAT_MP3) != SHOUTERR_SUCCESS){
      |             ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/dan/rtlsdrairband/RTLSDR-Airband-4.1.1/src/output.cpp:30:
/usr/include/shout/shout.h:361:5: note: declared here
  361 | int shout_set_format(shout_t *self, unsigned int format) SHOUT_ATTR_F_DEPRECATED("Use shout_set_content_format()"); // obsolete
      |     ^~~~~~~~~~~~~~~~
/home/dan/rtlsdrairband/RTLSDR-Airband-4.1.1/src/output.cpp:84:43: error: ‘int shout_set_name(shout_t*, const char*)’ is deprecated: use shout_set_meta() with SHOUT_META_NAME [-Werror=deprecated-declarations]
   84 |         if(icecast->name && shout_set_name(shouttemp, icecast->name) != SHOUTERR_SUCCESS) {
      |                             ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/shout/shout.h:348:5: note: declared here
  348 | int shout_set_name(shout_t *self, const char *name) SHOUT_ATTR_F_DEPRECATED("use shout_set_meta() with SHOUT_META_NAME"); // obsolete
      |     ^~~~~~~~~~~~~~
/home/dan/rtlsdrairband/RTLSDR-Airband-4.1.1/src/output.cpp:87:45: error: ‘int shout_set_genre(shout_t*, const char*)’ is deprecated: use shout_set_meta() with SHOUT_META_GENRE [-Werror=deprecated-declarations]
   87 |         if(icecast->genre && shout_set_genre(shouttemp, icecast->genre) != SHOUTERR_SUCCESS) {
      |                              ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/shout/shout.h:354:5: note: declared here
  354 | int shout_set_genre(shout_t *self, const char *genre) SHOUT_ATTR_F_DEPRECATED("use shout_set_meta() with SHOUT_META_GENRE"); // obsolete
      |     ^~~~~~~~~~~~~~~
/home/dan/rtlsdrairband/RTLSDR-Airband-4.1.1/src/output.cpp:90:57: error: ‘int shout_set_description(shout_t*, const char*)’ is deprecated: use shout_set_meta() with SHOUT_META_DESCRIPTION [-Werror=deprecated-declarations]
   90 |         if(icecast->description && shout_set_description(shouttemp, icecast->description) != SHOUTERR_SUCCESS) {
      |                                    ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/shout/shout.h:357:5: note: declared here
  357 | int shout_set_description(shout_t *self, const char *description) SHOUT_ATTR_F_DEPRECATED("use shout_set_meta() with SHOUT_META_DESCRIPTION"); // obsolete
      |     ^~~~~~~~~~~~~~~~~~~~~
/home/dan/rtlsdrairband/RTLSDR-Airband-4.1.1/src/output.cpp: In function ‘void process_outputs(channel_t*, int)’:
/home/dan/rtlsdrairband/RTLSDR-Airband-4.1.1/src/output.cpp:479:55: error: ‘int shout_set_metadata(shout_t*, shout_metadata_t*)’ is deprecated: Use shout_set_metadata_utf8() [-Werror=deprecated-declarations]
  479 |                                 if (shout_set_metadata(icecast->shout, meta) != SHOUTERR_SUCCESS) {
      |                                     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/usr/include/shout/shout.h:367:5: note: declared here
  367 | int shout_set_metadata(shout_t *self, shout_metadata_t *metadata) SHOUT_ATTR_F_WARN_UNUSED_RESULT SHOUT_ATTR_F_DEPRECATED("Use shout_set_metadata_utf8()");
      |     ^~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [src/CMakeFiles/rtl_airband_base.dir/build.make:153: src/CMakeFiles/rtl_airband_base.dir/output.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:117: src/CMakeFiles/rtl_airband_base.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
`

**Steps to Reproduce**
Try compiling per wiki instructions.

**Additional context**
A RPi setup with readsb/tar1090 running already.

**Your rtl_airband.conf file**
Didn't get there yet.

What did I do wrong?

Thanks!

yo2llq, 2023-10-13:
Ok, so it compiled after replacing the deprecated functions like so:

`        if (shout_set_content_format(shouttemp, SHOUT_FORMAT_MP3, SHOUT_USAGE_OTHER, "mp3") != SHOUTERR_SUCCESS){
                shout_free(shouttemp); return;
        }
        if(icecast->name && shout_set_meta(shouttemp, SHOUT_META_NAME, icecast->name) != SHOUTERR_SUCCESS) {
                shout_free(shouttemp); return;
        }
        if(icecast->genre && shout_set_meta(shouttemp, SHOUT_META_GENRE, icecast->genre) != SHOUTERR_SUCCESS) {
                shout_free(shouttemp); return;
        }
        if(icecast->description && shout_set_meta(shouttemp, SHOUT_META_DESCRIPTION, icecast->description) != SHOUTERR_SUCCESS) {
                shout_free(shouttemp); return;
....
        if (shout_set_metadata_utf8(icecast->shout, meta) != SHOUTERR_SUCCESS) {
                                        log(LOG_WARNING, "Failed to add shout metadata\n");

`
I'll have to see if it actually runs then.

yo2llq, 2023-10-13:
Below you have some outputs:

dan@rpi-adsb:~/rtlsdrairband/RTLSDR-Airband-4.1.1/build $ cmake -DPLATFORM=rpiv2 ../
-- Build type not specified: defaulting to Release
-- Found lame includes:	/usr/include/lame/lame.h
-- Found lame library: /usr/lib/arm-linux-gnueabihf/libmp3lame.so
-- Checking for module 'libmirisdr'
--   Package 'libmirisdr', required by 'virtual:world', not found
-- Could NOT find MiriSDR (missing: MIRISDR_LIBRARY MIRISDR_INCLUDE_DIR) 
-- Checking for SoapySDR
-- Could NOT find SoapySDR (missing: SoapySDR_DIR)
--   SoapySDR not found
-- Checking for module 'libpulse'
--   Package 'libpulse', required by 'virtual:world', not found
-- RTLSDR-Airband configuration summary:

-- - Build type:		Release
-- - Operating system:		Linux
-- - SDR drivers:
--   - librtlsdr:		requested: ON, enabled: TRUE
--   - mirisdr:			requested: ON, enabled: FALSE
--   - soapysdr:		requested: ON, enabled: FALSE
-- - Other options:
--   - Platform:		rpiv2
--   - Build Unit Tests:	FALSE
--   - Broadcom VideoCore GPU:	TRUE
--   - NFM support:		OFF
--   - PulseAudio:		requested: ON, enabled: FALSE
--   - Profiling:		requested: OFF, enabled: FALSE
--   - Icecast TLS support:	TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dan/rtlsdrairband/RTLSDR-Airband-4.1.1/build

dan@rpi-adsb:~/rtlsdrairband/RTLSDR-Airband-4.1.1/build $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

dan@rpi-adsb:~/rtlsdrairband/RTLSDR-Airband-4.1.1/build $ uname -a
Linux rpi-adsb 6.1.0-rpi4-rpi-v7 #1 SMP Raspbian 1:6.1.54-1+rpt2 (2023-10-05) armv7l GNU/Linux

Kinda in a hurry now, will try unstable later maybe.

charlie-foxtrot, 2023-10-14:
`main` is now `v4.2.0` that includes #382

closed 2023-10-14 by charlie-foxtrot

yo2llq, 2023-10-14:
Yep, 4.2.0 compiles now.

Thank you!