rtl-airband/RTLSDR-Airband #427: [FEATURE] Change the I/Q sample format for the rawfile output to a more standard one

LZ2DMV, 2023-10-16:
The current complex float 32-bit, little endian format that rtl_airband uses when configured to save raw I/Q data files differs from what appears to be the established standard in the RTL-SDR community for I/Q sample format - 48k/1 wav files, like the ones generated from GQRX and SDR++.

The DSP processing (especially things like the de-emphasis) of csdr is sometimes flawed and i think we could say it is complicated to arrange the right pipeline without sufficient knowledge in the DSP domain, so getting an useful audio is not always trivial with it, say, when we need high precision (digital modes).

Therefore, i think it will be best if rtl_airband exports I/Q data in a more standard format, like the ones exported from SDR++, GQRX or SDRSharp - many tools support this I/Q format as an input source, so it will certainly be easier to work with them instead of the current format.

It could be added as an additional rawfile option or replacing the current one.

charlie-foxtrot, 2023-10-17:
The `rawfile` output is a dump of the internal data structure with no additional processing.

I'm not familiar with `48k/1 wav` files, but if the `48k` is referring to the sample rate, that will require upsampling, and I dont think it makes sense to do that in `rtl_airband`.

Instead you can use `sox` to convert the output `.cf32` files to any format you want.  Looking at the man pages, this command may be what you're looking for (and if so please let me know and I'll add to the wiki):

```
sox −r 16k −e float −b 32 -L −c 2 input.raw output.wav
```

LZ2DMV, 2023-10-17:
Hm, interesting.  `soxi` reports the expected format for the samples - 32-bit Floating Point PCM, however, when i loaded the output wav file in GQRX and other software that accepts it, it fails to work with it.

During conversion, `sox` reports:

```
sox WARN sox: `input.raw' input clipped 2906524 samples
sox WARN sox: `output.wav' output clipped 1454773 samples; decrease volume?
```

closed 2023-11-09 by LZ2DMV