rtl-airband/RTLSDR-Airband #429: Possible Bookworm cmake issue on a clean install [BUG]

LoganSound, 2023-10-23:
**Describe your environment**

  - RTLSDR-Airband 4.2.0
  - `make` options used to build the program: -DPLATFORM=rpiv2
  - Hardware platform (eg. x86_64, Raspberry Pi v4): Raspi 3B
  - Operating system name and version: Bookworm (64-bit)

**What happened?**
cmake -DPLATFORM=rpiv2 ..
-- Build type not specified: defaulting to Release
-- Checking for module 'libconfig++'
--   Package 'libconfig++', required by 'virtual:world', not found
CMake Error at /usr/share/cmake-3.25/Modules/FindPkgConfig.cmake:607 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.25/Modules/FindPkgConfig.cmake:829 (_pkg_check_modules_internal)
  src/CMakeLists.txt:54 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/pi/RTLSDR-Airband-4.2.0/build/CMakeFiles/CMakeOutput.log".
/home/pi/RTLSDR-Airband-4.2.0/build/CMakeFiles/CMakeOutput.log
The system is: Linux - 6.1.0-rpi4-rpi-v8 - aarch64
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/pi/RTLSDR-Airband-4.2.0/build/CMakeFiles/3.25.1/CompilerIdCXX/a.out"

Detecting CXX compiler ABI info compiled with the following output:
Change Dir: /home/pi/RTLSDR-Airband-4.2.0/build/CMakeFiles/CMakeScratch/TryCompile-Z4SCm1

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_15179/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_15179.dir/build.make CMakeFiles/cmTC_15179.dir/build
gmake[1]: Entering directory '/home/pi/RTLSDR-Airband-4.2.0/build/CMakeFiles/CMakeScratch/TryCompile-Z4SCm1'
Building CXX object CMakeFiles/cmTC_15179.dir/CMakeCXXCompilerABI.cpp.o
/usr/bin/c++   -v -o CMakeFiles/cmTC_15179.dir/CMakeCXXCompilerABI.cpp.o -c /usr/share/cmake-3.25/Modules/CMakeCXXCompilerABI.cpp
Using built-in specs.
COLLECT_GCC=/usr/bin/c++
Target: aarch64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 12.2.0-14' --with-bugurl=file:///usr/share/doc/gcc-12/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-12 --program-prefix=aarch64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --enable-fix-cortex-a53-843419 --disable-werror --enable-checking=release --build=aarch64-linux-gnu --host=aarch64-linux-gnu --target=aarch64-linux-gnu
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.2.0 (Debian 12.2.0-14) 
COLLECT_GCC_OPTIONS='-v' '-o' 'CMakeFiles/cmTC_15179.dir/CMakeCXXCompilerABI.cpp.o' '-c' '-shared-libgcc' '-mlittle-endian' '-mabi=lp64' '-dumpdir' 'CMakeFiles/cmTC_15179.dir/'
 /usr/lib/gcc/aarch64-linux-gnu/12/cc1plus -quiet -v -imultiarch aarch64-linux-gnu -D_GNU_SOURCE /usr/share/cmake-3.25/Modules/CMakeCXXCompilerABI.cpp -quiet -dumpdir CMakeFiles/cmTC_15179.dir/ -dumpbase CMakeCXXCompilerABI.cpp.cpp -dumpbase-ext .cpp -mlittle-endian -mabi=lp64 -version -fasynchronous-unwind-tables -o /tmp/ccqiLMxX.s
GNU C++17 (Debian 12.2.0-14) version 12.2.0 (aarch64-linux-gnu)
	compiled by GNU C version 12.2.0, GMP version 6.2.1, MPFR version 4.1.1-p1, MPC version 1.3.1, isl version isl-0.25-GMP

warning: MPFR header version 4.1.1-p1 differs from library version 4.2.0.
GGC heuristics: --param ggc-min-expand=92 --param ggc-min-heapsize=116285
ignoring duplicate directory "/usr/include/aarch64-linux-gnu/c++/12"
ignoring nonexistent directory "/usr/local/include/aarch64-linux-gnu"
ignoring nonexistent directory "/usr/lib/gcc/aarch64-linux-gnu/12/include-fixed"
ignoring nonexistent directory "/usr/lib/gcc/aarch64-linux-gnu/12/../../../../aarch64-linux-gnu/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/include/c++/12
 /usr/include/aarch64-linux-gnu/c++/12
 /usr/include/c++/12/backward
 /usr/lib/gcc/aarch64-linux-gnu/12/include
 /usr/local/include
 /usr/include/aarch64-linux-gnu
 /usr/include
End of search list.
GNU C++17 (Debian 12.2.0-14) version 12.2.0 (aarch64-linux-gnu)
[... 204 lines cut ...]
Change Dir: /home/pi/RTLSDR-Airband-4.2.0/build/CMakeFiles/CMakeScratch/TryCompile-Kd2lBp

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_70fcc/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_70fcc.dir/build.make CMakeFiles/cmTC_70fcc.dir/build
gmake[1]: Entering directory '/home/pi/RTLSDR-Airband-4.2.0/build/CMakeFiles/CMakeScratch/TryCompile-Kd2lBp'
Building CXX object CMakeFiles/cmTC_70fcc.dir/CheckSymbolExists.cxx.o
/usr/bin/c++ -D_GNU_SOURCE  -Wall -Wextra  -o CMakeFiles/cmTC_70fcc.dir/CheckSymbolExists.cxx.o -c /home/pi/RTLSDR-Airband-4.2.0/build/CMakeFiles/CMakeScratch/TryCompile-Kd2lBp/CheckSymbolExists.cxx
Linking CXX executable cmTC_70fcc
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_70fcc.dir/link.txt --verbose=1
/usr/bin/c++  -Wall -Wextra  -rdynamic CMakeFiles/cmTC_70fcc.dir/CheckSymbolExists.cxx.o -o cmTC_70fcc  -lm 
gmake[1]: Leaving directory '/home/pi/RTLSDR-Airband-4.2.0/build/CMakeFiles/CMakeScratch/TryCompile-Kd2lBp'

File CheckSymbolExists.cxx:
/* */
#include <math.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef sincosf
  return ((int*)(&sincosf))[argc];
#else
  (void)argc;
  return 0;
#endif
}

LoganSound, 2023-10-23:
I have tried sudo apt-get install libconfig++

charlie-foxtrot, 2023-10-23:
Try installing the dev package (read the notes carefully to get the quoting right), see https://github.com/charlie-foxtrot/RTLSDR-Airband/wiki/Dependencies#mandatory-dependencies

LoganSound, 2023-11-04:
I got through the cmake but now I get:
make
[  8%] Generating version.cpp, _version.cpp
[  8%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/config.cpp.o
c++: error: unrecognized command-line option ‘-mfpu=neon-vfpv4’
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
make[2]: *** [src/CMakeFiles/rtl_airband_base.dir/build.make:83: src/CMakeFiles/rtl_airband_base.dir/config.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:117: src/CMakeFiles/rtl_airband_base.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

charlie-foxtrot, 2023-11-04:
Remove the DPLATFORM flag to cmake

SullivanChrisJ, 2025-03-07:
Removing the `DPLATFORM` flag from cmake removes the Broadcom Videocore GPU support. Leaving it in causes the error on make. All the packages required by the wiki are installed. Raspberry Pi 3B Bookworm Lite 64 bit. I just found [this](https://forums.raspberrypi.com/viewtopic.php?t=332690) that states the flags only work on 32-bit, but are default on 64 bit, but will it compile the GPU support into RTL-SDR Airband if cmake states that support is FALSE?

It might help others to put something about this build issue in the Wiki, as it states that DPLATFORM=rpiv2 is required and says nothing about 64 bit OS.
```
ve3nrt@doppler:~/RTLSDR-Airband $ mkdir build
ve3nrt@doppler:~/RTLSDR-Airband $ cd build
ve3nrt@doppler:~/RTLSDR-Airband/build $ cmake -DNFM=ON ../
-- The CXX compiler identification is GNU 12.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to Release
-- Performing Test CXX_HAS_PTHREAD
-- Performing Test CXX_HAS_PTHREAD - Success
-- Performing Test CXX_HAS_FFAST_MATH
-- Performing Test CXX_HAS_FFAST_MATH - Success
-- Looking for sincosf
-- Looking for sincosf - found
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Checking for module 'libconfig++'
--   Found libconfig++, version 1.5
-- Found lame includes: /usr/include/lame/lame.h
-- Found lame library: /usr/lib/aarch64-linux-gnu/libmp3lame.so
-- Checking for module 'shout'
--   Found shout, version 2.4.6
-- Looking for SHOUT_TLS_AUTO
-- Looking for SHOUT_TLS_AUTO - found
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN - found
-- Looking for SHOUT_TLS_RFC2818
-- Looking for SHOUT_TLS_RFC2818 - found
-- Looking for SHOUT_TLS_RFC2817
-- Looking for SHOUT_TLS_RFC2817 - found
-- Looking for SHOUT_TLS_DISABLED
-- Looking for SHOUT_TLS_DISABLED - found
-- Looking for shout_set_tls
-- Looking for shout_set_tls - found
-- Looking for shout_set_content_format
-- Looking for shout_set_content_format - found
-- Looking for shout_set_metadata_utf8
-- Looking for shout_set_metadata_utf8 - found
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0
-- Found RTLSDR: /usr/lib/aarch64-linux-gnu/librtlsdr.so
-- Checking for module 'libmirisdr'
--   Package 'libmirisdr', required by 'virtual:world', not found
-- Could NOT find MiriSDR (missing: MIRISDR_LIBRARY MIRISDR_INCLUDE_DIR)
-- Checking for SoapySDR
--   SoapySDR found, /usr/local/include, SoapySDR
-- Checking for module 'libpulse'
--   Found libpulse, version 16.1
-- Performing Test CXX_HAS_MARCH_NATIVE
-- Performing Test CXX_HAS_MARCH_NATIVE - Success
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.10
-- RTLSDR-Airband configuration summary:

-- - Version string:            v5.0.12
-- - Build type:                Release
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: ON, enabled: FALSE
--   - soapysdr:                requested: ON, enabled: TRUE
-- - Other options:
--   - Platform:                native
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  FALSE
--   - NFM support:             ON
--   - PulseAudio:              requested: ON, enabled: TRUE
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- Configuring done
```
-- Generating done
-- Build files have been written to: /home/ve3nrt/RTLSDR-Airband/build
```

pcazador2021, 2025-03-27:
Greetings, I have been trying to install RTL-SDR AIRBAND for days but my attempts have been very unsuccessful.

All the installation goes well but when I get to      cmake -DPLATFORM=<platform_name>

My operating system is this       https://imgur.com/a/hQzWivA      

The installation seems to be going well    `https://imgur.com/Tcr7Qd3

pi@raspberrypi:~/RTLSDR-Airband $ git clone https://github.com/rtl-airband/RTLSDR-Airband.git
cd RTLSDR-Airband
git checkout main && git pull
Cloning into 'RTLSDR-Airband'...
remote: Enumerating objects: 2873, done.
remote: Counting objects: 100% (757/757), done.
remote: Compressing objects: 100% (151/151), done.
remote: Total 2873 (delta 681), reused 609 (delta 606), pack-reused 2116 (from 2)
Receiving objects: 100% (2873/2873), 1.19 MiB | 253.00 KiB/s, done.
Resolving deltas: 100% (1920/1920), done.
Already on 'main'
Your branch is up to date with 'origin/main'.
Already up to date.
pi@raspberrypi:~/RTLSDR-Airband/RTLSDR-Airband $ mkdir build
cd build
cmake ../
-- The CXX compiler identification is GNU 12.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to Release
-- Performing Test CXX_HAS_PTHREAD
-- Performing Test CXX_HAS_PTHREAD - Success
-- Performing Test CXX_HAS_FFAST_MATH
-- Performing Test CXX_HAS_FFAST_MATH - Success
-- Looking for sincosf
-- Looking for sincosf - found
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Checking for module 'libconfig++'
--   Found libconfig++, version 1.5
-- Found lame includes: /usr/include/lame/lame.h
-- Found lame library: /usr/lib/arm-linux-gnueabihf/libmp3lame.so
-- Checking for module 'shout'
--   Found shout, version 2.4.6
-- Looking for SHOUT_TLS_AUTO
-- Looking for SHOUT_TLS_AUTO - found
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN - found
-- Looking for SHOUT_TLS_RFC2818
-- Looking for SHOUT_TLS_RFC2818 - found
-- Looking for SHOUT_TLS_RFC2817
-- Looking for SHOUT_TLS_RFC2817 - found
-- Looking for SHOUT_TLS_DISABLED
-- Looking for SHOUT_TLS_DISABLED - found
-- Looking for shout_set_tls
-- Looking for shout_set_tls - found
-- Looking for shout_set_content_format
-- Looking for shout_set_content_format - found
-- Looking for shout_set_metadata_utf8
-- Looking for shout_set_metadata_utf8 - found
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0
-- Found RTLSDR: /usr/lib/arm-linux-gnueabihf/librtlsdr.so
-- Checking for module 'libmirisdr'
--   Package 'libmirisdr', required by 'virtual:world', not found
-- Could NOT find MiriSDR (missing: MIRISDR_LIBRARY MIRISDR_INCLUDE_DIR)
-- Checking for SoapySDR
-- Could NOT find SoapySDR (missing: SoapySDR_DIR)
--   SoapySDR not found
-- Checking for module 'libpulse'
--   Package 'libpulse', required by 'virtual:world', not found
-- Performing Test CXX_HAS_MARCH_NATIVE
-- Performing Test CXX_HAS_MARCH_NATIVE - Success
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.10
-- RTLSDR-Airband configuration summary:

-- - Version string:            v5.1.1
-- - Build type:                Release
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: ON, enabled: FALSE
--   - soapysdr:                requested: ON, enabled: FALSE
-- - Other options:
--   - Platform:                native
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  FALSE
--   - NFM support:             OFF
--   - PulseAudio:              requested: ON, enabled: FALSE
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/RTLSDR-Airband/RTLSDR-Airband/build
pi@raspberrypi:~/RTLSDR-Airband/RTLSDR-Airband/build $

've come this far and I'm not moving forward.

-- Installing: /usr/local/bin/rtl_airband
pi@raspberrypi:~/RTLSDR-Airband/build $ cmake -DPLATFORM=rpiv2
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/home/pi/RTLSDR-Airband/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
pi@raspberrypi:~/RTLSDR-Airband/build $

I have tried to search for step-by-step video tutorials but I cannot find how to do these installations.
`

SullivanChrisJ, 2025-03-27:
It look like the `cmake ../` process worked. Why do are you trying to run it again with `cmake -DPLATFORM=rpiv2`? The next step should be just `make` to build the executable, then `sudo make install` to install it.  Without the `..` at the end of the command cmake will not find the file it is complaining about.

Additionally, I see that the second time you run it you are using `-DPLATFORM=rpiv2`, which specifies that the Broadcom GPU FFT should be used instead of `libfftw3`. Is that what you want? Are you running on a Pi2 or Pi3? If so, make sure that the support for the GPU is installed (`sudo apt-get install libraspberrypi-dev`). 

Lastly, I've had issues on the Pi 3 with the 64 bit operating system when trying to use the GPU support. The 32 bit OS is fine.

pcazador2021, 2025-03-27:
Greetings friend, thanks for responding, 

Raspberry Pi 3 Model B Rev 1.2pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a52082
Serial          : 00000000788482dd
Model           : Raspberry Pi 3 Model B Rev 1.2

----------------------------------------------------
My system 

Operating System: Raspbian GNU/Linux 12 (bookworm)
 Kernel: Linux 6.6.74+rpt-rpi-v7
 Architecture: arm

-----------------------------------------------------

I have little knowledge of Raspberry, but I would like to learn more. If you have any tutorials, I would appreciate it. I want to set up my airband frequency.

pcazador2021, 2025-03-27:
installed system
system I have installed

Raspberry Pi OS Lite
Release date: November 19th 2024
System: 32-bit
Kernel version: 6.6
Debian version: 12 (bookworm)
Size: 508MB

SullivanChrisJ, 2025-03-27:
After you run `cmake ../`, which you have already done, run `make` from the build directory and let us know what happens.

pcazador2021, 2025-03-29:
Hello, I started formatting my Raspberry Pi from scratch, but I'm still stuck on the cmake -DPLATFORM=<platform_name> command. 
What am I doing wrong? Can you help me?

mkdir build
cd build
cmake ../

-------------------------------------------------------
--   Found fftw3f, version 3.3.10
-- RTLSDR-Airband configuration summary:

-- - Version string:            v5.1.1
-- - Build type:                Release
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: ON, enabled: FALSE
--   - soapysdr:                requested: ON, enabled: FALSE
-- - Other options:
--   - Platform:                native
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  FALSE
--   - NFM support:             OFF
--   - PulseAudio:              requested: ON, enabled: TRUE
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/RTLSDR-Airband/build

pi@raspberrypi:~/RTLSDR-Airband/build $ make
sudo make install
[  5%] Generating version.cpp, _version.cpp
[ 10%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/config.cpp.o
[ 15%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-common.cpp.o
[ 21%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-file.cpp.o
[ 26%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-helpers.cpp.o
[ 31%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/mixer.cpp.o
[ 36%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/output.cpp.o
[ 42%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/rtl_airband.cpp.o
[ 47%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/squelch.cpp.o
[ 52%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/ctcss.cpp.o
[ 57%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/util.cpp.o
[ 63%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/udp_stream.cpp.o
[ 68%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/logging.cpp.o
[ 73%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/filters.cpp.o
[ 78%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/helper_functions.cpp.o
[ 78%] Generating version.cpp, _version.cpp
[ 84%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/version.cpp.o
[ 89%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-rtlsdr.cpp.o
[ 94%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/pulse.cpp.o
[ 94%] Built target rtl_airband_base
[100%] Linking CXX executable rtl_airband
[100%] Built target rtl_airband
[  5%] Generating version.cpp, _version.cpp
[ 10%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/version.cpp.o
[ 94%] Built target rtl_airband_base
[100%] Linking CXX executable rtl_airband
[100%] Built target rtl_airband

Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/rtl_airband

pi@raspberrypi:~/RTLSDR-Airband/build $ cmake -DPLATFORM=native
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/home/pi/RTLSDR-Airband/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
pi@raspberrypi:~/RTLSDR-Airband/build $

https://imgur.com/a/Wd7nQtz

SullivanChrisJ, 2025-03-29:
You have successfully built and installed it. Then you ran cmake again from the build directory without using `../` at the end of the command, which is why it generates the error.  The `..` tells cmake to look for the file `CMakeLists.txt` in the parent directory `/home/pi/RTLSDR-Airband`, not in the current directory `/home/pi/RTLSDR-Airband/build`.

If you need to run cmake again with different parameters, use something like `cmake -DPLATFORM=native ../`

`-DPLATFORM=native` is the default, so there's no reason to add that command.

However, if you are using a Pi3B, the appropriate command is `cmake -DPLATFORM=rpiv2 ../` to get the GPU support, assuming you have installed the `libraspberrypi-dev` package as it says in the instructions.

pcazador2021, 2025-03-29:
Thanks for your help. I continue to move forward like a school child. The problem is in the guide.  https://github.com/rtl-airband/RTLSDR-Airband/wiki/Installation     gives as an example but omits the periods at the end of the command     **cmake -DPLATFORM=<platform_name>**         

https://imgur.com/EcDG16u

This has been my result advancing in my installation now what step do I have to follow since the guide in this step ends here my air frequencies will be 118-136 what step should I follow?

https://imgur.com/8mIOOx7

pi@raspberrypi:~/RTLSDR-Airband/build $ cmake -DPLATFORM=rpiv2 ../
-- Build type not specified: defaulting to Release
-- Found lame includes: /usr/include/lame/lame.h
-- Found lame library: /usr/lib/arm-linux-gnueabihf/libmp3lame.so
-- Checking for module 'libmirisdr'
--   Package 'libmirisdr', required by 'virtual:world', not found
-- Could NOT find MiriSDR (missing: MIRISDR_LIBRARY MIRISDR_INCLUDE_DIR)
-- Checking for SoapySDR
-- Could NOT find SoapySDR (missing: SoapySDR_DIR)
--   SoapySDR not found
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/c++
-- Found BCM_VC: /usr/lib/arm-linux-gnueabihf/libbcm_host.so
-- RTLSDR-Airband configuration summary:

-- - Version string:            v5.1.1
-- - Build type:                Release
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: ON, enabled: FALSE
--   - soapysdr:                requested: ON, enabled: FALSE
-- - Other options:
--   - Platform:                rpiv2
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  TRUE
--   - NFM support:             OFF
--   - PulseAudio:              requested: ON, enabled: TRUE
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/RTLSDR-Airband/build
pi@raspberrypi:~/RTLSDR-Airband/build $

SullivanChrisJ, 2025-03-29:
I did not notice that, but you are right. The wiki is in error. It is also a little misleading that the build options comes after the Building the Program section, and I see how you might think it is in addition to, not instead of, the previous cmake command. You are on the right track. Now that you have run cmake successfully, all you need to do to install the program is:
```
make
sudo make install
```
Once you have done that, you can continue with the Wiki instructions starting with "Configuration Essentials:. Be aware that the syntax is quite strict. Make sure you don't mix up `{` with `(`, or miss a semi-colon. The configuration file `/usr/local/etc/rtl_airband.conf` will contain the frequency and mode.

By the way, if you want to receive NFM (e.g. Marine Radio or Ham Radio repeaters) you probably want to re-run cmake with 
```
cmake -DPLATORM=rpiv2  -DNFM=ON ../
``` 
before doing the `make` & `sudo make install`.

Good luck!

pcazador2021, 2025-03-29:
Greetings, I have already configured Ice Cast 2 and it is already working on my receiver. I have also configured the frequencies in sudo nano /usr/local/etc/rtl_airband.conf   

ICE CAST2
https://imgur.com/w6eIt7P

editing my frequencies to use

https://imgur.com/RrPTb8q

Servidor listo 
https://imgur.com/d4ID1zN

but when I test my SDR dongle with the command 

sudo rtl_airband -f

https://imgur.com/K4yK5Nm

Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Allocating 10 zero-copy buffers

SullivanChrisJ, 2025-03-30:
If it is working then ignore the PLL not locked message. To program an RTL-SDR there are a lot of steps, and depending on the order in which they're executed you may get the message. In that case it is harmless.

pcazador2021, 2025-03-30:
Greetings again, I'm still stuck here. I see that the guide doesn't explain step by step how to continue with the configuration. I have a question: how to configure the output.   

https://imgur.com/ZwYQzFz

where do I edit or enter this Syntax  I have consulted different guides on Icecast but there is no practical explanation for beginners.  

outputs: (
  {
    type = "icecast";
    server = "icecast.server.example.org";
    port = 8080;
    mountpoint = "TWR.mp3";
    username = "source";
    password = "mypassword";
#   tls = "auto|auto_no_plain|transport|upgrade|disabled";
#   name = "Tower";
#   description = "This is a Tower feed from my local airport";
#   genre = "ATC";
  }
);

Check the mount status and it is blank. I apologize, but if you need your help, how do I edit and configure the mount?

https://imgur.com/gypuBiY

https://imgur.com/MuLZNUi