rtl-airband/RTLSDR-Airband #468: Gain Error Failed to initialize gain for device #-536870912: error -1071395636

mattfox27, 2024-03-11:
When trying to run as a service I get this error

`Failed to initialize gain for device #-536870912: error -1071395636
`

When I check the status of the service

nrussell88, 2024-03-14:
Hi @mattfox27, can you share your config file?

mattfox27, 2024-03-15:
`
devices:
({
type = "rtlsdr";
index = 0;
gain = 32;    #Gain Setting
correction = 0;   #ppm
mode = "scan";
channels:
(
{
freqs = ( 123.1, 119.0, 120.5, 132.2, 120.2, 119.3, 135.3, 124.3, 128.375, 119.>
outputs: (
  {
    type = "pulse";
    server = "192.168.7.155";
#   sink = ... /* default sink */
#   stream_name = "Utility channels";
#   continuous = false;
}
);
}
);
}
);
`

mattfox27, 2024-03-15:
I fixed it, I had the wrong gain value in my config file

nrussell88, 2024-03-15:
> I fixed it, I had the wrong gain value in my config file

Great news! Glad everything worked out well.

mattfox27, 2024-03-16:
So I was wrong, still gives that error

mattfox27, 2024-03-17:
> Mar 15 17:37:28 airband rtl_airband[13304]: Failed to initialize gain for device #-536870912: error -1071395636

so I was wrong, still get the error
im using a cheap blue SDR off Aliexpress not sure if that is the issue

**Mar 15 17:37:28 airband rtl_airband[13304]: Failed to initialize gain for device #-536870912: error -1071395636**

R1ch1e-Lee, 2024-03-18:
Have you determined the supported gain values of your sdr dongle? 

If you will install rtl-sdr and run a simple test it will show you the supported gain values.   Try this:

sudo apt install rtl-sdr

rtl_test -t

This should show you information about your dongle including gain values.   My NESDR dongle shows these values:

Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6

Hope this helps

OA6DXV, 2025-03-10:
> im using a cheap blue SDR off Aliexpress not sure if that is the issue

those are based on FC0012, gain values are different , the max gain value is around 20, use rtl_test to display the supported gain values