rtl-airband/RTLSDR-Airband #469: Error on Service Start Failed to connect to Pulse

mattfox27, 2024-03-11:
Triying to setup the autostart service on RPI

I get this error

`Mar 10 19:43:05 airband rtl_airband[1717]: pulse: 192.168.7.155: connection failed: Connection refused
`
No matter what I do I cannot get it to work, seems to work fine when I start from a command line

gabeschine, 2025-02-27:
@mattfox27 have you been able to make progress on this issue? Thanks for your patience.

firefighter76, 2025-02-27:
I don't use the autostart (systemd/service), but I could imagine that the following should be in the file rtl_airband.service under [Unit]: 
```
After=network.target pipewire-pulse.target pipewire.target
Wanted=network.target pipewire-pulse.target pipewire.target
```

mattfox27, 2025-02-27:
> [@mattfox27](https://github.com/mattfox27) have you been able to make progress on this issue? Thanks for your patience.

No I never got any further

gabeschine, 2025-02-28:
> > [@mattfox27](https://github.com/mattfox27) have you been able to make progress on this issue? Thanks for your patience.
> 
> No I never got any further

Please give the suggestions from @firefighter76 a try. I'm not seeing evidence so far this is an RTLSDR-Airband issue (could be wrong), but would be great to get you sorted anyway.