rtl-airband/RTLSDR-Airband #481: Is this project still alive?

jammedjackal, 2024-08-18:
Dear All,

I've been around for a while, as I made RTLSDR-Airband the engine of my ultimate radioscanning center. This software does a fantastic job of utilizing high-sample rate devices like HackRF as efficiently as possible. To be honest, I don't know if there's anything even close to this solution for monitoring a wide spectrum.

Unfortunately, I noticed that the latest release was in February, and the author seems to be inactive. Is this project still alive, and is anyone taking care of it? Were there any announcements about the termination of support?

It would be a shame to lose such an excellent piece of software, which could be even better, so I hope and keep my fingers crossed that the author will be back.

charlie-foxtrot, 2024-08-19:
I have unfortunately had to take a step back. I do intend to return but things are taking longer than I expected.  If I am unable to resume I will work to find someone who can take over.

gabeschine, 2025-02-27:
👋  I've stepped in to help @charlie-foxtrot while he's out. Hopefully he'll come back soon.

closed 2025-03-26 by gabeschine