rtl-airband/RTLSDR-Airband #482: [BUG] UDP throughput seems too low based on expected data types?

kristian1108, 2024-10-14:
**Describe your environment**

I'm using an RTLSDR dongle on the latest branch of RTLSDR-Airband (at this time, commit `e7cd0ece5b3f1a1c054d926bcecefa8166a3af62`).

I'm on my Mac OSX 14.1.1 Sonoma, with Apple M1 silicon.

**What happened?**

When I set up a UDP stream, I'm getting 8kb/s through the socket.

**What you expected to happen?**

I would expect to get 32kb/s through the socket based on this documentation: https://github.com/charlie-foxtrot/RTLSDR-Airband/wiki/Configuring-UDP-outputs

NFM disabled: 8000hz, 32-bit float samples = 8000 samples x 4 bytes per sample = 32 kilobytes per second.

**Steps to Reproduce**

`src/rtl_airband -c ../../bladerf-conf/rtl_airband.conf -f -e`

**Your rtl_airband.conf file**

```plaintext
devices:
({
  type = "rtlsdr";
  index = 0;
  gain = 20.0;
  centerfreq = 128.5;
  channels: (
    {
     freq = 128.175;
     modulation = "am";
     squelch_threshold = -100;
     outputs: (
          {
           type = "udp_stream";
           dest_address = "127.0.0.1";
           dest_port = 8081;
           continuous = true;
          },
          {
           type = "file";
           continuous = true;
           directory = "/Users/me/code/bladerf-conf";
           filename_template = "hello";
          }
      );
     }
  );
});
```

Note the output in the `.mp3` file sounds just fine, but I'd love to have the live stream. Thank you!!

SullivanChrisJ, 2025-02-22:
Enable NFM. See https://github.com/charlie-foxtrot/RTLSDR-Airband/wiki/Configuring-UDP-outputs

I'm set up for NFM (and raw IQ) only and get 16kHz.

SullivanChrisJ, 2025-02-22:
With NFM enabled I get the full data rate 16kHz 32-bit floats. I haven't tried with NFM disabled, and NFM is what I use it for.