rtl-airband/RTLSDR-Airband #486: [FEATURE] rdio-scanner API support

mscottco, 2024-12-05:
**Describe the solution you'd like**
I'd like if RTLSDR-Airband could upload recordings direct to Rdio-scanner using their API

**Is your feature request related to a problem? If so, please describe.**
I've currently got a P25 system setup in docker that uploads to rdio-scanner that works really well. For a while I've been playing with RTLSDR-Airband and working on getting the audio files uploaded to the same rdio-scanner instance however because I run all in docker, Rdio-scanners dirwatch feature doesn't work. 

**Describe alternative solutions or features you've considered**
I've tried setting up a non-docker instance of rdio-scanner but have had limited success with its reliability. I also prefer docker because its cleaner for running multiple applications on the same server.

Metadata support: I've also found that because of how rdio-scanner reads the file names, it's often difficult to quickly tell which channel you're listening to. The rdio-scanner API has options to include metadata about the transmission which would make it friendlier for silencing or detecting which frequency you're listening to.

**Additional context**
Details about the API for Rdio-scanner can be found here: [https://github.com/chuot/rdio-scanner/blob/master/docs/api.md](https://github.com/chuot/rdio-scanner/blob/master/docs/api.md)

It's something I'd happily help with debugging but I have limited python programming abilities and C is definitely a new language to me.

gabeschine, 2024-12-13:
It's possible @chuot can help here.

chris021, 2024-12-27:
I have a working docker config where rtl_airband runs in one container saving to a /recordings and a second container running a python script that uploads to rdio and deletes the recording from the same /recordings folder. happy to share that python script if its of any use. native support in rtl-airband would be neat tho!

mscottco, 2025-01-05:
@chris021 If you're willing to share, that would make an excellent interim solution!

blackwellj, 2025-03-15:
@chris021 I too would really appreciate your script