rtl-airband/RTLSDR-Airband #501: [BUG] adding flag to CMAKE doesn't seem to have effect

munnik, 2025-03-04:
**Describe your environment**

  - RTLSDR-Airband version you are using: v5.0.9
  - `make` options used to build the program: `cmake ../ -DNFM=ON -DRTLSDR=ON -DSOAPYSDR=ON -DMIRISDR=OFF -DPULSEAUDIO=OFF -DCMAKE_INSTALL_SYSCONFDIR=/etc && make`
  - Hardware platform (eg. x86_64, Raspberry Pi v4): x86
  - Operating system name and version: NixOS 24.11

**What happened?**
```
 munnik@turing ~/..../build  cmake ../ -DNFM=ON -DRTLSDR=ON -DSOAPYSDR=ON -DPULSEAUDIO=OFF -DCMAKE_INSTALL_SYSCONFDIR=/etc -DMIRISDR=OFF
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The CXX compiler identification is GNU 13.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/4apajimszc47rxwcpvc3g3rj2icinl71-gcc-wrapper-13.3.0/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to Release
-- Performing Test CXX_HAS_PTHREAD
-- Performing Test CXX_HAS_PTHREAD - Success
-- Performing Test CXX_HAS_FFAST_MATH
-- Performing Test CXX_HAS_FFAST_MATH - Success
-- Looking for sincosf
-- Looking for sincosf - found
-- Found PkgConfig: /nix/store/16h31npnbgpkhxp1041dsifiklbw4mpr-pkg-config-wrapper-0.29.2/bin/pkg-config (found version "0.29.2")
-- Checking for module 'libconfig++'
--   Found libconfig++, version 1.7.3
-- Found lame includes: /nix/store/9h8ga1x62h6spkycpjzskspb6257yn29-lame-3.100/include/lame/lame.h
-- Found lame library: /nix/store/qa6xya7klldgfhmjvlspg35xb3srgdqm-lame-3.100-lib/lib/libmp3lame.so
-- Checking for module 'shout'
--   Found shout, version 2.4.6
-- Looking for SHOUT_TLS_AUTO
-- Looking for SHOUT_TLS_AUTO - found
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN - found
-- Looking for SHOUT_TLS_RFC2818
-- Looking for SHOUT_TLS_RFC2818 - found
-- Looking for SHOUT_TLS_RFC2817
-- Looking for SHOUT_TLS_RFC2817 - found
-- Looking for SHOUT_TLS_DISABLED
-- Looking for SHOUT_TLS_DISABLED - found
-- Looking for shout_set_tls
-- Looking for shout_set_tls - found
-- Looking for shout_set_content_format
-- Looking for shout_set_content_format - found
-- Looking for shout_set_metadata_utf8
-- Looking for shout_set_metadata_utf8 - found
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.9.0
-- Found RTLSDR: /nix/store/5l1l0bnikawglc9ls2n85gr8i57j2v3w-rtl-sdr-librtlsdr-0.9.0/lib/librtlsdr.so
-- Checking for SoapySDR
--   SoapySDR found, /nix/store/zmapjm65fx587qa3cvynxqpxd1yriqd2-soapysdr-0.8.2-pre/include, SoapySDR
-- Performing Test CXX_HAS_MARCH_NATIVE
-- Performing Test CXX_HAS_MARCH_NATIVE - Success
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.10
-- RTLSDR-Airband configuration summary:

-- - Version string:            5.0.9
-- - Build type:                Release
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: OFF, enabled: FALSE
--   - soapysdr:                requested: ON, enabled: TRUE
-- - Other options:
--   - Platform:                native
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  FALSE
--   - NFM support:             ON
--   - PulseAudio:              requested: OFF, enabled: FALSE
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- Configuring done (9.0s)
-- Generating done (0.0s)
-- Build files have been written to: /home/munnik/Code/RTLSDR-Airband-5.0.9/build
 munnik@turing ~/..../build  make
[  5%] Generating version.cpp, _version.cpp
[ 10%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/config.cpp.o
[ 15%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-common.cpp.o
[ 21%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-file.cpp.o
[ 26%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-helpers.cpp.o
[ 31%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/mixer.cpp.o
[ 36%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/output.cpp.o
[ 42%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/rtl_airband.cpp.o
[ 47%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/squelch.cpp.o
[ 52%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/ctcss.cpp.o
[ 57%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/util.cpp.o
[ 63%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/udp_stream.cpp.o
[ 68%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/logging.cpp.o
[ 73%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/filters.cpp.o
[ 78%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/helper_functions.cpp.o
[ 78%] Generating version.cpp, _version.cpp
[ 84%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/version.cpp.o
[ 89%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-rtlsdr.cpp.o
[ 94%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-soapysdr.cpp.o
[ 94%] Built target rtl_airband_base
[100%] Linking CXX executable rtl_airband
[100%] Built target rtl_airband
 munnik@turing ~/..../build  ./src/rtl_airband
Cannot read configuration file /usr/local/etc/rtl_airband.conf
```

**What you expected to happen?**
`rtl_airband` is using `/etc/rtl_airband.conf` for it's configuration

**Steps to Reproduce**
See the commands above

**Additional context**
-

**Your rtl_airband.conf file**
Not modified and not relevant for this issue

gabeschine, 2025-03-04:
Can you try `-DSYSCONFDIR=/etc`?

Can I ask what made you expect `CMAKE_INSTALL_SYSCONFDIR` to define something? Is there documentation somewhere? I couldn't find anything.

munnik, 2025-03-04:
Thank you for your reply!

I tried it before and gave a warning (see the end of the cmake command output):
```
 munnik@turing ~/..../build  cmake ../ -DNFM=ON -DRTLSDR=ON -DSOAPYSDR=ON -DPULSEAUDIO=OFF -DSYSCONFDIR=/etc -DMIRISDR=OFF
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The CXX compiler identification is GNU 13.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/4apajimszc47rxwcpvc3g3rj2icinl71-gcc-wrapper-13.3.0/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to Release
-- Performing Test CXX_HAS_PTHREAD
-- Performing Test CXX_HAS_PTHREAD - Success
-- Performing Test CXX_HAS_FFAST_MATH
-- Performing Test CXX_HAS_FFAST_MATH - Success
-- Looking for sincosf
-- Looking for sincosf - found
-- Found PkgConfig: /nix/store/16h31npnbgpkhxp1041dsifiklbw4mpr-pkg-config-wrapper-0.29.2/bin/pkg-config (found version "0.29.2")
-- Checking for module 'libconfig++'
--   Found libconfig++, version 1.7.3
-- Found lame includes: /nix/store/9h8ga1x62h6spkycpjzskspb6257yn29-lame-3.100/include/lame/lame.h
-- Found lame library: /nix/store/qa6xya7klldgfhmjvlspg35xb3srgdqm-lame-3.100-lib/lib/libmp3lame.so
-- Checking for module 'shout'
--   Found shout, version 2.4.6
-- Looking for SHOUT_TLS_AUTO
-- Looking for SHOUT_TLS_AUTO - found
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN
-- Looking for SHOUT_TLS_AUTO_NO_PLAIN - found
-- Looking for SHOUT_TLS_RFC2818
-- Looking for SHOUT_TLS_RFC2818 - found
-- Looking for SHOUT_TLS_RFC2817
-- Looking for SHOUT_TLS_RFC2817 - found
-- Looking for SHOUT_TLS_DISABLED
-- Looking for SHOUT_TLS_DISABLED - found
-- Looking for shout_set_tls
-- Looking for shout_set_tls - found
-- Looking for shout_set_content_format
-- Looking for shout_set_content_format - found
-- Looking for shout_set_metadata_utf8
-- Looking for shout_set_metadata_utf8 - found
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.9.0
-- Found RTLSDR: /nix/store/5l1l0bnikawglc9ls2n85gr8i57j2v3w-rtl-sdr-librtlsdr-0.9.0/lib/librtlsdr.so
-- Checking for SoapySDR
--   SoapySDR found, /nix/store/zmapjm65fx587qa3cvynxqpxd1yriqd2-soapysdr-0.8.2-pre/include, SoapySDR
-- Performing Test CXX_HAS_MARCH_NATIVE
-- Performing Test CXX_HAS_MARCH_NATIVE - Success
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.10
-- RTLSDR-Airband configuration summary:

-- - Version string:            5.0.9
-- - Build type:                Release
-- - Operating system:          Linux
-- - SDR drivers:
--   - librtlsdr:               requested: ON, enabled: TRUE
--   - mirisdr:                 requested: OFF, enabled: FALSE
--   - soapysdr:                requested: ON, enabled: TRUE
-- - Other options:
--   - Platform:                native
--   - Build Unit Tests:        FALSE
--   - Broadcom VideoCore GPU:  FALSE
--   - NFM support:             ON
--   - PulseAudio:              requested: OFF, enabled: FALSE
--   - Profiling:               requested: OFF, enabled: FALSE
--   - Icecast TLS support:     TRUE
-- Configuring done (6.3s)
-- Generating done (0.0s)
CMake Warning:
  Manually-specified variables were not used by the project:

    SYSCONFDIR

-- Build files have been written to: /home/munnik/Code/RTLSDR-Airband-5.0.9/build
 munnik@turing ~/..../build  make
[  5%] Generating version.cpp, _version.cpp
[ 10%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/config.cpp.o
[ 15%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-common.cpp.o
[ 21%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-file.cpp.o
[ 26%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-helpers.cpp.o
[ 31%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/mixer.cpp.o
[ 36%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/output.cpp.o
[ 42%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/rtl_airband.cpp.o
[ 47%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/squelch.cpp.o
[ 52%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/ctcss.cpp.o
[ 57%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/util.cpp.o
[ 63%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/udp_stream.cpp.o
[ 68%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/logging.cpp.o
[ 73%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/filters.cpp.o
[ 78%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/helper_functions.cpp.o
[ 78%] Generating version.cpp, _version.cpp
[ 84%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/version.cpp.o
[ 89%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-rtlsdr.cpp.o
[ 94%] Building CXX object src/CMakeFiles/rtl_airband_base.dir/input-soapysdr.cpp.o
[ 94%] Built target rtl_airband_base
[100%] Linking CXX executable rtl_airband
[100%] Built target rtl_airband
 munnik@turing ~/..../build  ./src/rtl_airband
Cannot read configuration file /usr/local/etc/rtl_airband.conf
```

I'm a cmake newbie, so I'm just trying. This documentation, https://cmake.org/cmake/help/latest/module/GNUInstallDirs.html let me believe that `CMAKE_INSTALL_SYSCONFDIR` was the way to set the `/etc/` directory.

gabeschine, 2025-03-05:
Ok. I'm no `cmake` expert either :). The official way to specify a different config file is through the `-c` flag.

https://github.com/charlie-foxtrot/RTLSDR-Airband/wiki/Running#running-in-foreground

munnik, 2025-03-05:
I've fixed it with a patch in my Nix config for this package. See the `substituteInPlace` command. I've also overwritten the `find_version` script to just echo the version because the original script is not working in Nix.

```nix
{ stdenv, lib, fetchFromGitHub, pkgs }:
stdenv.mkDerivation rec {
  pname = "RTLSDR-Airband";
  version = "5.0.9";

  src = fetchFromGitHub {
    owner = "charlie-foxtrot";
    repo = "${pname}";
    rev = "v${version}";
    hash = "sha256-Rgsxim7FESVt1ZnVJFvVfMqJMgzVCfP1yBDBBO4zEsE=";
  };

  postPatch = ''
    echo '#!${pkgs.bash}/bin/bash' > scripts/find_version
    echo 'echo ${version}' >> scripts/find_version

    substituteInPlace src/rtl_airband.h --replace "/usr/local/etc" "$out/etc"
  '';

  buildInputs = with pkgs; [
    lame
    libshout
    libconfig
    fftwFloat
    rtl-sdr-librtlsdr
    soapysdr-with-plugins
  ];
  nativeBuildInputs = with pkgs; [ cmake pkg-config ];

  cmakeFlags = [
    "-DNFM=ON"
    "-DRTLSDR=ON"
    "-DSOAPYSDR=ON"
    "-DMIRISDR=OFF"
    "-DPULSEAUDIO=OFF"
  ];

  meta = with lib; {
    description =
      "RTLSDR-Airband receives analog radio voice channels and produces audio streams which can be routed to various outputs";
    homepage = "https://github.com/charlie-foxtrot/RTLSDR-Airband";
    license = licenses.gpl3Only;
    platforms = platforms.linux;
  };
}
```