rtl-airband/RTLSDR-Airband #522: [FEATURE] add argument to call script/command after file is written

jschmall, 2025-06-27:
**Is your feature request related to a problem? If so, please describe.**
Sort of. As of right now, there’s no way to call a script or command after a file is written.

**Describe the solution you'd like**
A simple argument to call a script or command after an output is finished. I understand this could be complicated for continuous streams but I write mp3 files to disk after each call.

**Describe alternative solutions or features you've considered**
Right now I use inotifywatch to watch the file directories and then rsync those files to my RDIO-Scanner instance to ingest. This is clunky and often takes upwards of 20 mins before it’s operational.

**Additional context**
Ideally this would allow a simpler integration with other systems for notifications and audio file playback.

yegors, 2025-06-29:
I've added this feature in my fork: https://github.com/yegors/RTLSDR-Airband/pull/8 

When using `type=file` and `split_on_transmission=true` you can define a new config param: `post_write_script = "/tmp/script.sh";`

The script will receive the filename that was written as the first arg. Quick example:

```
pi@adsbexchange:/tmp $ sudo cat /usr/local/etc/rtl_airband_file.conf
devices:
({
  type = "rtlsdr";
  index = 2;
  gain = 33.8;
  correction = 80;
  mode = "scan";
  channels:
  (
    {
      freqs = ( 128.8, 118.2, 124.75 );
      labels = ( "Tower", "Ground", "Approach" );
      outputs: (
        {
          type = "file";
          directory = "/tmp";
          filename_template = "test";
          split_on_transmission = true;
          min_rx_seconds = 3.0;
          post_write_script = "/tmp/script.sh";
        }
      );
    }
  );
 }
);
pi@adsbexchange:/tmp $ cat /tmp/script.sh
#!/bin/bash

# Check if exactly one argument is provided
if [ $# -ne 1 ]; then
    echo "Error: Exactly one argument (file path) is required"
    exit 1
fi

# Get the file path argument
FILE_PATH=$1

# Define log file path
LOG_FILE="/tmp/post_exec_hook.log"

# Get current date and time
CURRENT_DATE=$(date '+%Y-%m-%d %H:%M:%S')

# Write to log file
echo "[$CURRENT_DATE] File path: $FILE_PATH" >> $LOG_FILE

exit 0
pi@adsbexchange:/tmp $ cat post_exec_hook.log
[2025-06-29 18:05:36] File path: /tmp/test_20250629_180533.mp3
[2025-06-29 18:06:55] File path: /tmp/test_20250629_180651.mp3
[2025-06-29 18:09:20] File path: /tmp/test_20250629_180916.mp3
[2025-06-29 18:11:10] File path: /tmp/test_20250629_181106.mp3
```

jschmall, 2025-06-29:
This is exactly what I’m looking for. I’ll spin up a test environment tonight and test this out. Thank you!

jschmall, 2025-11-15:
> Sure, let me know how it goes. I tested it only for ~30 mins. Be mindful, my fork contains other new stuff which I made for myself: [#523](https://github.com/rtl-airband/RTLSDR-Airband/pull/523)
> 
> I'm not a C++ dev, and all of this is AI coded, so... use at own risk :)

Following back up on this.

The script call works perfectly, and I have no issues there. However, I'm switching to udp_stream output for some channels and I've noticed that this does NOT work with your fork. Is this something you've come across?